kraj/meta-clang #573: How can I know what standard is being used?

RaGrCobham, 2022-02-16:
I'm aiming for C++14, where can I specify this?

kraj, 2022-02-16:
@RaGrCobham if you are using new enough clang  ( clang >= 6.0 ) then compiler defaults is already set to use `c++14` so nothing would be required to be set. However if you want to be explicit, you can add `-std=c++14` to `CXXFLAGS`

kraj, 2022-02-16:
you can try to compile this example to see if compiler supports c++14 by default
`clang++ a.c` should compile this file if it supports c++14 by default. Otherwise try to add `-std=c++14`

```c++
#include <iostream>
#include <tuple>
#include <functional>

auto f() // this function returns multiple values
{
    int x = 5;
    return std::make_tuple(x, 7); // not "return {x,7};" because the corresponding
                                  // tuple constructor is explicit (LWG 2051)
}

int main()
{
    // heterogeneous tuple construction
    int n = 1;
    auto t = std::make_tuple(10, "Test", 3.14, std::ref(n), n);
    n = 7;
    std::cout << "The value of t is "  << "("
              << std::get<0>(t) << ", " << std::get<1>(t) << ", "
              << std::get<2>(t) << ", " << std::get<3>(t) << ", "
              << std::get<4>(t) << ")\n";

    // function returning multiple values
    int a, b;
    std::tie(a, b) = f();
    std::cout << a << " " << b << "\n";
}
```

RaGrCobham, 2022-02-18:
Thanks for your reply. Turns out we can't use clang as part of meta-sca static analysis layer because it will be removed.

But upon trying to remove the meta-clang layer from our bblayers.conf.sample, I'm hit with bitbake trying to build:

```
binutils-cross-aarch64-2.36
rpm-native_1.4.16
glibc-2.33
```
We never used these packages because they throw a fetch error, something that shouldn't happen because we have local storage. It's also weird that our repository's origin is changed to binutils repo. Our tracking info is changed.

Any ideas what could be going on?

RaGrCobham, 2022-02-22:
Sorry, we fixed the issue. It had nothing to do with the layer itself, our local source mirror was tainted with incomplete files that didn't have a checksum in the recipe I assume, so the error it threw was ambiguous.

closed 2022-02-22 by RaGrCobham