kraj/meta-clang #576: clang.bbclass: Handle big.LITTLE mtune options on arm/aarch64

yangpeng1995, 2022-02-18:
Hi @kraj ,
There may be some problems with this submission.
https://github.com/kraj/meta-clang/commit/46db90bf7ac1d7ab1078770423317071154c495c
line 35:
TUNE_CCARGS_append_toolchain-clang = "${@bb.utils.contains_any("TUNE_FEATURES", "cortexa75-cortex-a55 cortexa76-cortex-a55", " -mtune=cortex-a55", "", d)}"

I refer to default configuration of Yocto/Poky for cortexa76-cortexa55: 
https://git.yoctoproject.org/poky/tree/meta/conf/machine/include/arm/armv8-2a/tune-cortexa76-cortexa55.inc

I think "cortexa75-cortex-a55 cortexa76-cortex-a55" should modified to "cortexa75-cortexa55 cortexa76-cortexa55".
When I applying the cortexa76-cortex-A55 configuration in Yocto/Poky, then use Clang to compile some recipes. I noticed this minor error.

kraj, 2022-02-18:
@yangpeng1995 yes you are right its a typo. Would you mind submitting a patch ?

yangpeng1995, 2022-02-19:
Of course, I'd be happy to submit a patch to fix this minor problem.

yangpeng1995, 2022-02-19:
submitting a patch.
https://github.com/kraj/meta-clang/pull/577

closed 2022-03-17 by kraj