kraj/meta-clang #591: Support of meta-clang, kirkstone with Clang 12/LLVM 12

PreetiSachan, 2022-03-14:
**We really need support of meta-clang kirkstone branch with clang12 (LLVM12). The kirkstone is Long Term Support (until Apr. 2024).**

This is regarding applications failure (opencl-clang, intel-compute-runtime and intel-graphics-compiler) from [meta-intel, master/kirkstone](https://git.yoctoproject.org/meta-intel/tree/dynamic-layers/clang-layer/recipes-opencl?h=master) against meta-clang, [master/kirkstone](https://github.com/kraj/meta-clang/tree/master) branch.
Currently intel-graphics-compiler does not support [**LLVM14**](https://github.com/intel/intel-graphics-compiler/blob/master/documentation/build_ubuntu.md).

We did sanity build at our end and below three patches are required on top of meta-clang, hardknott branch to make it work with meta-intel, master/kirkstone branch.

**We request you to support us and if can create a new branch (e.g. **kirkstone-clang12**) based on **meta-clang, hardknott** and pushing these three commits.**

https://github.com/kraj/meta-clang/compare/hardknott...PreetiSachan:kirkstone-clang12?expand=1

1. https://github.com/PreetiSachan/meta-clang/commit/ee084c2b7d3a713c5ea4b6380d75920014fb2a76
It is only having new override syntax changes (Used script 'scripts/contrib/convert-overrides.py' from oe-core layer to convert).

2. https://github.com/PreetiSachan/meta-clang/commit/65a40c2e62c635edd9b170eb38877c6b91349901
Add kirkstone to LAYERSERIES_COMPAT.

3. https://github.com/PreetiSachan/meta-clang/commit/249568cf5608276dc4e21b2bbc3e1eb5b3664a1d
A newer busybox v1.35 exists in openembedded-core layer, thus deleted busybox_1.33%.bbappend. Changes are picked up from meta-clang, master branch.

Looking forward to hear from you ASAP.

Thanks and Regards
Preeti Sachan
preeti.sachan@intel.com

kraj, 2022-03-14:
kirkstone-clang12 branch is available, Please submit a pull with needed changes, I would like to know plan for clang14 as well. Since there are few months to stabilize it. I hope clang12 is just an intermediate step for meta-intel for kirkstone.

PreetiSachan, 2022-03-15:
Thanks for creating branch kirkstone-clang12.
Yes, those projects are working on fixing and enabling support for LLVM 13/14 but it will probably take some time before that is fully supported. Yes, clang12 support is just an intermediate step for meta-intel for kirkstone.

closed 2022-03-17 by kraj