kraj/meta-clang #603: libclc with LLVM 14 won't build

zboszor, 2022-03-29:
**Describe the bug**

There's a linker error with LLVM 14 when trying to build `libclc`:
```
| [114/2871] : && /data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-oe-linux/x86_64-oe-linux-clang++ -target x86_64-oe-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -target x86_64-oe-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now   -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now CMakeFiles/prepare_builtins.dir/utils/prepare-builtins.cpp.o -o prepare_builtins -L/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot/usr/usr/lib  && :
| FAILED: prepare_builtins
| : && /data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-oe-linux/x86_64-oe-linux-clang++ -target x86_64-oe-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -target x86_64-oe-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now   -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0=/usr/src/debug/libclc/14.0.0-r0                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now CMakeFiles/prepare_builtins.dir/utils/prepare-builtins.cpp.o -o prepare_builtins -L/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot/usr/usr/lib  && :
| /data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-oe-linux/x86_64-oe-linux-ld: /data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-oe-linux/x86_64-oe-linux-ld: DWARF error: invalid or unhandled FORM value: 0x23
| CMakeFiles/prepare_builtins.dir/utils/prepare-builtins.cpp.o: in function `main':
| prepare-builtins.cpp:(.text+0x2d): undefined reference to `llvm::LLVMContext::LLVMContext()'
| /data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/libclc/14.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-oe-linux/x86_64-oe-linux-ld: prepare-builtins.cpp:(.text+0x4f): undefined reference to `llvm::cl::ParseCommandLineOptions(int, char const* const*, llvm::StringRef, llvm::raw_ostream*, char const*, bool)'
[... cut ...] nValueCopyINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEE[_ZTVN4llvm2cl15OptionValueCopyINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEE]+0x18): undefined reference to `llvm::cl::GenericOptionValue::anchor()'
| clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
```

**To Reproduce**
Steps to reproduce the behavior:
1. bitbake libclc or enable opencl PACKAGECONFIG for mesa in Yocto 4.0 master

**Expected behavior**
libclc should build without errors.

kraj, 2022-03-29:
Interesting, I can build it fine for `qemux86-64` using latest meta-clang/master branch. Can you confirm that issue exists with latest top of trunk meta-clang ?

zboszor, 2022-03-30:
Previous build was using `42dd5484867b772767a33340ee219fb9f42799e4`.
Trying now with `17a6480b5e03a73d38b2d26c072aa51957ae747d`.
I am using `genericx86-64` from meta-yocto, or `intel-corei7-64` from meta-intel.
These machine setting should make no difference compared to `qemux86-64`.

zboszor, 2022-03-30:
`17a6480b5e03a73d38b2d26c072aa51957ae747d` fixed this linker error.

closed 2022-03-30 by zboszor