kraj/meta-clang #604: mesa 22.0.0 from Yocto 4.0 doesn't find LLVM 14

zboszor, 2022-03-29:
**Describe the bug**

```
| Run-time dependency libdrm_intel found: YES 2.4.110
| Run-time dependency libdrm_amdgpu found: YES 2.4.110
| Run-time dependency libdrm_radeon found: YES 2.4.110
| Run-time dependency libdrm_nouveau found: YES 2.4.110
| Run-time dependency libdrm found: YES 2.4.110
| Run-time dependency libudev found: NO (tried pkgconfig and cmake)
| ../mesa-22.0.0/meson.build:1663: WARNING: shared_llvm option "true" deprecated, please use "enabled" instead.
| ../mesa-22.0.0/meson.build:1676: WARNING: llvm option "true" deprecated, please use "enabled" instead.
| WARNING: Ignoring LLVM CMake dependency because dynamic was requested
| llvm-config found: YES (/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/mesa/2_22.0.0-r0/recipe-sysroot/usr/bin/crossscripts/llvm-config14.0.0) 14.0.0
| Run-time dependency LLVM (modules: amdgpu, asmparser, bitreader, bitwriter, core, engine, executionengine, instcombine, ipo, mcdisassembler, mcjit, native, scalaropts, transformutils, coroutines) found: NO (tried cmake and config-tool)
| Looking for a fallback subproject for the dependency llvm (modules: bitwriter, engine, mcdisassembler, mcjit, core, executionengine, scalaropts, transformutils, instcombine, amdgpu, native, bitreader, ipo, asmparser, native)
| Building fallback subproject with default_library=shared
| 
| ../mesa-22.0.0/meson.build:1696:2: ERROR: Neither a subproject directory nor a llvm.wrap file was found.
| 
| A full log can be found at /data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/mesa/2_22.0.0-r0/build/meson-logs/meson-log.txt
```
and
```
CMake binary for 1 is cached as not found
CMake binary for machine 1 not found. Giving up.
Run-time dependency libudev found: NO (tried pkgconfig and cmake)
../mesa-22.0.0/meson.build:1663: WARNING: shared_llvm option "true" deprecated, please use "enabled" instead.
../mesa-22.0.0/meson.build:1676: WARNING: llvm option "true" deprecated, please use "enabled" instead.
CMake binary for 1 is cached as not found
CMake binary for machine 1 not found. Giving up.
WARNING: Ignoring LLVM CMake dependency because dynamic was requested
llvm-config binary for 1 specified from cross file, native file, or env var as ['llvm-config14.0.0']
llvm-config found: YES (/data/genericx86-64-yocto-3.5/tmp-sicom-glibc/work/core2-64-oe-linux/mesa/2_22.0.0-r0/recipe-sysroot/usr/bin/crossscripts/llvm-config14.0.0) 14.0.0
llvm-config --shared-mode returned an error
Run-time dependency LLVM (modules: amdgpu, asmparser, bitreader, bitwriter, core, engine, executionengine, instcombine, ipo, mcdisassembler, mcjit, native, scalaropts, transformutils, coroutines) found: NO (tried cmake and config-tool)
Looking for a fallback subproject for the dependency llvm (modules: bitwriter, engine, mcdisassembler, mcjit, core, executionengine, scalaropts, transformutils, instcombine, amdgpu, native, bitreader, ipo, asmparser, native)
Building fallback subproject with default_library=shared
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kraj, 2022-03-29:
I can build it fine on `qemux86-64` how to reproduce it ?

zboszor, 2022-03-30:
I was using the `genericx86-64` BSP from meta-yocto but the machine should not make a difference.
Previous build was using `42dd5484867b772767a33340ee219fb9f42799e4`.
Trying now with `17a6480b5e03a73d38b2d26c072aa51957ae747d` for which the commit message looks like it would fix this problem.

zboszor, 2022-03-30:
`17a6480b5e03a73d38b2d26c072aa51957ae747d` fixed this problem.

closed 2022-03-30 by zboszor