kraj/meta-clang #614: Question: How to build kernel with clang

SquallATF, 2022-05-23:
I have try set
```
KERNEL_CC:pn-linux-rockchip = "${CCACHE}${HOST_PREFIX}clang ${HOST_CC_KERNEL_ARCH} -fuse-ld=bfd"
TOOLCHAIN:pn-linux-rockchip = "clang"
```
but build failed
```
Using .kernel-meta/configs///defconfig as base
Merging .kernel-meta/configs///cgroups.cfg
Merging .kernel-meta/configs///ext4.cfg
make[1]: Entering directory '/home/squallatf/yocto/build/tmp/work/rockchip_rk3566_evb-poky-linux/linux-rockchip/4.19-r0/linux-rockchip_rk3566_evb-standard-build'
/bin/sh: 1: aarch64-poky-linux-clang: not found
  HOSTCC  scripts/basic/fixdep
/bin/sh: 1: aarch64-poky-linux-clang: not found
  GEN     ./Makefile
/bin/sh: 1: aarch64-poky-linux-clang: not found
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --alldefconfig Kconfig
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-version.sh: 26: aarch64-poky-linux-clang: not found
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-version.sh: 27: aarch64-poky-linux-clang: not found
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-version.sh: 29: aarch64-poky-linux-clang: not found
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-version.sh: 26: aarch64-poky-linux-clang: not found
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-version.sh: 27: aarch64-poky-linux-clang: not found
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-version.sh: 29: aarch64-poky-linux-clang: not found
init/Kconfig:17: syntax error
init/Kconfig:16: invalid option
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/clang-version.sh: 15: aarch64-poky-linux-clang: not found
/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/gcc-plugin.sh: 11: aarch64-poky-linux-clang: not found
make[2]: *** [/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/scripts/kconfig/Makefile:69: alldefconfig] Error 1
make[1]: *** [/home/squallatf/yocto/build/tmp/work-shared/rockchip-rk3566-evb/kernel-source/Makefile:578: alldefconfig] Error 2
make[1]: Leaving directory '/home/squallatf/yocto/build/tmp/work/rockchip_rk3566_evb-poky-linux/linux-rockchip/4.19-r0/linux-rockchip_rk3566_evb-standard-build'
make: *** [Makefile:146: sub-make] Error 2
```

kraj, 2022-05-23:
interesting, do you see clang staged under `recipe-sysroot-native` directory in kernel build area ?

SquallATF, 2022-05-23:
only gcc toolchain under `recipe-sysroot-native` directory

kraj, 2022-05-23:
> only gcc toolchain under `recipe-sysroot-native` directory

That means setting TOOLCHAIN was not effective

SquallATF, 2022-05-24:
`TOOLCHAIN = "gcc"` is defined in [kernel-arch.bbclass#L67](https://github.com/openembedded/openembedded-core/blob/579906b9f4ea32c8d4e0e11a87ee6c9e27cdf0dc/meta/classes/kernel-arch.bbclass#L67), is there any way to override it or just modify it?

kraj, 2022-05-24:
Yeah I think turn that into 

‘TOOLCHAIN ?= “gcc”’

SquallATF, 2022-05-24:
> Yeah I think turn that into
> 
> `TOOLCHAIN ?= “gcc”`

I tried but still not work, I changed TOOLCHAIN to clang in `kernel-arch.bbclass`, but sill only gcc toolchain copied to `recipe-sysroot-native` directory

SquallATF, 2022-05-25:
I successfully compiled the kernel with clang, but not very elegant 
```
DEPENDS:pn-linux-rockchip:append = " clang-cross-aarch64"
KERNEL_CC:pn-linux-rockchip = "${CCACHE}clang ${HOST_CC_KERNEL_ARCH} -fuse-ld=lld ${DEBUG_PREFIX_MAP} -fdebug-prefix -map=${STAGING_KERNEL_DIR}=${KERNEL_SRC_PATH}"
KERNEL_LD:pn-linux-rockchip = "${CCACHE}ld.lld"
KERNEL_AR:pn-linux-rockchip = "${CCACHE}llvm-ar"
```

force add `clang-cross-aarch64` to `DEPENDS`

kraj, 2022-05-25:
> I successfully compiled the kernel with clang, but not very elegant
>
> ```
> DEPENDS:pn-linux-rockchip:append = " clang-cross-aarch64"
> KERNEL_CC:pn-linux-rockchip = "${CCACHE}clang [...]

yeah this is because kernel sets `INHIBIT_DEFAULT_DEPS` and then uses its own deps for toolchain components and assumes gcc toolchain. What you have is perhaps ok. You might want to move these all into a bbappend for kernel recipe and change
`DEPENDS:pn-linux-rockchip:append = " clang-cross-aarch64"` to `DEPENDS:append:toolchain-clang = " clang-cross-${TARGET_ARCH}"
`

SquallATF, 2022-05-26:
Thanks, here is final done `linux-rockchip_%bbappend` and `lto.cfg` for `linux-rockchip` package
```
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

TOOLCHAIN:forcevariable = "clang"

DEPENDS:append:toolchain-clang = " clang-cross-${TARGET_ARCH}"
KERNEL_CC:toolchain-clang = "${CCACHE}clang ${HOST_CC_KERNEL_ARCH} -fuse-ld=lld ${DEBUG_PREFIX_MAP} -fdebug-prefix-map=${STAGING_KERNEL_DIR}=${KERNEL_SRC_PATH}"
KERNEL_LD:toolchain-clang = "${CCACHE}ld.lld"
KERNEL_AR:toolchain-clang = "${CCACHE}llvm-ar"
SRC_URI:append:toolchain-clang = "\
	file://lto.cfg \
"

do_compile:prepend() {
	export LLVM_IAS=1
}

do_compile_kernelmodules:prepend() {
	export LLVM_IAS=1
}
```

```
CONFIG_LTO_CLANG=y
CONFIG_LTO=y
CONFIG_LTO_CLANG_THIN=y
```

closed 2022-05-26 by SquallATF

kraj, 2022-05-26:
@SquallATF awesome ! I would like to see this  retained perhaps as a section on README on how to compile kernel with clang or some such section. Would you might doing a doc patch like that ?

fcuzzocrea, 2022-07-01:
Hi! I am trying to do the same, but on riscv64.

After adding the lines suggested by @SquallATF the kernel compiler is recognized correctly, however config step fails.

```
DEBUG: Executing python function do_kernel_configcheck
Traceback (most recent call last):
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2783, in _expand_macro
    res += args[int(new_args[0])]
ValueError: invalid literal for int() with base 10: 'error-if'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/symbol_why.py", line 847, in <module>
    conf = kconfiglib.Kconfig( kconf, show_errors, show_errors )
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2914, in _parse_block
    while self._next_line():
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2236, in _next_line
    self._tokens = self._tokenize(line)
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2371, in _tokenize
    self._parse_assignment(s)
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2611, in _parse_assignment
    s, i = self._expand_macro(s, i, ())
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2787, in _expand_macro
    res += self._fn_val(new_args)
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 2850, in _fn_val
    return py_fn(self, *args)
  File "/media/fcuzzocrea/Yocto/kirkstone/build-core3/tmp-glibc/work/core3-oe-linux/do-linux/5.18.0-do1+gitAUTOINC+2604e4fd64-r0/recipe-sysroot-native/usr/bin/kconfiglib.py", line 6792, in _error_if_fn
    raise KconfigError("{}:{}: {}".format(
kconfiglib.KconfigError: scripts/Kconfig.include:50: Sorry, this assembler is not supported.
ERROR: config analysis failed: 
DEBUG: Python function do_kernel_configcheck finished
```

It seems that the kernel build system does not regonize clang assembler ? 

Do you have any hints on this? Thanks!!!

fcuzzocrea, 2022-07-01:
Alright, figured out that I need to add -no-integrated-as to KERNEL_CC:toolchain-clang!

SquallATF, 2022-07-03:
> Alright, figured out that I need to add -no-integrated-as to KERNEL_CC:toolchain-clang!

should set `LLVM_IAS=0` to build kernel