kraj/meta-clang #623: Task "meta-clang/recipes-devtools/clang/clang_git.bb:do_compile" failed with exit code '1'

TejasAbhang4042, 2022-06-29:
Hello,

I am a beginner with Yocto and I want to run chromium browser with Yocto dunfell. I read your comment in one of the issues that latest dunfell-clang12 branch of meta-clang is needed by meta-chromium.

So I cloned the branch in sources directory, added the path in my "bblayers.conf" file.
I have also enabled virtualization in the DISTRO_FEATURES.
But I am unable to compile meta-clang.
The task of compiling "meta-clang/recipes-devtools/clang/clang_git.bb" is failing.

I have also tried other branches like dunfell and clang12.

Am I missing something or doing something wrong?

Please help.

TejasAbhang4042, 2022-07-01:
Well I got the issue. I was using gcc version 5.4. I guess due to this it was failing to compile.
So after upgrading the gcc version to 8, it got compiled successfully.

kraj, 2022-08-24:
closing based on above comment. Please re-open if its not resolved.

closed 2022-08-24 by kraj