kraj/meta-clang #632: Can't get TARGET_PREFIX-clang etc to end up in SDK

dir-ableton, 2022-07-26:
**Describe the bug**
In our process of upgrading to dunfell from zeus we don't see the symbolic links in the form of ${TARGET_PREFIX}$tool being created. I do see these in `build/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/clang/10.0.1-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-clang`, but they do not end up in the sdk.sh. As a result the meta-environment.bbappend creates non-functional env vars.

**To Reproduce**
Steps to reproduce the behavior:

I just added the layer and placed CLANGSDK="1" in my local.conf, plus CORE_IMAGE_EXTRA_INSTALL_append = " clang " in my image recipe. 

**Expected behavior**
Symbolic links to clang/clang++ with the TARGET_PREFIX inside the SDK.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 -

dir-ableton, 2022-07-27:
A bit of follow-up: in my desperation, I took the zeus-branch and made it dunfell-compatible. It was simple, besides declaring it compatible just replacing pythonnative with python3native, and cherry-picking a limits-include patch you did later. This builds  (clang 9 obviously), but *including* the required arm-oe-linux-gnueabi-prefix.

kraj, 2022-08-24:
@dir-ableton are you expecting these symlinks inside the target sysroot or inside the native sysroot inside SDK

dir-ableton, 2022-08-24:
This is where they live in a zeus-based release:

```
<prefix>/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux/arm-oe-linux-clang++
<prefix>/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-musl/arm-oe-linux-musl-clang++
<prefix>/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-musleabi/arm-oe-linux-musleabi-clang++
<prefix>/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-clang++
```
And my understandig of https://github.com/kraj/meta-clang/blob/3bb001d3f364bbf6588fed04b9ee2c7e74a0beba/recipes-devtools/clang/clang-crosssdk_git.bb#L20 is that this responsible for this. It didn't change since zeus, but seems not to work anymore.

dir-ableton, 2022-09-27:
I'm closing this because in a second attempt at upgrading to dunfell, this issue didn't surface. It's a bit frustrating for me not to understand why, but then that's on me ;)

closed 2022-09-27 by dir-ableton