kraj/meta-clang #645: buildpath warnings on master for compiler-rt, libcxx

rwmacleod, 2022-08-24:
When building chromium on master, I see:

WARNING: libcxx-14.0.6-r0 do_package_qa: QA Issue: File /usr/lib/.debug/libc++abi.so.1.0 in package libcxx-dbg contains reference to TMPDIR [buildpaths]
WARNING: compiler-rt-14.0.6-r0 do_package_qa: QA Issue: File /usr/lib/clang/14.0.6/lib/linux/libclang_rt.builtins-armhf.a in package compiler-rt-staticdev contains reference to TMPDIR [buildpaths]
Setscene tasks: 2827 of 2827 

**To Reproduce**
Steps to reproduce the behavior:
1. use poky+meta-oe+meta-clang+meta-browser/meta-chromium on ubunutu-18.04.6 and maybe other distros.
2. bitbake chromium 

**Expected behavior**
No such warnings.

kraj, 2022-08-24:
this is known problem with clang older than 15.x, it has been fixed thats the good news. I have 15.x in tests right now https://github.com/YoeDistro/meta-clang/commits/kraj/clang15
its still WIP but if you are interested give it a shot and let me know.

rossburton, 2023-01-04:
More buildpaths warnings:
```
WARNING: clang-15.0.6-r0 do_package_qa: QA Issue: File /usr/bin/llvm-config in package clang contains reference to TMPDIR [buildpaths]
WARNING: clang-15.0.6-r0 do_package_qa: QA Issue: File /usr/lib/cmake/llvm/LLVMConfig.cmake in package clang-dev contains reference to TMPDIR [buildpaths]
WARNING: openmp-15.0.6-r0 do_package_qa: QA Issue: File /usr/lib/libomptarget.devicertl.a in package openmp-staticdev contains reference to TMPDIR [buildpaths]
```

kraj, 2023-01-04:
yes have seen first two of these too.

rossburton, 2023-01-11:
The openmp workaround is now merged in b573f41170c4e3be8d5fc585bb8166218ed09d8b.

clang is interesting and I think I'm going to leave it to you Khem. :)

One of the entries in `/usr/lib/cmake/llvm/LLVMConfig.cmake` is:

```
set(LLVM_DEFAULT_EXTERNAL_LIT "/yocto/ross/build/tmp/work/armv8a-poky-linux/clang/15.0.6-r0/build/./bin/llvm-lit")
```

That's the default location for `llvm-lit`, which isn't going to work.  There is a chain of assignments:
```
LLVMConfig.cmake.in: set(LLVM_DEFAULT_EXTERNAL_LIT "@LLVM_CONFIG_DEFAULT_EXTERNAL_LIT@")
llvm/cmake/modules/CMakeLists.txt: set(LLVM_CONFIG_DEFAULT_EXTERNAL_LIT "${LLVM_CONFIG_TOOLS_BINARY_DIR}/llvm-lit")
llvm/cmake/modules/CMakeLists.txt: set(LLVM_CONFIG_TOOLS_BINARY_DIR "${LLVM_TOOLS_BINARY_DIR}")
clang/CMakeLists.txt: set(TOOLS_BINARY_DIR "${LLVM_TOOLS_BINARY_DIR}")
clang/CMakeLists.txt: set(LLVM_TOOLS_BINARY_DIR "${TOOLS_BINARY_DIR}" CACHE PATH "Path to llvm/bin")
```

I thought I could set `-DLLVM_TOOLS_BINARY_DIR=${bindir}` but that didn't seem to do the right thing.

rossburton, 2023-03-08:
With current master I can't replicate any of these, I think this can be closed.

closed 2023-08-31 by rwmacleod