kraj/meta-clang #646: Support for building with Clang inside Yocto the same way as against SDK

tronical, 2022-08-25:
**Is your feature request related to a problem? Please describe.**

We have a piece of software (Rust Skia bindings) that, among other things, compiles some sources with clang. Thanks to meta-clang it's quite easy to build against a Yocto SDK that's equipped with clang: `clang-environment.inc` takes care of making sure that the `CLANGCC`/`CLANGCXX`/`CLANGCPP` environment variables are set. The software in question will pick up clang as specified in these environment variables and everything works out of the box.

However when building the same software inside Yocto, via a dedicated bitbake recipe, these environment variables are not set and we end up picking up the wrong compiler.

**Describe the solution you'd like**

It would be great to have an equivalent of `clang-environment.inc`, perhaps as bbclass, that - instead of directing the entire build to use clang - just sets up the environment in a similar way.

As a workaround we can of course emulate that `clang-environment.inc` does, but it seems that this would perhaps be useful to support right here.

kraj, 2022-10-06:
@tronical SDKs are more or less dev friendly environments. When including clang we ship both gcc and clang based toolchains and CC is still gcc so CLANGCC is added to point to clang in SDK. Howver, if you then bring the package into yocto build system via recipe, you will have to do few tweaks. Firstly you want to add

```
TOOLCHAIN = "clang" 
```

in recipe. This should tell the build system that it should use clang to build this one component. You also need to include
meta-clang ofcourse in your bblayers.

kraj, 2023-03-30:
closing this if its still an issue please reopen.

closed 2023-03-30 by kraj