kraj/meta-clang #648: Dunfell-clang12: Build issues for mesa 20.0.2 (gallium)

fwuehr95, 2022-08-30:
Hi @kraj ,

I am facing issues with compiling mesa since I updated meta-clang to dunfell-clang12 in our Yocto project. We are still on dunfell though, with the dunfell branch of meta-clang everything works fine.

The following error occurs:

x86_64-rational-linux-gcc -m64 -march=x86-64 -mtune=generic -msse2 -mfpmath=sse -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot -Isrc/gallium/auxiliary/eb820e8@@gallium@sta -Isrc/gallium/auxiliary -I../mesa-20.0.2/src/gallium/auxiliary -Isrc/loader -I../mesa-20.0.2/src/loader -I../mesa-20.0.2/src/gallium/include -Isrc -I../mesa-20.0.2/src -Iinclude -I../mesa-20.0.2/include -I../mesa-20.0.2/src/gallium/auxiliary/util -Isrc/compiler/nir -I../mesa-20.0.2/src/compiler/nir -Isrc/util -I../mesa-20.0.2/src/util -I/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot/usr/include/libdrm -I/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot/usr/include -fdiagnostics-color=always -DNDEBUG -pipe -D_FILE_OFFSET_BITS=64 -std=c99 '-DPACKAGE_VERSION="20.0.2"' '-DPACKAGE_BUGREPORT="https://gitlab.freedesktop.org/mesa/mesa/issues"' -DUSE_ELF_TLS -DENABLE_ST_OMX_BELLAGIO=0 -DENABLE_ST_OMX_TIZONIA=0 -DHAVE_X11_PLATFORM -DGLX_INDIRECT_RENDERING -DGLX_DIRECT_RENDERING -DGLX_USE_DRM -DHAVE_DRM_PLATFORM -DHAVE_SURFACELESS_PLATFORM -DENABLE_SHADER_CACHE -DHAVE___BUILTIN_BSWAP32 -DHAVE___BUILTIN_BSWAP64 -DHAVE___BUILTIN_CLZ -DHAVE___BUILTIN_CLZLL -DHAVE___BUILTIN_CTZ -DHAVE___BUILTIN_EXPECT -DHAVE___BUILTIN_FFS -DHAVE___BUILTIN_FFSLL -DHAVE___BUILTIN_POPCOUNT -DHAVE___BUILTIN_POPCOUNTLL -DHAVE___BUILTIN_UNREACHABLE -DHAVE_FUNC_ATTRIBUTE_CONST -DHAVE_FUNC_ATTRIBUTE_FLATTEN -DHAVE_FUNC_ATTRIBUTE_MALLOC -DHAVE_FUNC_ATTRIBUTE_PURE -DHAVE_FUNC_ATTRIBUTE_UNUSED -DHAVE_FUNC_ATTRIBUTE_WARN_UNUSED_RESULT -DHAVE_FUNC_ATTRIBUTE_WEAK -DHAVE_FUNC_ATTRIBUTE_FORMAT -DHAVE_FUNC_ATTRIBUTE_PACKED -DHAVE_FUNC_ATTRIBUTE_RETURNS_NONNULL -DHAVE_FUNC_ATTRIBUTE_VISIBILITY -DHAVE_FUNC_ATTRIBUTE_ALIAS -DHAVE_FUNC_ATTRIBUTE_NORETURN -DHAVE_UINT128 -DUSE_SSE41 -DUSE_GCC_ATOMIC_BUILTINS -DMAJOR_IN_SYSMACROS -DHAVE_SYS_SYSCTL_H -DHAVE_LINUX_FUTEX_H -DHAVE_ENDIAN_H -DHAVE_DLFCN_H -DHAVE_EXECINFO_H -DHAVE_SYS_SHM_H -DHAVE_CET_H -DHAVE_STRTOF -DHAVE_MKOSTEMP -DHAVE_TIMESPEC_GET -DHAVE_MEMFD_CREATE -DHAVE_RANDOM_R -DHAVE_FLOCK -DHAVE_STRTOK_R -DHAVE_PROGRAM_INVOCATION_NAME -DHAVE_POSIX_MEMALIGN -DHAVE_DIRENT_D_TYPE -DHAVE_STRTOD_L -DHAVE_DLADDR -DHAVE_DL_ITERATE_PHDR -DHAVE_ZLIB -DHAVE_PTHREAD -DHAVE_PTHREAD_SETAFFINITY -DHAVE_LIBDRM -DLLVM_AVAILABLE '-DMESA_LLVM_VERSION_STRING="12.0.0"' -DHAVE_DRI3 -DHAVE_DRI3_MODIFIERS -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=return-type -Werror=empty-body -Werror=incompatible-pointer-types -Werror=int-conversion -Wno-missing-field-initializers -Wno-format-truncation -fno-math-errno -fno-trapping-math -Werror=format -Wformat-security -O2 -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0=/usr/src/debug/mesa/2_20.0.2-r0 -fdebug-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0=/usr/src/debug/mesa/2_20.0.2-r0 -fdebug-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot= -fdebug-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot-native= -fno-omit-frame-pointer -fPIC -pthread -D__STDC_FORMAT_MACROS -D_GNU_SOURCE -D__STDC_LIMIT_MACROS -D__STDC_CONSTANT_MACROS -fvisibility=hidden -Werror=pointer-arith -Werror=vla -MD -MQ 'src/gallium/auxiliary/eb820e8@@gallium@sta/gallivm_lp_bld_format_srgb.c.o' -MF 'src/gallium/auxiliary/eb820e8@@gallium@sta/gallivm_lp_bld_format_srgb.c.o.d' -o 'src/gallium/auxiliary/eb820e8@@gallium@sta/gallivm_lp_bld_format_srgb.c.o' -c ../mesa-20.0.2/src/gallium/auxiliary/gallivm/lp_bld_format_srgb.c
../mesa-20.0.2/src/gallium/auxiliary/gallivm/lp_bld_format_srgb.c: In function 'lp_build_linear_to_srgb':
../mesa-20.0.2/src/gallium/auxiliary/gallivm/lp_bld_format_srgb.c:194:44: error: implicit declaration of function 'powf' [-Werror=implicit-function-declaration]
  194 |                                  exp2f_c * powf(coeff_f, 1.0f / exp_f));
      |                                            ^~~~
../mesa-20.0.2/src/gallium/auxiliary/gallivm/lp_bld_format_srgb.c:194:44: warning: incompatible implicit declaration of built-in function 'powf'
../mesa-20.0.2/src/gallium/auxiliary/gallivm/lp_bld_format_srgb.c:78:1: note: include '<math.h>' or provide a declaration of 'powf'
   77 | #include "lp_bld_format.h"
  +++ |+#include <math.h>
   78 | 
cc1: some warnings being treated as errors

FAILED: src/gallium/auxiliary/eb820e8@@gallium@sta/gallivm_lp_bld_init.c.o 
x86_64-rational-linux-gcc -m64 -march=x86-64 -mtune=generic -msse2 -mfpmath=sse -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot -Isrc/gallium/auxiliary/eb820e8@@gallium@sta -Isrc/gallium/auxiliary -I../mesa-20.0.2/src/gallium/auxiliary -Isrc/loader -I../mesa-20.0.2/src/loader -I../mesa-20.0.2/src/gallium/include -Isrc -I../mesa-20.0.2/src -Iinclude -I../mesa-20.0.2/include -I../mesa-20.0.2/src/gallium/auxiliary/util -Isrc/compiler/nir -I../mesa-20.0.2/src/compiler/nir -Isrc/util -I../mesa-20.0.2/src/util -I/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot/usr/include/libdrm -I/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot/usr/include -fdiagnostics-color=always -DNDEBUG -pipe -D_FILE_OFFSET_BITS=64 -std=c99 '-DPACKAGE_VERSION="20.0.2"' '-DPACKAGE_BUGREPORT="https://gitlab.freedesktop.org/mesa/mesa/issues"' -DUSE_ELF_TLS -DENABLE_ST_OMX_BELLAGIO=0 -DENABLE_ST_OMX_TIZONIA=0 -DHAVE_X11_PLATFORM -DGLX_INDIRECT_RENDERING -DGLX_DIRECT_RENDERING -DGLX_USE_DRM -DHAVE_DRM_PLATFORM -DHAVE_SURFACELESS_PLATFORM -DENABLE_SHADER_CACHE -DHAVE___BUILTIN_BSWAP32 -DHAVE___BUILTIN_BSWAP64 -DHAVE___BUILTIN_CLZ -DHAVE___BUILTIN_CLZLL -DHAVE___BUILTIN_CTZ -DHAVE___BUILTIN_EXPECT -DHAVE___BUILTIN_FFS -DHAVE___BUILTIN_FFSLL -DHAVE___BUILTIN_POPCOUNT -DHAVE___BUILTIN_POPCOUNTLL -DHAVE___BUILTIN_UNREACHABLE -DHAVE_FUNC_ATTRIBUTE_CONST -DHAVE_FUNC_ATTRIBUTE_FLATTEN -DHAVE_FUNC_ATTRIBUTE_MALLOC -DHAVE_FUNC_ATTRIBUTE_PURE -DHAVE_FUNC_ATTRIBUTE_UNUSED -DHAVE_FUNC_ATTRIBUTE_WARN_UNUSED_RESULT -DHAVE_FUNC_ATTRIBUTE_WEAK -DHAVE_FUNC_ATTRIBUTE_FORMAT -DHAVE_FUNC_ATTRIBUTE_PACKED -DHAVE_FUNC_ATTRIBUTE_RETURNS_NONNULL -DHAVE_FUNC_ATTRIBUTE_VISIBILITY -DHAVE_FUNC_ATTRIBUTE_ALIAS -DHAVE_FUNC_ATTRIBUTE_NORETURN -DHAVE_UINT128 -DUSE_SSE41 -DUSE_GCC_ATOMIC_BUILTINS -DMAJOR_IN_SYSMACROS -DHAVE_SYS_SYSCTL_H -DHAVE_LINUX_FUTEX_H -DHAVE_ENDIAN_H -DHAVE_DLFCN_H -DHAVE_EXECINFO_H -DHAVE_SYS_SHM_H -DHAVE_CET_H -DHAVE_STRTOF -DHAVE_MKOSTEMP -DHAVE_TIMESPEC_GET -DHAVE_MEMFD_CREATE -DHAVE_RANDOM_R -DHAVE_FLOCK -DHAVE_STRTOK_R -DHAVE_PROGRAM_INVOCATION_NAME -DHAVE_POSIX_MEMALIGN -DHAVE_DIRENT_D_TYPE -DHAVE_STRTOD_L -DHAVE_DLADDR -DHAVE_DL_ITERATE_PHDR -DHAVE_ZLIB -DHAVE_PTHREAD -DHAVE_PTHREAD_SETAFFINITY -DHAVE_LIBDRM -DLLVM_AVAILABLE '-DMESA_LLVM_VERSION_STRING="12.0.0"' -DHAVE_DRI3 -DHAVE_DRI3_MODIFIERS -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=return-type -Werror=empty-body -Werror=incompatible-pointer-types -Werror=int-conversion -Wno-missing-field-initializers -Wno-format-truncation -fno-math-errno -fno-trapping-math -Werror=format -Wformat-security -O2 -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0=/usr/src/debug/mesa/2_20.0.2-r0 -fdebug-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0=/usr/src/debug/mesa/2_20.0.2-r0 -fdebug-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0/recipe-sysroot= -fdebug-prefix-map=/home/wufl/git/yocto-rational-os/build-x86-64/tmp/work/x86_64-rational-linux/mesa/2_20.0.2-r0 [... cut ...]
  171 |       LLVMAddConstantPropagationPass(gallivm->passmgr);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       LLVMAddCorrelatedValuePropagationPass
cc1: some warnings being treated as errors

Of course these are warning, so it should be quite straight forward to mask them, but the error prints keep me wondering. Is there some incompatibility between the LLVM versions between dunfell and dunfell-clang that causes this issue ?

If you need further information or I can help anyway please let me know.

Kind regards,

Florian

fwuehr95, 2022-08-30:
Looks to be an issue with mesa, I have opened up a pull request to be able to build mesa with the newer LLVM as shipped by meta-clang´s dunfell-clang12 branch. Changes have been extracted out of official mesa git repository.

kraj, 2022-10-06:
This is fixed with https://github.com/kraj/meta-clang/pull/649

closed 2022-10-06 by kraj