kraj/meta-clang #651: [CRITICAL] new clang-native.bbclass in incorrect folder

bmx666, 2022-09-03:
build failed due to this commit -> https://github.com/kraj/meta-clang/commit/d89de99cbbb171a21f37ce64dd9ec2ef79fa033a
because bbclass MUST BE located in **classes** folder, not in **classes-recipe**

kraj, 2022-09-03:
are you on latest `master` branch of bitbake/poky ? if not please update retry and report back.

bmx666, 2022-09-03:
Hi @kraj thanks for quick update, sorry for that.
But now our servers build NXP and BoundaryDevices which pointed somehow on meta-clang "master" in all branches, seems they all affected now... Like example
https://github.com/boundarydevices/boundary-bsp-platform/blob/kirkstone/default.xml

closed 2022-09-03 by bmx666

kraj, 2022-09-03:
> Hi @kraj thanks for quick update, sorry for that. But now our servers build NXP and BoundaryDevices that points on **meta-clang** "master" in all branches, seems they all affected now... Like example https://github.com/boundarydevices/boundary-bsp-platform/blob/kirkstone/default.xml

hmm you are mixing kirkstone and master but I guess keeping this in classes/ is a good workaround for now. I will push such a change into meta-clang master for now.