kraj/meta-clang #672: clang-tools-extra into a seperate package without clang compiler for sdk

Lorac, 2022-10-05:
**Is your feature request related to a problem? Please describe.**
I would like to have clang-tools-extra in a separate package so that I can install those tools without clang itself into an SDK.
One issue I ran into is that if I include meta-clang layer, mesa will now build with llvm provided by meta-clang instead of poky.

With `CLANGSDK = 0` if we are using mesa with gallium-llvm, the compiler is still present in the SDK.

**Describe the solution you'd like**
I would like to be able to install inside an SDK something like `nativesdk-clang-tools-extra` without the clang compiler, etc.

**Describe alternatives you've considered**
The current workaround is to override the current override for the PACKAGECONFIG[gallium-llvm] to restore the one from poky mesa recipe.

We have to redefine the YOCTO_ALTERNATE_EXE_PATH to the original mesa from poky instead of the one from meta-clang.

re-override mesa in our layer
```
PACKAGECONFIG[gallium-llvm] = "-Dllvm=enabled -Dshared-llvm=enabled, -Dllvm=disabled, llvm${MESA_LLVM_RELEASE} llvm-native \
                               elfutils"

export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm${MESA_LLVM_RELEASE}/llvm-config"

```

override on clang
```
PROVIDES:remove = "llvm llvm-native"
PROVIDES:remove:class-native = " llvm-native"
```

Lorac, 2022-10-06:
@kraj In the best case scenario for us if I can do something like this:

```
TOOLCHAIN_HOST_TASK += " \
    nativesdk-clang-tools-extra \
"
```
And only install stuff from that target.

At the moment I tried something like this:

```
PACKAGE_BEFORE_PN:append = "clang-tools-extra"
RDEPENDS:clang-tools-extra = ""
RDEPENDS:clang-tools-extra:append = " ${PN}-libllvm libclang python3"

FILES:libclang = "\
  ${libdir}/libclang.so.*\
"

FILES:clang-tools-extra = " \
    ${bindir}/clang-tidy \
    ${bindir}/run-clang-tidy \
    ${bindir}/clang-format \
"
```

But for some reason, it will pull the clang compiler anyway and I haven't figured why yet.

kraj, 2022-10-06:
check the shared libraries that clang-tidy needs. Perhaps it depens on libclang. bitbake -g might be able to show the depchain.

Lorac, 2022-10-06:
clang-tidy depends on libclang and libllvm, but when running bitbake -g it doesn't help with `PACKAGES`, afaik. It just tells me that my meta-toolchain recipe requires nativesdk-clang which makes sense because I added:
```
TOOLCHAIN_HOST_TASK += " \
    nativesdk-clang-tools-extra \
"
```

But that specific package doesn't require the clang compiler itself, just the .so.

kraj, 2022-10-06:
OK `CLANGSDK="0"` in local.conf perhaps will help in removing the direct RDEPENDS in nativesdk-packagegroup-sdk-host

Lorac, 2022-10-06:
`CLANGSDK` is already at 0. This is why it's the most confusing.

kraj, 2022-10-07:
@Lorac can you try the proposed fix in https://github.com/kraj/meta-clang/pull/674

Lorac, 2022-10-07:
@kraj looks good, but I haven't tested it yet. I wonder why the clang-tidy package doesn't RDEPENDS on `libllvm libclang`. When you look at libraries that clang-tidy links to, libllvm and libclang are present. I'll be away for a couple of days. Is it possible to backport changes to kirkstone since it's an LTS for poky?

I think `clang-apply-replacements` is needed for clang-tidy. Or at least, afaik run-clang-tidy can use it to apply replacements.

Another subject, maybe another issue to talk about, can kirkstone branch be supported with newer version of clang too?

kraj, 2022-10-07:
@Lorac its deps are looking right to me here

```
PV = 15.0.2
PR = r0
PKGR = r0.8
RPROVIDES =
RDEPENDS = clang-libclang-cpp (>= 15.0.2) clang-libllvm (>= 15.0.2) libcxx (>= 15.0.2) musl (>= 1.2.3+git1+dc9285ad1d)
RRECOMMENDS =
PKGSIZE = 27658533
FILES = /usr/bin/*clang-tidy*
FILELIST = /usr/bin/clang-tidy /usr/bin/run-clang-tidy
```

I am all for backporting this change to kirkstone, regarding clang15 it might be a bit involved process since compiler has changed a lot and all packages may not compile so perhaps creating a separate branch `kirkstone_clang15` or somesuch and then maturing it might be the way to go.

for clang-apply-replacements its perhaps better to install `clang-tools` in sdk as well or maybe create a RRECOMMENDS for it in `clang-tidy` package

Lorac, 2022-10-07:
Interesting output is that oe-pkg-util script?

If you are doing a kirkstone_clang15 are you going to do a
kirkstone_clang16 too. When to stop? Or it’s more of an experiment and when
kirkstone_clang15 is stable enough it will merge in kirkstone?

I agree that the rrecommends is probably a good idea for clang-tidy and
clang-apply-replacement.

closed 2022-10-08 by kraj

Lorac, 2022-10-11:
@kraj When do you think the changes will be available to kirkstone?

kraj, 2022-10-11:
Help me with backport of you can

Lorac, 2022-10-11:
> Help me with backport of you can

I'll be testing the backported change onto kirkstone branch and will submit it when I confirm it works.

Lorac, 2022-10-11:
When I install 
```
    nativesdk-clang-tidy \
    nativesdk-clang-format \
```

To the SDK, `./sysroots/x86_64-ekosdk-linux/usr/bin/clang` is present inside the native sdk.

There is `./lib/clang/14.0.3/include/mmintrin.h` headers file too that are not necessary, if I don't want the clang compiler.