kraj/meta-clang #676: When installing nativesdk-clang-tools, for an sdk, it installs clang compiler

Lorac, 2022-10-12:
**Describe the bug**
Given that I only want the clang tooling but not the compiler. If I decide to install `nativesdk-clang-tools` it will install clang compiler which isn't what I wanted to install.

**To Reproduce**
Add `nativesdk-clang-tools` to TOOLCHAIN_HOST_TASK

**Expected behavior**
Only install `nativesdk-clang-tools` binaries and the needed .so, but not the compiler and its headers, intrinsic, etc.

**Desktop (please complete the following information):**
 - OS: Linux, Debian 11

kraj, 2022-10-12:
its interesting who is pulling nativesdk-clang. Can you look into output of `bitbake -g nativesdk-clang-tools` or post the output here as well.

Lorac, 2022-10-12:
nativesdk-clang-tools is a package and can't be used to do that on kirkstone. I think you mean `bitbake -g nativesdk-clang` ?

Lorac, 2022-10-12:
I know that mesa depends on clang. But it doesn't RDEPENDS, maybe the way mesa depends on clang is incorrect?

kraj, 2022-10-12:
Well may be add -g to sdk target build

Lorac, 2022-10-12:
task-depends.dot of the -g sdk target build: https://file.io/qFVjQv9bSvNX

kraj, 2022-10-13:
who is pulling `nativesdk-clang-glue` I think thats what is pulling it in.

Lorac, 2022-10-13:
`clang-crosssdk-x86_64.do_prepare_recipe_sysroot` is needed to create nativesdk-clang-glue

Lorac, 2022-10-24:
@kraj do you have any updates on this?

kraj, 2023-03-30:
I finally found some time to look into it and I see

```
% readelf -d clang-apply-replacements

Dynamic section at offset 0x1dd08 contains 32 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libclang-cpp.so.16]
 0x0000000000000001 (NEEDED)             Shared library: [libLLVM-16.so]

```

Which means the tools depend on llvm and clang library. Which will be pulled in as dependency and it will automatically pull in `nativesdk-clang-libclang-cpp` and `nativesdk-clang-libllvm`

closed 2023-03-30 by kraj