kraj/meta-clang #686: Build of nativesdk-castxml fails due to CMAKE_AR issue

h5b, 2022-10-31:
**Describe the bug**
Building nativesdk-castxml with poky/kirkstone (4.0.4) fails due to an issue with CMAKE_AR as depicted below:

$ bitbake nativesdk-castxml
`ERROR: nativesdk-castxml-0.4.4-r0 do_configure: ExecutionError('/opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/temp/run.do_configure.714558', 1, None, None)
ERROR: Logfile of failure stored in: /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/temp/log.do_configure.714558
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/opt/yocto/issue_22/../poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/opt/yocto/issue_22/../poky/meta/recipes-devtools/gcc/gcc-crosssdk_11.3.bb:do_populate_sysroot', '/opt/yocto/issue_22/../poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/issue_22/../poky/meta/recipes-devtools/chrpath/chrpath_0.16.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/issue_22/../poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/issue_22/../poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/issue_22/../poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/issue_22/../poky/meta/recipes-devtools/python/python3_3.10.4.bb:do_populate_sysroot', 'virtual:nativesdk:/opt/yocto/issue_22/../meta-clang/recipes-devtools/clang/clang_git.bb:do_populate_sysroot', 'virtual:nativesdk:/opt/yocto/issue_22/../poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', 'virtual:nativesdk:/opt/yocto/issue_22/../poky/meta/recipes-devtools/gcc/gcc-runtime_11.3.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['cmake-native', 'gcc-crosssdk-x86_64-pokysdk-linux', 'quilt-native', 'chrpath-native', 'ninja-native', 'pkgconfig-native', 'pseudo-native', 'python3-native', 'nativesdk-clang', 'nativesdk-glibc', 'nativesdk-gcc-runtime', 'nativesdk-zlib', 'nativesdk-libxml2', 'nativesdk-libedit', 'nativesdk-binutils', 'nativesdk-python3', 'nativesdk-libffi', 'libtool-native', 'zlib-native', 'ncurses-native', 'xz-native', 'zstd-native', 'bzip2-native', 'curl-native', 'nativesdk-linux-libc-headers', 'gmp-native', 'libmpc-native', 'binutils-crosssdk-x86_64-pokysdk-linux', 'mpfr-native', 'gnu-config-native', 'texinfo-dummy-native', 'flex-native', 'gettext-native', 're2c-native', 'openssl-native', 'libnsl2-native', 'gdbm-native', 'libtirpc-native', 'readline-native', 'util-linux-libuuid-native', 'libffi-native', 'sqlite3-native', 'nativesdk-libgcc', 'nativesdk-ncurses', 'nativesdk-opkg-utils', 'nativesdk-elfutils', 'nativesdk-bzip2', 'nativesdk-openssl', 'nativesdk-libnsl2', 'nativesdk-gdbm', 'nativesdk-xz', 'nativesdk-libtirpc', 'nativesdk-util-linux-libuuid', 'nativesdk-sqlite3', 'nativesdk-readline', 'nativesdk-libxcrypt', 'gettext-minimal-native', 'm4-native', 'perl-native', 'nativesdk-libarchive', 'nativesdk-curl', 'nativesdk-libmicrohttpd', 'make-native', 'nativesdk-zstd', 'nativesdk-lzo', 'nativesdk-libgcrypt', 'nativesdk-gnutls', 'nativesdk-file', 'nativesdk-libgpg-error', 'nativesdk-libcap', 'nativesdk-libidn2', 'nativesdk-nettle', 'nativesdk-libunistring', 'nativesdk-gmp']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| CMake Error at /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/toolchain.cmake:11 (find_program):
|   Could not find CMAKE_AR using the following names:
|   x86_64-pokysdk-linux-llvm-ar
| Call Stack (most recent call first):
|   /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:131 (include)
|   CMakeLists.txt:17 (project)
| 
| 
| -- Configuring incomplete, errors occurred!
| WARNING: /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/temp/run.do_configure.714558:167 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-linux/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=bin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-linux/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-linux/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython_EXECUTABLE:PATH=/opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython3_EXECUTABLE:PATH=/opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DFETCHCONTENT_FULLY_DISCONNECTED=ON -Wno-dev'
| WARNING: Backtrace (BB generated script):
| 	#1: cmake_do_configure, /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/temp/run.do_configure.714558, line 167
| 	#2: do_configure, /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/temp/run.do_configure.714558, line 146
| 	#3: main, /opt/yocto/issue_22/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.4-r0/temp/run.do_configure.714558, line 203
ERROR: Task (virtual:nativesdk:/opt/yocto/issue_22/../meta-clang/recipes-devtools/castxml/castxml_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1115 tasks of which 646 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:nativesdk:/opt/yocto/issue_22/../meta-clang/recipes-devtools/castxml/castxml_git.bb:do_configure
Summary: There was 1 ERROR message, returning a non-zero exit code.
`

**To Reproduce**
git clone https://git.yoctoproject.org/git/poky
cd poky && git checkout kirkstone
$ . ./oe-init-build-env
$ git clone git://github.com/kraj/meta-clang.git
// Add poky/meta-clang to conf/bblayers.conf

**Expected behavior**
Running bitbake nativesdk-castxml succeeds

**HOST OS:**
- Fedora release 34 (Thirty Four), x86_64
- all required host tools installed

**local.conf**
MACHINE ?= "genericx86"
SDKMACHINE = "x86_64"

h5b, 2022-10-31:
Please let me know if any information is missing or if i should switch to main/HEAD instead of kirkstone to reproduce the issue.

kraj, 2022-11-01:
yes try to see if you can reproduce it on master as well using poky/master as well.

h5b, 2022-11-02:
I was able to reproduce using poky/master with meta-clang/master.

´´
$ bitbake-layers show-layers
NOTE: Starting bitbake server...
layer                      path                                                 priority
==========================================================================
meta-clang            /opt/yocto/meta-clang                      7
meta                      /opt/yocto/poky/meta                       5
meta-poky             /opt/yocto/poky/meta-poky              5
meta-yocto-bsp     /opt/yocto/poky/meta-yocto-bsp       5
´´´

```
$ bitbake nativesdk-castxml
[..] // cut for brevity
Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "genericx86"
DISTRO               = "poky"
DISTRO_VERSION       = "4.1"
TUNE_FEATURES        = "m32 core2"
TARGET_FPU           = ""
meta-clang           = "HEAD:4e5ab07cae26e31d04590d372a777bb8b2a599d1"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:a501f13c500ffe0f5f6a7a971dfeb56f80f09955"

[...] // cut for brevity
ERROR: nativesdk-castxml-0.4.5-r0 do_configure: ExecutionError('/opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/temp/run.do_configure.2286279', 1, None, None)
ERROR: Logfile of failure stored in: /opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/temp/log.do_configure.2286279
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/opt/yocto/poky/meta/recipes-devtools/cmake/cmake-native_3.24.2.bb:do_populate_sysroot', '/opt/yocto/poky/meta/recipes-devtools/gcc/gcc-crosssdk_12.2.bb:do_populate_sysroot', '/opt/yocto/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/poky/meta/recipes-devtools/chrpath/chrpath_0.16.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/poky/meta/recipes-devtools/ninja/ninja_1.11.1.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/poky/meta/recipes-devtools/python/python3_3.11.0.bb:do_populate_sysroot', 'virtual:nativesdk:/opt/yocto/meta-clang/recipes-devtools/clang/clang_git.bb:do_populate_sysroot', 'virtual:nativesdk:/opt/yocto/poky/meta/recipes-core/glibc/glibc_2.36.bb:do_populate_sysroot', 'virtual:nativesdk:/opt/yocto/poky/meta/recipes-devtools/gcc/gcc-runtime_12.2.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['cmake-native', 'gcc-crosssdk-x86_64-pokysdk-linux', 'quilt-native', 'chrpath-native', 'ninja-native', 'pkgconfig-native', 'pseudo-native', 'python3-native', 'nativesdk-clang', 'nativesdk-glibc', 'nativesdk-gcc-runtime', 'libtool-native', 'xz-native', 'gmp-native', 'libmpc-native', 'texinfo-dummy-native', 'zlib-native', 'binutils-crosssdk-x86_64-pokysdk-linux', 'gettext-native', 'zstd-native', 'gnu-config-native', 'flex-native', 'mpfr-native', 'nativesdk-libffi', 'nativesdk-libedit', 'nativesdk-python3', 'nativesdk-zlib', 'nativesdk-zstd', 'nativesdk-libxml2', 'nativesdk-binutils', 'util-linux-libuuid-native', 'bzip2-native', 'openssl-native', 'sqlite3-native', 'libffi-native', 'readline-native', 'ncurses-native', 'gdbm-native', 'libtirpc-native', 'libnsl2-native', 'curl-native', 'nativesdk-libgcc', 'nativesdk-linux-libc-headers', 're2c-native', 'gettext-minimal-native', 'm4-native', 'nativesdk-ncurses', 'nativesdk-opkg-utils', 'nativesdk-gdbm', 'nativesdk-xz', 'nativesdk-libnsl2', 'nativesdk-readline', 'nativesdk-sqlite3', 'nativesdk-util-linux-libuuid', 'nativesdk-libxcrypt', 'nativesdk-libtirpc', 'nativesdk-openssl', 'nativesdk-bzip2', 'nativesdk-elfutils', 'perl-native', 'nativesdk-libmicrohttpd', 'nativesdk-libarchive', 'nativesdk-curl', 'make-native', 'nativesdk-file', 'nativesdk-gnutls', 'nativesdk-libgcrypt', 'nativesdk-gmp', 'nativesdk-libidn2', 'nativesdk-libunistring', 'nativesdk-nettle', 'nativesdk-libcap', 'nativesdk-libgpg-error']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| CMake Error at /opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/toolchain.cmake:11 (find_program):
|   Could not find CMAKE_AR using the following names:
|   x86_64-pokysdk-linux-llvm-ar
| Call Stack (most recent call first):
|   /opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/recipe-sysroot-native/usr/share/cmake-3.24/Modules/CMakeDetermineSystem.cmake:131 (include)
|   CMakeLists.txt:17 (project)
| 
| 
| -- Configuring incomplete, errors occurred!
| WARNING: /opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/temp/run.do_configure.2286279:167 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-linux/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=bin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-lin [... cut ...]
| 	#3: main, /opt/yocto/clang_issue_686/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-castxml/0.4.5-r0/temp/run.do_configure.2286279, line 203
ERROR: Task (virtual:nativesdk:/opt/yocto/meta-clang/recipes-devtools/castxml/castxml_0.4.5.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1095 tasks of which 637 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:nativesdk:/opt/yocto/meta-clang/recipes-devtools/castxml/castxml_0.4.5.bb:do_configure
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

h5b, 2022-11-02:
> yes try to see if you can reproduce it on master as well using poky/master as well.
:+1:

h5b, 2022-12-15:
@kraj Currently i'm stuck with the problem. Any input what might be left to test is highly appreciated.

kraj, 2023-03-30:
> @kraj Currently i'm stuck with the problem. Any input what might be left to test is highly appreciated.

I have been able to reproduce this locally on master. The issue is that we use gcc for building all nativesdk packages
but TOOLCHAIN setting for castxml is set to clang

closed 2023-03-30 by kraj