kraj/meta-clang #690: Issue while building opensbi with clang

vaibhavvivekdeshpande, 2022-11-14:
Facing issue while building opensbi with clang from kirkstone branch.
Please see below error logs

Log data follows:
| DEBUG: Executing python function externalsrc_compile_prefunc
| NOTE: opensbi: compiling from external source tree /home/vaibhav/clang/build/../source/opensbi
| DEBUG: Python function externalsrc_compile_prefunc finished
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'riscv-common', 'common-linux', 'common-glibc', 'riscv64-linux', 'riscv64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 PLATFORM=generic I=/home/vaibhav/clang/build/tmp-glibc/work/riscv64-oe-linux/opensbi/1.0-r0/image FW_PIC=y FW_OPTIONS=0x0 -C /home/vaibhav/clang/build/../source/opensbi FW_PAYLOAD_PATH=/home/vaibhav/clang/build/tmp-glibc/deploy/images/qemuriscv64/Image-qemuriscv64.bin
| make: Entering directory '/home/vaibhav/clang/source/opensbi'
|  ELF       platform/generic/firmware/payloads/test.elf
| /home/vaibhav/clang/build/tmp-glibc/hosttools/ld.bfd: cannot represent machine `riscv'
| clang-14: error: ld.lld command failed with exit code 1 (use -v to see invocation)
| ERROR: oe_runmake failed
| make: Leaving directory '/home/vaibhav/clang/source/opensbi'
| WARNING: exit code 1 from a shell command.

Thank you

kraj, 2022-11-14:
@vaibhavvivekdeshpande it seems to be not finding cross-linker and then defaulting to what it finds in path which is ld.bfd from hosttools. Looking at the opensbi recipe it uses
```
EXTRA_OEMAKE += "PLATFORM=${RISCV_SBI_PLAT} I=${D} FW_PIC=n CLANG_TARGET= "
```

so perhaps try to match these options and see if they match.

kraj, 2023-03-30:
is this still an issue with latest master