kraj/meta-clang #691: Request: `langdale` release tag/branch.

jainvikas8, 2022-11-21:
**Is your feature request related to a problem? Please describe.**
Since `langdale` release is out there, can we please create one, as relying on `master` would break things in the future or have an overhead of backporting changes to `krikstone`

kraj, 2022-11-23:
@jainvikas8 We defer creating branch as late as possible so we can share work with master. Do you see build issues with master on otherwise langdale release ?

jainvikas8, 2022-11-23:
@kraj , Thanks for the response, we don't see any issues at the moment when using `master`, please let us know if there is a date for `langdale` release. Thanks

shr-project, 2022-12-26:
@kraj I've just noticed one issue when using latest meta-clang/master (d89e06ad94a46f6810d0a8787004b71b8ecaf87d) with latest oe-core/langdale (df9a2b69748d8a24c3390f812225231e9e9acb66)

And it caused target clang to fail in do_compile with:
```
-- Found SWIG: /OE/lge/build/webos/langdale/BUILD/work/qemux86_64-webos-linux/clang/15.0.6-r0/recipe-sysroot-native/usr/bin/swig (found suitable version "4.0.2", minimum required is "3.0")
-- Could NOT find Lua (missing: LUA_LIBRARIES LUA_INCLUDE_DIR) (Required is exact version "5.3")
-- Could NOT find LuaAndSwig (missing: LUA_LIBRARIES LUA_INCLUDE_DIR)
-- Enable Lua scripting support in LLDB: FALSE
-- Could NOT find Python3 (missing: Development Development.Module Development.Embed) (found version "3.11.0")
-- Found PythonAndSwig: /OE/lge/build/webos/langdale/BUILD/work/qemux86_64-webos-linux/clang/15.0.6-r0/recipe-sysroot-native/usr/lib/libpython3.10.so
-- Enable Python scripting support in LLDB: TRUE
-- Found LibXml2: /OE/lge/build/webos/langdale/BUILD/work/qemux86_64-webos-linux/clang/15.0.6-r0/recipe-sysroot/usr/lib/libxml2.so (found suitable version "2.9.14", minimum required is "2.8")
-- Enable Libxml 2 support in LLDB: TRUE
-- Enable libfbsdvmcore support in LLDB: 0
-- Looking for el_winsertstr
-- Looking for el_winsertstr - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of el_rfunc_t
-- Check size of el_rfunc_t - done
-- Performing Test CXX_SUPPORTS_DEPRECATED_DECLARATIONS
-- Performing Test CXX_SUPPORTS_DEPRECATED_DECLARATIONS - Success
-- Performing Test CXX_SUPPORTS_UNKNOWN_PRAGMAS
-- Performing Test CXX_SUPPORTS_UNKNOWN_PRAGMAS - Success
-- Performing Test CXX_SUPPORTS_STRICT_ALIASING
-- Performing Test CXX_SUPPORTS_STRICT_ALIASING - Success
-- Performing Test CXX_SUPPORTS_STRINGOP_TRUNCATION
-- Performing Test CXX_SUPPORTS_STRINGOP_TRUNCATION - Failed
-- Performing Test CXX_SUPPORTS_DEPRECATED_REGISTER
-- Performing Test CXX_SUPPORTS_DEPRECATED_REGISTER - Success
-- Performing Test CXX_SUPPORTS_VLA_EXTENSION
-- Performing Test CXX_SUPPORTS_VLA_EXTENSION - Success
-- LLDB version: 15.0.6
-- Looking for ppoll
-- Looking for ppoll - found
-- Looking for ptsname_r
-- Looking for ptsname_r - found
-- Looking for accept4
-- Looking for accept4 - found
-- Looking for include files sys/types.h, sys/event.h
-- Looking for include files sys/types.h, sys/event.h - not found
-- Looking for process_vm_readv
-- Looking for process_vm_readv - found
-- Looking for __NR_process_vm_readv
-- Looking for __NR_process_vm_readv - found
-- Looking for compression_encode_buffer in compression
-- Looking for compression_encode_buffer in compression - not found
Traceback (most recent call last):
  File "/OE/lge/build/webos/langdale/BUILD/work-shared/llvm-project-source-15.0.6-r0/git/lldb/bindings/python/get-python-config.py", line 70, in <module>
    main()
  File "/OE/lge/build/webos/langdale/BUILD/work-shared/llvm-project-source-15.0.6-r0/git/lldb/bindings/python/get-python-config.py", line 34, in main
    print(relpath_nodots(sysconfig.get_path("platlib"), sys.prefix))
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/sysconfig.py", line 662, in get_path
    return get_paths(scheme, vars, expand)[name]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/sysconfig.py", line 652, in get_paths
    return _expand_vars(scheme, vars)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/sysconfig.py", line 289, in _expand_vars
    _extend_dict(vars, get_config_vars())
                       ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/sysconfig.py", line 706, in get_config_vars
    _init_posix(_CONFIG_VARS)
  File "/usr/lib/python3.11/sysconfig.py", line 567, in _init_posix
    _temp = __import__(name, globals(), locals(), ['build_time_vars'], 0)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named '_sysconfigdata'
[... 41 lines cut ...]
```
Could NOT find Python3 (missing: Development Development.Module Development.Embed) (found version "3.11.0")
```
part is also a bit strange, because python3 and python3native are 3.10.6:
```
clang/15.0.6-r0 $ ./recipe-sysroot-native/usr/bin/python3-native/python3 --v
ersion
Python 3.10.6
```

host (docker container) python is 3.10.8:
```
clang/15.0.6-r0 $ /usr/bin/python3 --version
Python 3.10.8
```

and python3 on my real host is 3.11.1:
```
$ /usr/bin/python3 --version
Python 3.11.1
```

So not sure where it found 3.11.0.

Also not sure if it's worth creating langdale branch just without this python3targetconfig or if we should instead try to figure out something which works with both oe-core versions.

kraj, 2022-12-26:
I think its time to create langdale branch just before this change. OE-Core has several changes in this area which are not in langdale.

kraj, 2022-12-26:
[`langdale` branch](https://github.com/kraj/meta-clang/tree/langdale) has been created.

closed 2022-12-26 by kraj