kraj/meta-clang #693: clang-native fails to compile on poky master after swig update

miwojcik, 2022-11-23:
**Describe the bug**
clang-native fails to compile on poky master:
```
2022-11-23 08:16:56 - INFO     - | [622/5820] cd /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python && /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/recipe-sysroot-native/usr/bin/swig -c++ -features autodoc -I/builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/include -I/builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings -D__STDC_LIMIT_MACROS -D__STDC_CONSTANT_MACROS -I/builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings/python -c++ -shadow -python -py3 -threads -outdir /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python -o /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python/LLDBWrapPython.cpp /builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings/python/python.swig
2022-11-23 08:16:56 - INFO     - | FAILED: tools/lldb/bindings/python/LLDBWrapPython.cpp tools/lldb/bindings/python/lldb.py /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python/LLDBWrapPython.cpp /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python/lldb.py
2022-11-23 08:16:56 - INFO     - | cd /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python && /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/recipe-sysroot-native/usr/bin/swig -c++ -features autodoc -I/builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/include -I/builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings -D__STDC_LIMIT_MACROS -D__STDC_CONSTANT_MACROS -I/builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings/python -c++ -shadow -python -py3 -threads -outdir /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python -o /builder/build/tmp/work/x86_64-linux/clang-native/15.0.4-r0/build/tools/lldb/bindings/python/LLDBWrapPython.cpp /builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings/python/python.swig
2022-11-23 08:16:56 - INFO     - | Deprecated command line option: -py3. Ignored, this option is no longer supported.
2022-11-23 08:16:56 - INFO     - | /builder/build/tmp/work-shared/llvm-project-source-15.0.4-r0/git/lldb/bindings/interfaces.swig:5: Error: Macro '__STDC_LIMIT_MACROS' redefined,
2022-11-23 08:16:56 - INFO     - | :1: Error: previous definition of '__STDC_LIMIT_MACROS'.
```

I bisected the poky repo and build failure appeared first after swig update :
```
deef918ab212a1549ffec25fd8318dec044d1387 is the first bad commit
commit deef918ab212a1549ffec25fd8318dec044d1387
Author: Alexander Kanavin <alex.kanavin@gmail.com>
Date:   Sun Nov 20 14:31:08 2022 +0100

    swig: update 4.0.2 -> 4.1.0

    swig has transitioned pcre -> pcre2
    Update 0001-configure-use-pkg-config-for-pcre-detection.patch
    accordingly.

    (From OE-Core rev: 53e51ded320fb35fb53d496e832b6527372f58ed)

    Signed-off-by: Alexander Kanavin <alex@linutronix.de>
    Signed-off-by: Alexandre Belloni <alexandre.belloni@bootlin.com>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>

 meta/recipes-devtools/swig/swig.inc                |  2 +-
 ...nfigure-use-pkg-config-for-pcre-detection.patch | 44 ++++++++++++----------
 meta/recipes-devtools/swig/swig_4.0.2.bb           |  7 ----
 meta/recipes-devtools/swig/swig_4.1.0.bb           |  7 ++++
 4 files changed, 32 insertions(+), 28 deletions(-)
 delete mode 100644 meta/recipes-devtools/swig/swig_4.0.2.bb
 create mode 100644 meta/recipes-devtools/swig/swig_4.1.0.bb
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build Configuration:
```
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "qemuarm64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.1"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa57"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "HEAD:deef918ab212a1549ffec25fd8318dec044d1387"
meta-oe              = "master:c4829fa338dae008c75cfffd4a712813a99c33c4"
meta-clang           = "master:f71d12dc480550ebf88d7bc47a34d7b20ee0707f"
meta-arm-toolchain
meta-arm             = "master:9699e6698e1e4de2b4b76c076a8ffc585ebab100"
meta-python          = "master:c4829fa338dae008c75cfffd4a712813a99c33c4"
```
2. Run `bitbake clang-native`

**Desktop (please complete the following information):**
 - Ubuntu 22.04.1 LTS

kraj, 2022-11-23:
@miwojcik yes I am seeing it too.

kraj, 2022-11-23:
fixed with https://github.com/kraj/meta-clang/pull/633

miwojcik, 2022-11-23:
Thank you for a quick fix. Works like a charm 👍