kraj/meta-clang #696: xserver crashes in pixman since change to assembler

rossburton, 2022-11-28:
With oe-core langdale or master and meta-clang master, xserver crashes on startup:

```
#0  0x0000007ff7f5d6cc in pixman_fill32 (bits=0x7ff660f010, stride=1280, x=0, y=0, width=1280, height=<optimized out>, filler=0)
    at ../pixman-0.40.0/pixman/pixman-fast-path.c:2119
#1  fast_path_fill (imp=<optimized out>, bits=0x7ff660f010, stride=1280, bpp=32, x=0, y=0, width=1280, height=<optimized out>, filler=0)
    at ../pixman-0.40.0/pixman/pixman-fast-path.c:2152
#2  0x0000007ff7f74f88 in _pixman_implementation_fill (imp=0x55557d3ac0, bits=0x7ff660f010, stride=1280, bpp=32, x=0, y=0, width=1280,
    height=800, filler=0) at ../pixman-0.40.0/pixman/pixman-implementation.c:277
#3  0x0000007ff7f28f4c in pixman_fill (bits=0x7ff660f010, stride=1280, bpp=32, x=x@entry=0, y=y@entry=0, width=width@entry=1280, height=800,
    filler=<optimized out>) at ../pixman-0.40.0/pixman/pixman.c:759
#4  0x00000055556fca50 in fbFill (pDrawable=<optimized out>, pGC=0x55557fcee0, x=x@entry=0, y=y@entry=0, width=1280, height=800)
    at ../xorg-server-21.1.4/fb/fbfill.c:125
#5  0x00000055556fd324 in fbPolyFillRect (pDrawable=0x555583a3b0, pGC=0x55557fcee0, nrect=<optimized out>, prect=<optimized out>)
    at ../xorg-server-21.1.4/fb/fbfillrect.c:72
#6  0x0000005555615920 in damagePolyFillRect (pDrawable=0x555583a3b0, pGC=0x55557fcee0, nRects=1, pRects=0x5555865010)
    at ../xorg-server-21.1.4/miext/damage/damage.c:1224
#7  0x000000555559daec in miPaintWindow (pWin=<optimized out>, prgn=<optimized out>, what=<optimized out>)
    at ../xorg-server-21.1.4/mi/miexpose.c:536
#8  0x000000555559d6d0 in miWindowExposures (pWin=0x555583a3b0, prgn=0x7ffffffaa0) at ../xorg-server-21.1.4/mi/miexpose.c:387
#9  0x00000055555f23c8 in MapWindow (pWin=0x555583a3b0, client=0x55557de3f0) at ../xorg-server-21.1.4/dix/window.c:2711
#10 0x00000055555ca1e0 in dix_main (argc=1, argv=0x7ffffffcf8, envp=<optimized out>) at ../xorg-server-21.1.4/dix/main.c:243
#11 0x0000007ff76c77b8 in __libc_start_call_main (main=main@entry=0x5555707ca0 <main>, argc=argc@entry=1, argv=argv@entry=0x7ffffffcf8)
    at /usr/src/debug/glibc/2.36-r0/sysdeps/nptl/libc_start_call_main.h:58
#12 0x0000007ff76c789c in __libc_start_main_impl (main=0x5555707ca0 <main>, argc=1, argv=0x7ffffffcf8, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=<optimized out>) at /usr/src/debug/glibc/2.36-r0/csu/libc-start.c:381
#13 0x0000005555591ef0 in _start () at ../sysdeps/aarch64/start.S:81
```

Bisecting confirmed that it worked until 84dbafa42d8141b00da75d6664aef07c252a52ee.

kraj, 2022-12-06:
@rossburton can you try replacing disabling it will gcc and see if that helps ?
```
TOOLCHAIN:pn-pixman:aarch64 = "gcc"
```

rossburton, 2023-01-04:
We have the same build (eg qemuarm64) for clang and gcc: gcc works, clang doesn't.

kraj, 2023-01-04:
yeah I am seeing it all over meta-clang and Yoe distribution CI :( for some time now. I have been turning a blind eye due to lack of time.

rossburton, 2023-01-04:
Congratulations, this is now failing on the yocto autobuilder: https://autobuilder.yoctoproject.org/typhoon/#/builders/142/builds/3

:)

I presume that nobody is actually using langdale+arm64+clang+pixman?

I suggest removing the integrated-as workaround and just set `TOOLCHAIN="gcc"` as an interim fix as this is very broken right now.

kraj, 2023-01-04:
> Congratulations, this is now failing on the yocto autobuilder: https://autobuilder.yoctoproject.org/typhoon/#/builders/142/builds/3
>
> :)
>
> I presume that nobody is actually using [...]

Does this fix work ?

rossburton, 2023-01-04:
Yes, it appears to be a bad interaction with clang and pixman.

kraj, 2023-01-04:
OK.

kraj, 2023-03-30:
We have switched back to using gcc to build pixman for now.

closed 2023-03-30 by kraj