kraj/meta-clang #701: Is this an ok way to replace llvm recipe from oe-core?

bkauler, 2022-12-08:
I'm doing a kirkstone build, have oe-core kirkstone branch, etc.
Also have meta-clang kirkstone branch.

Ran into a conflict, with two providers of llvm, "bitbake -g core-image-minimal" gave an error message. 
They are also different versions, llvm in oe-core is 13.0.1, in meta-clang is 14.0.3

I have applied a "fix", that is probably not really the best solution. Doing a build right now, to see if it works, but could you guys tell me in principle whether this is a sane thing to do...

I deleted meta/recipes-devtools/llvm, so it is completely out of the picture.

I then created recipes-devtools/clang/clang_%.bbappend, to rprovide everything that llvm did:

```
RPROVIDES:clang:append = " llvm llvm-locale llvm-libllvm llvm-liboptremarks \
    llvm-staticdev llvm-dev llvm-src llvm-dbg llvm-bugpointpasses \
    llvm-liblto llvm-doc llvm-llvmhello"
```

My computer isn't the fastest horse out of the gate, and have some big packages in the build, so it is going to take over 24 hours to complete, but can you tell me, is there anything in particular that is wrong with what I have done?

kraj, 2022-12-09:
usually yes, however, the focus of llvm library from meta-clang is for building the compiler, so if your usecase is something more, it might need some tweaking. 

We should match the providers between two recipes. Which one is missing it ?

bkauler, 2022-12-11:
The build complete successfully. It took a few days, as I am updating from dunfell, a couple of hundred recipes, lots of fixing required.

I will have to get back to you later on what was causing the clash before.

kraj, 2023-03-30:
if it worked then report back and we can close this issue.

bkauler, 2023-03-31:
Yes, no problems, closing.

closed 2023-03-31 by bkauler