kraj/meta-clang #707: perf fails to build in kirkstone branch

rossburton, 2022-12-16:
```
arm-poky-linux-gnueabi-clang -target arm-poky-linux-gnueabi  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a15 -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9=/usr/src/debug/perf/1.0-r9                      -fdebug-prefix-map=.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9=/usr/src/debug/perf/1.0-r9                      -fdebug-prefix-map=.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/recipe-sysroot=                      -fdebug-prefix-map=.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/recipe-sysroot-native=  --sysroot=.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/recipe-sysroot -Wp,-MD,.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/perf-1.0/fs/.fs.o.d -Wp,-MT,.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/perf-1.0/fs/fs.o -Wbad-function-cast -Wdeclaration-after-statement -Wformat-security -Wformat-y2k -Winit-self -Wmissing-declarations -Wmissing-prototypes -Wnested-externs -Wno-system-headers -Wold-style-definition -Wpacked -Wredundant-decls -Wstrict-prototypes -Wswitch-default -Wswitch-enum -Wundef -Wwrite-strings -Wformat -Wno-type-limits -Wshadow -ldw -ggdb3 -Wall -Wextra -std=gnu99 -U_FORTIFY_SOURCE -fPIC -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -I.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/lib/api -I.../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/include -D"BUILD_STR(s)=#s" -c -o .../tmp/work/qemuarm-poky-linux-gnueabi/perf/1.0-r9/perf-1.0/fs/fs.o fs/fs.c

clang-14: error: no such file or directory: 'arm-poky-linux-gnueabi'
```

kraj, 2022-12-17:
we need to backport https://lore.kernel.org/linux-perf-users/Y5d4k7fDxfRP7hcN@kernel.org/T/#t to linux-yocto kernels on kirkstone get it going on kirkstone.  Perhaps work with @zeddi on this

rossburton, 2023-01-03:
Fails in langdale too now.

rossburton, 2023-01-04:
I can replicate this with master too:
https://autobuilder.yoctoproject.org/typhoon/#/builders/142/builds/4/steps/11/logs/stdio

kraj, 2023-01-04:
see the patches on `yoe/mut` branch on poky-contrib, Bruce has posted a v1 of kernel patches with these  but it did not go through so until a v2 is posted you can use following patch from my branch 

perf: Fix build with clang

rossburton, 2023-01-11:
Fixed in master.

closed 2023-01-11 by rossburton