kraj/meta-clang #722: Fails yocto-check-layer

rossburton, 2023-01-11:
In a surprise to no one, meta-clang fails the layer compat:

```
AssertionError: Adding layer meta-clang changed signatures.
769 signatures changed, initial differences (first hash before, second after):
   mesa-native:do_deploy_source_date_epoch: bf0ce6f1491be8522fd2c5cf271b231601d89406b8570e6f411b79fa530b7ce5 -> c4a2c6a375cf9fd2257f4c22e2d9cd28d391b7326050a3e183979b3f5c4c6e56
      bitbake-diffsigs --task mesa-native do_deploy_source_date_epoch --signature bf0ce6f1491be8522fd2c5cf271b231601d89406b8570e6f411b79fa530b7ce5 c4a2c6a375cf9fd2257f4c22e2d9cd28d391b7326050a3e183979b3f5c4c6e56
      Task dependencies changed from:
      ['AR', 'AS', 'BUILD_AR', 'BUILD_AS', 'BUILD_AS_ARCH', 'BUILD_CC', 'BUILD_CCLD', 'BUILD_CC_ARCH', 'BUILD_CFLAGS', 'BUILD_CPP', 'BUILD_CPPFLAGS', 'BUILD_CXX', 'BUILD_CXXFLAGS', 'BUILD_FC', 'BUILD_LD', 'BUILD_LDFLAGS', 'BUILD_LD_ARCH', 'BUILD_NM', 'BUILD_OPTIMIZATION', 'BUILD_OS', 'BUILD_PREFIX', 'BUILD_RANLIB', 'BUILD_STRIP', 'BUILD_VENDOR', 'CC', 'CCLD', 'CFLAGS', 'CONFIG_SITE', 'CPP', 'CPPFLAGS', 'CXX', 'CXXFLAGS', 'DEBUG_BUILD', 'EXTENDPE', 'EXTRA_NATIVE_PKGCONFIG_PATH', 'FC', 'GETTEXTDATADIRS', 'HOST_PREFIX', 'LC_ALL', 'LD', 'LDFLAGS', 'MAKE', 'MULTIMACH_TARGET_SYS', 'NATIVE_PACKAGE_PATH_SUFFIX', 'NM', 'OBJCOPY', 'OBJDUMP', 'PACKAGE_ARCH', 'PE', 'PERL_HASH_SEED', 'PKG_CONFIG_DIR', 'PKG_CONFIG_DISABLE_UNINSTALLED', 'PKG_CONFIG_LIBDIR', 'PKG_CONFIG_PATH', 'PKG_CONFIG_SYSROOT_DIR', 'PN', 'PR', 'PSEUDO_DISABLED', 'PV', 'PYTHON', 'PYTHONHASHSEED', 'PYTHONNOUSERSITE', 'PYTHON_ABI', 'PYTHON_BASEVERSION', 'PYTHON_DIR', 'PYTHON_INCLUDE_DIR', 'PYTHON_LIBRARY', 'RANLIB', 'READELF', 'RECIPE_SYSROOT_NATIVE', 'SDE_DEPLOYDIR', 'SDE_DIR', 'SDE_FILE', 'SSTATECREATEFUNCS', 'SSTATEPOSTCREATEFUNCS', 'SSTATEPOSTINSTFUNCS', 'SSTATEPOSTUNPACKFUNCS', 'SSTATE_BUILDDIR', 'SSTATE_CURRTASK', 'SSTATE_EXTRAPATH', 'SSTATE_INSTDIR', 'SSTATE_MANIFESTS', 'SSTATE_MANMACH', 'SSTATE_PKG', 'SSTATE_PKGNAME', 'SSTATE_PKGSPEC', 'SSTATE_SCAN_FILES', 'SSTATE_SKIP_CREATION', 'SSTATE_VERSION', 'STAGING_BASE_LIBDIR_NATIVE', 'STAGING_BINDIR_NATIVE', 'STAGING_DATADIR', 'STAGING_DATADIR_NATIVE', 'STAGING_DIR_NATIVE', 'STAGING_ETCDIR_NATIVE', 'STAGING_INCDIR', 'STAGING_INCDIR_NATIVE', 'STAGING_LIBDIR', 'STAGING_LIBDIR_NATIVE', 'STAGING_SBINDIR_NATIVE', 'STAMP', 'STRINGS', 'STRIP', 'TARGET_OS', 'TARGET_VENDOR', 'TZ', 'base_bindir', 'base_libdir', 'base_libdir_native', 'base_prefix', 'base_sbindir', 'bindir', 'bindir_native', 'datadir', 'datadir_native', 'do_deploy_source_date_epoch[network]', 'do_deploy_source_date_epoch[sstate-plaindirs]', 'docdir', 'exec_prefix', 'generate_sstatefn', 'includedir', 'includedir_native', 'infodir', 'libdir', 'libdir_native', 'libexecdir', 'localstatedir', 'lt_cv_sys_lib_dlsearch_path_spec', 'mandir', 'nonarch_base_libdir', 'nonarch_libdir', 'oe.path.copyhardlinktree', 'oe.path.copytree', 'oe.path.remove', 'oe.sstatesig.sstate_get_manifest_filename', 'oe.types.boolean', 'oe.utils.vartrue', 'oldincludedir', 'prefix', 'prefix_native', 'root_prefix', 'sbindir', 'sbindir_native', 'servicedir', 'sharedstatedir', 'sstate_add', 'sstate_clean', 'sstate_clean_manifest', 'sstate_init', 'sstate_install', 'sstate_installpkgdir', 'sstate_package', 'sstate_state_fromvars', 'sstate_task_postfunc', 'sstate_task_prefunc', 'sysconfdir', 'sysconfdir_native', 'systemd_system_unitdir', 'systemd_unitdir', 'systemd_user_unitdir']
      to:
      ['AR', 'AS', 'BUILD_AR', 'BUILD_AS', 'BUILD_AS_ARCH', 'BUILD_CC', 'BUILD_CCLD', 'BUILD_CC_ARCH', 'BUILD_CFLAGS', 'BUILD_CPP', 'BUILD_CPPFLAGS', 'BUILD_CXX', 'BUILD_CXXFLAGS', 'BUILD_FC', 'BUILD_LD', 'BUILD_LDFLAGS', 'BUILD_LD_ARCH', 'BUILD_NM', 'BUILD_OPTIMIZATION', 'BUILD_OS', 'BUILD_PREFIX', 'BUILD_RANLIB', 'BUILD_STRIP', 'BUILD_VENDOR', 'CC', 'CCLD', 'CFLAGS', 'CONFIG_SITE', 'CPP', 'CPPFLAGS', 'CXX', 'CXXFLAGS', 'DEBUG_BUILD', 'EXTENDPE', 'EXTRA_NATIVE_PKGCONFIG_PATH', 'FC', 'GETTEXTDATADIRS', 'HOST_PREFIX', 'LC_ALL', 'LD', 'LDFLAGS', 'MAKE', 'MULTIMACH_TARGET_SYS', 'NATIVE_PACKAGE_PATH_SUFFIX', 'NM', 'OBJCOPY', 'OBJDUMP', 'PACKAGE_ARCH', 'PE', 'PERL_HASH_SEED', 'PKG_CONFIG_DIR', 'PKG_CONFIG_DISABLE_UNINSTALLED', 'PKG_CONFIG_LIBDIR', 'PKG_CONFIG_PATH', 'PKG_CONFIG_SYSROOT_DIR', 'PN', 'PR', 'PSEUDO_DISABLED', 'PV', 'PYTHON', 'PYTHONHASHSEED', 'PYTHONNOUSERSITE', 'PYTHON_ABI', 'PYTHON_BASEVERSION', 'PYTHON_DIR', 'PYTHON_INCLUDE_DIR', 'PYTHON_LIBRARY', 'RANLIB', 'READELF', 'RECIPE_SYSROOT_NATIVE', 'SDE_DEPLOYDIR', 'SDE_DIR', 'SDE_FILE', 'SSTATECREATEFUNCS', 'SSTATEPOSTCREATEFUNCS', 'SSTATEPOSTINSTFUNCS', 'SSTATEPOSTUNPACKFUNCS', 'SSTATE_BUILDDIR', 'SSTATE_CURRTASK', 'SSTATE_EXTRAPATH', 'SSTATE_INSTDIR', 'SSTATE_MANIFESTS', 'SSTATE_MANMACH', 'SSTATE_PKG', 'SSTATE_PKGNAME', 'SSTATE_PKGSPEC', 'SSTATE_SCAN_FILES', 'SSTATE_SKIP_CREATION', 'SSTATE_VERSION', 'STAGING_BASE_LIBDIR_NATIVE', 'STAGING_BINDIR_NATIVE', 'STAGING_DATADIR', 'STAGING_DATADIR_NATIVE', 'STAGING_DIR_NATIVE', 'STAGING_ETCDIR_NATIVE', 'STAGING_INCDIR', 'STAGING_INCDIR_NATIVE', 'STAGING_LIBDIR', 'STAGING_LIBDIR_NATIVE', 'STAGING_SBINDIR_NATIVE', 'STAMP', 'STRINGS', 'STRIP', 'TARGET_OS', 'TARGET_VENDOR', 'TZ', 'YOCTO_ALTERNATE_EXE_PATH', 'base_bindir', 'base_libdir', 'base_libdir_native', 'base_prefix', 'base_sbindir', 'bindir', 'bindir_native', 'datadir', 'datadir_native', 'do_deploy_source_date_epoch[network]', 'do_deploy_source_date_epoch[sstate-plaindirs]', 'docdir', 'exec_prefix', 'generate_sstatefn', 'includedir', 'includedir_native', 'infodir', 'libdir', 'libdir_native', 'libexecdir', 'localstatedir', 'lt_cv_sys_lib_dlsearch_path_spec', 'mandir', 'nonarch_base_libdir', 'nonarch_libdir', 'oe.path.copyhardlinktree', 'oe.path.copytree', 'oe.path.remove', 'oe.sstatesig.sstate_get_manifest_filename', 'oe.types.boolean', 'oe.utils.vartrue', 'oldincludedir', 'prefix', 'prefix_native', 'root_prefix', 'sbindir', 'sbindir_native', 'servicedir', 'sharedstatedir', 'sstate_add', 'sstate_clean', 'sstate_clean_manifest', 'sstate_init', 'sstate_install', 'sstate_installpkgdir', 'sstate_package', 'sstate_state_fromvars', 'sstate_task_postfunc', 'sstate_task_prefunc', 'sysconfdir', 'sysconfdir_native', 'systemd_system_unitdir', 'systemd_unitdir', 'systemd_user_unitdir']
      basehash changed from abde446c078f466a7386adf1d64b163015079cfd3343fb128fb9ebcdfb0334e3 to 06f361e17bbb383b8b1380cada26bfc2c83323d0a0f75cb545f1801a71fb667c
      Dependency on variable YOCTO_ALTERNATE_EXE_PATH was added

   packagegroup-core-standalone-sdk-target:do_package: b7bf9ae99ea656d0386004f8bfbdcdaba2f9e91e2fbb02477f60f56b3b91ca10 -> 727253b0729721da991b65c47f03c3aec8f6da29e73f819bd745039bc60b0b37
      bitbake-diffsigs --task packagegroup-core-standalone-sdk-target do_package --signature b7bf9ae99ea656d0386004f8bfbdcdaba2f9e91e2fbb02477f60f56b3b91ca10 727253b0729721da991b65c47f03c3aec8f6da29e73f819bd745039bc60b0b37
      basehash changed from 8672fe90be752980bbf3b2e53fee6e0274563985f91939240dbe7e3ef327810a to afaf877af7652077183958d9696dc8409e91ebf82f92ba8e4a03832b041073a3
      Variable RRECOMMENDS:packagegroup-core-standalone-sdk-target value changed from '' to ' ${@bb.utils.contains('CLANGSDK', '1', 'libcxx-dev libcxx-staticdev compiler-rt-dev compiler-rt-staticdev', '', d)}
      CLANGSDK{1} = Unset'
```

rossburton, 2023-01-11:
The former is due to `mesa_%.bbappend`:

```
recipes-graphics/mesa/mesa_%.bbappend:export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config"
```

The latter is `packagegroup-cross-canadian.bbappend`:

```
recipes-core/packagegroups/packagegroup-cross-canadian.bbappend:RDEPENDS:${PN} += "${@bb.utils.contains('CLANGSDK', '1', '${CLANGCROSSCANADIANDEPS}', '', d)}"
```

rossburton, 2023-01-11:
Oh commenting out those lines gave more:
```
   mesa-native:do_configure: 82ebb144cda2c7cd4383f539e4626943023b6a4e6c594cf5922ab7e7be228158 -> d356c01d0e46b9e5a3ff8a605498f5019f5aac509f400425383b092f0d418630      Variable PACKAGECONFIG_CONFARGS value changed:
      " -Ddri3=enabled -Degl=enabled -Dgallium-drivers=${@strip_comma('${GALLIUMDRIVERS}')} [- -Dllvm=disabled-] {+-Dllvm=false+} -Dgbm=enabled -Dgles1=enabled -Dgles2=enabled  -Dglvnd=false -Dlmsensors=disabled -Dgallium-opencl=disabled -Dopencl-spirv=false -Dopengl=true -Dosmesa=false -Dperfetto=false -Dlibunwind=disabled -Dgallium-va=disabled -Dgallium-vdpau=disabled  -Dvulkan-drivers='' -Dvulkan-beta=false  -Dgallium-xa=disabled -Dgallium-xvmc=disabled"
```

```
   packagegroup-core-standalone-sdk-target:do_package: b7bf9ae99ea656d0386004f8bfbdcdaba2f9e91e2fbb02477f60f56b3b91ca10 -> 727253b0729721da991b65c47f03c3aec8f6da29e73f819bd745039bc60b0b37
      Variable RRECOMMENDS:packagegroup-core-standalone-sdk-target value changed from '' to ' ${@bb.utils.contains('CLANGSDK', '1', 'libcxx-dev libcxx-staticdev compiler-rt-dev compiler-rt-staticdev', '', d)}
      CLANGSDK{1} = Unset'
```

rossburton, 2023-01-11:
Finally, this patch simply commenting out lines obviously breaks the build but does pass layer compat, so that's the scope of what needs to be tweaked or merged into oe-core:

```diff
diff --git a/recipes-core/packagegroups/packagegroup-core-standalone-sdk-target.bbappend b/recipes-core/packagegroups/packagegroup-core-standalone-sdk-target.bbappend
index 29990f0..2891c03 100644
--- a/recipes-core/packagegroups/packagegroup-core-standalone-sdk-target.bbappend
+++ b/recipes-core/packagegroups/packagegroup-core-standalone-sdk-target.bbappend
@@ -1 +1 @@
-RRECOMMENDS:${PN} += "${@bb.utils.contains('CLANGSDK', '1', 'libcxx-dev libcxx-staticdev compiler-rt-dev compiler-rt-staticdev', '', d)}"
+#RRECOMMENDS:${PN} += "${@bb.utils.contains('CLANGSDK', '1', 'libcxx-dev libcxx-staticdev compiler-rt-dev compiler-rt-staticdev', '', d)}"
diff --git a/recipes-core/packagegroups/packagegroup-cross-canadian.bbappend b/recipes-core/packagegroups/packagegroup-cross-canadian.bbappend
index 3199a69..aea7001 100644
--- a/recipes-core/packagegroups/packagegroup-cross-canadian.bbappend
+++ b/recipes-core/packagegroups/packagegroup-cross-canadian.bbappend
@@ -1,3 +1,3 @@
 CLANGCROSSCANADIAN = "clang-cross-canadian-${TRANSLATED_TARGET_ARCH}"
 CLANGCROSSCANADIANDEPS += "${@all_multilib_tune_values(d, 'CLANGCROSSCANADIAN')}"
-RDEPENDS:${PN} += "${@bb.utils.contains('CLANGSDK', '1', '${CLANGCROSSCANADIANDEPS}', '', d)}"
+#RDEPENDS:${PN} += "${@bb.utils.contains('CLANGSDK', '1', '${CLANGCROSSCANADIANDEPS}', '', d)}"
diff --git a/recipes-graphics/mesa/mesa_%.bbappend b/recipes-graphics/mesa/mesa_%.bbappend
index 74888ba..269bf6e 100644
--- a/recipes-graphics/mesa/mesa_%.bbappend
+++ b/recipes-graphics/mesa/mesa_%.bbappend
@@ -6,6 +6,6 @@ DEPENDS:append:toolchain-clang = " libatomic-ops"
 EXTRA_OEMASON:append:toolchain-clang:x86 = " -Dasm=false"
 EXTRA_OEMASON:append:toolchain-clang:x86-64 = " -Dasm=false"

-export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config"
+#export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config"

-PACKAGECONFIG[gallium-llvm] = "-Dllvm=true -Dshared-llvm=true -Ddraw-use-llvm=true,-Dllvm=false,clang clang-native elfutils"
+#PACKAGECONFIG[gallium-llvm] = "-Dllvm=true -Dshared-llvm=true -Ddraw-use-llvm=true,-Dllvm=false,clang clang-native elfutils"
```

rossburton, 2023-03-08:
There's also a slew of missing Upstream-Status values:
```
AssertionError: 51 != 0 : Found following patches with malformed or missing upstream status:
/home/ross/Yocto/meta-clang/recipes-devtools/gdb/gdb/0001-gdb-Link-with-latomic-for-riscv-clang-alone.patch
/home/ross/Yocto/meta-clang/recipes-devtools/clang/clang/0010-clang-Define-releative-gcc-installation-dir.patch
...
```

kraj, 2023-03-30:
https://github.com/kraj/meta-clang/pull/772 fixes some everything except one issue which is due to clang projecting itself to provide llvm. I dont know if we should even solve that.