kraj/meta-clang #732: libcxx-14.0.3 do_compile: ExecutionError

p870613, 2023-01-31:
Hi, 
I compile libcxx, and it meet some trouble

```
ERROR: libcxx-14.0.3-r0 do_compile: ExecutionError('/mfs/mtkslt1261/mtk30464/m/build/tmp/work/aarch64-poky-linux/libcxx/14.0.3-r0/temp/run.do_compile.2229', 1, None, None)
ERROR: Logfile of failure stored in: /mfs/mtkslt1261/mtk30464/m/build/tmp/work/aarch64-poky-linux/libcxx/14.0.3-r0/temp/log.do_compile.2229
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /mfs/mtkslt1261/mtk30464/m/build/tmp/work/aarch64-poky-linux/libcxx/14.0.3-r0/build --target cxxabi cxx --
| Error: could not load cache
| WARNING: exit code 1 from a shell command.
NOTE: recipe libcxx-14.0.3-r0: task do_compile: Failed
ERROR: Task (/mfs/mtkslt1261/mtk30464/m/meta/poky/../meta-clang/recipes-devtools/clang/libcxx_git.bb:do_compile) failed with exit code '1'
```

environment
```
OS: ubuntu 18.04
cmake version 3.10.2
```

branch: kirtstone

thanks !!

kraj, 2023-01-31:
```
| Error: could not load cache

```

do you have some sort of caching enabled ? like ccache or some such

p870613, 2023-01-31:
Thanks for your reply !!

how to enable cache ? 

thanks !!

kraj, 2023-03-30:
are you building it on some sort of NFS share ? its something related to your build environment, it works fine for me on ubuntu 18.04 VM here.

closed 2024-05-04 by p870613