kraj/meta-clang #736: Environment setup script of yocto SDK points to wrong compiler

rungemar, 2023-02-03:
**Description** 
Variables CLANGCC, CLANGCXX, CLANGCPP, etc..  are prefixed with TARGET_PREFIX but clang itself isn't

Different than GCC, clang supports all target architectures with one single binary. So no need for a target tuple prefix.

**To Reproduce**
1) include CLANGSDK ="1" in local.conf as described in the README.md meta-clang
2) bitbake core-image-minimal -c populate_sdk
3) in the SDK, there is a script "environment-setup-\<machine\>-\<distro\>-linux"
clang in the SDK is located in:  sysroots/x86_64-pokysdk-linux/usr/bin/clang
variable CLANGCC points to "clang-\<target-tuple\> \<options\>"

**Expected behavior**
CLANGCC should point to "clang  \<options\>"

kraj, 2023-02-03:
Its a cross compiler SDK, `CC` and its ilk is pointing to cross compilers and necessary cmdline options to use them. `CLANGCC` and ilk are same variables to use clang conveniently, where use can set
```
CC=${CLANGCC}
```
in the environment and painlessly switch to using clang without changing the component Makefiles etc.

rungemar, 2023-02-03:
Yes, I understang the idea of that variable ;-)  On my target, it is set like follows:

`export CLANGCC="aarch64-poky-linux-clang --target=aarch64-poky-linux  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc ...`

but there is no "aarch64-poky-linux-clang" in the SDK generated via meta-clang. Instead the clang binary is just named "clang" (located in <SDKPATH>/sysroots/x86_64-pokysdk-linux/usr/bin) and the target architecture is passed to clang via the "--target=aarch-poky-linux -mcpu ... --march ...".

It should be:

`export CLANGCC="clang --target=aarch64-poky-linux  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc ...`

In a x86-64 cross SDK, I see also a "x86_64-poky-linux-clang" in the target sysroot. But I can not execute this binary on the build host, of cause, as the binary is build against the target sysroot. The right clang binary for cross compiling wound be the one in sysroot for the buildhost.

for "CC=${CLANGCC}" to work, CLANGCC must point to the binary in .../sysroots/x86_64-pokysdk-linux/usr/bin

kraj, 2023-02-03:
Do you have 

```
CLANGSDK = "1"
```

set in your distro config ?

rungemar, 2023-02-06:
I have it in local.conf, not in disto.conf

kraj, 2023-02-07:
> I have it in local.conf, not in disto.conf

using `local.conf` to specify it is OK,  `but there is no "aarch64-poky-linux-clang" in the SDK generated via meta-clang.`
this seems to be an unexpected behavior. So I would like to understand this a bit more. 
Do you also set `TOOLCHAIN = "clang"` in `local.conf`

These symlinks should be added via `recipes-core/packagegroups/packagegroup-cross-canadian.bbappend`
which does this

```
CLANGCROSSCANADIAN = "clang-cross-canadian-${TRANSLATED_TARGET_ARCH}"
CLANGCROSSCANADIANDEPS += "${@all_multilib_tune_values(d, 'CLANGCROSSCANADIAN')}"
RDEPENDS:${PN} += "${@bb.utils.contains('CLANGSDK', '1', '${CLANGCROSSCANADIANDEPS}', '', d)}"
```

rungemar, 2023-02-07:
No, I don't have `TOOLCHAIN = "clang"` set. I use gcc to build the yocto image (that would work without meta-clang), but I want clang to be included in the Standard-SDK additionally so I can offer clang as alternative compiler to the application developers.

I see that the extensible SDK has a different directory structure, but I refer to the Standard SDK. It has two sysroots, the "target sysroot" and the "host sysroot" (illustrated here in the yocto manual: https://docs.yoctoproject.org/sdk-manual/appendix-obtain.html#installed-standard-sdk-directory-structure)

I don't expect the compiler to be available in the target sysroot as I can't execute those binaries on the build host.

```
|── environment-setup-aarch64-poky-linux
├── site-config-aarch64-poky-linux
├── sysroots
│   ├── aarch64-poky-linux
│   │   ├── bin                   # ARM binaries, can't execute on build host
│   │   ├── lib                   # ARM libraries for target, clang links against these when cross compiling
|   |  ...
│   └── x86_64-pokysdk-linux      # host sysroot, tools run on build host
│       ├── bin
│       ├── usr
|       │   └── bin
|       │         └── clang       #  <- I would expect CLANGCC to point here
|      ...   
```

in `recipes-core/meta/clang-environment.inc` the variable CLANGCC is appended to the environment setup script:

```
create_sdk_files:append() {
        script=${SDK_OUTPUT}/${SDKPATH}/environment-setup-${REAL_MULTIMACH_TARGET_SYS}
        if ${@bb.utils.contains('CLANGSDK', '1', 'true', 'false', d)}; then
            echo 'export CLANGCC="${TARGET_PREFIX}clang --target=${TARGET_SYS} ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
            echo 'export CLANGCXX="${TARGET_PREFIX}clang++ --target=${TARGET_SYS} ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
            echo 'export CLANGCPP="${TARGET_PREFIX}clang -E --target=${TARGET_SYS} ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
            echo 'export CLANG_TIDY_EXE="${TARGET_PREFIX}clang-tidy"' >> $script
        fi
}
```

I don't understand, why ${TARGET_PREFIX} is prepended to clang. I find it as "clang" in the build host sysroot.
Thank you very much for your support, I appreciate it very much :-)

kraj, 2023-02-07:
Can you search for `aarch64-poky-linux-clang` in your SDK install ? Does it exist say under `sysroots/x86_64-pokysdk-linux`

rungemar, 2023-02-07:
I searched  "*clang". nothing found, just ../sysroots/x86_64-pokysdk-linux/usr/bin/clang   (and all the other clang tools). Do you have a <target-tuple>-clang in that folder? Maybe I can try to reproduce with a very minimalistic setup, just poky and meta-clang?

rungemar, 2023-03-20:
I did some further research and found the reason why the symlinks are missing in our setup:

we had a CLANGCROSSCANADIAN = "" and a NATIVESDKCLANG = "" somewhere hidden in a bbappend of an other layer. This was originally done to preserve space. 
When I included clang in the SDK with "CLANGSDK = 1" the result was:
* the clang executable was included in the SDK as sysroots/x86_64-pokysdk-linux/usr/bin/clang
* the symlinks from the "cang-cross-canadian-${MACHINE}" package where not included due to the variables set empty. That's why the symlinks in sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/ were missing.

closed 2023-03-20 by rungemar