kraj/meta-clang #739: compiler-rt: do_configure failed at compiler version check

JustPretender, 2023-02-12:
I need `libclang` as a dependency for a Rust recipe that uses bindgen. When building `clang` I get the following:

```
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is Clang 14.0.3
| -- The CXX compiler identification is Clang 14.0.3
| -- The ASM compiler identification is Clang with GNU-like command-line
| -- Found assembler: /home/usersetup/yocto/build/all/tmp/work/corei7-64-vendor-linux/compiler-rt/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-vendor-linux/x86_64-vendor-linux-clang
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /home/usersetup/yocto/build/all/tmp/work/corei7-64-vendor-linux/compiler-rt/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-vendor-linux/x86_64-vendor-linux-clang - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /home/usersetup/yocto/build/all/tmp/work/corei7-64-vendor-linux/compiler-rt/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-vendor-linux/x86_64-vendor-linux-clang++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- No build type selected, default to Debug
| -- bolt project is disabled
| -- clang project is disabled
| -- clang-tools-extra project is disabled
| -- compiler-rt project is enabled
| -- cross-project-tests project is disabled
| -- libc project is disabled
| -- libclc project is disabled
| -- libcxx project is disabled
| -- libcxxabi project is disabled
| -- libunwind project is disabled
| -- lld project is disabled
| -- lldb project is disabled
| -- mlir project is disabled
| -- openmp project is disabled
| -- polly project is disabled
| -- pstl project is disabled
| -- flang project is disabled
| -- Performing Test LLVM_LIBSTDCXX_MIN
| -- Performing Test LLVM_LIBSTDCXX_MIN - Failed
| CMake Error at cmake/modules/CheckCompilerVersion.cmake:98 (message):
|   libstdc++ version must be at least 5.1.
| Call Stack (most recent call first):
|   cmake/config-ix.cmake:14 (include)
|   CMakeLists.txt:732 (include)
|
|
| -- Configuring incomplete, errors occurred!
```

I'm the `kirkstone` branch of Yocto.
Maybe unrelated, but I've already tried fiddling with `CLANG_EXTRA_OE_VENDORS` and specifying the target vendor I use.

JustPretender, 2023-02-22:
Closing as I figured out that this was happening due to a change we made to the `libgcc`:
```
do_install:append () {
    rm -rf ${D}${libdir}/gcc
    ln -sf . ${D}${libdir}/gcc
}
```
The "hack" above was added before integrating `meta-clang` that fixes the causing issue correctly.

closed 2023-02-22 by JustPretender

kraj, 2023-02-22:
> Closing as I figured out that this was happening due to a change we made to the `libgcc`:
> 
> ```
> do_install:append () {
>     rm -rf ${D}${libdir}/gcc
>     ln -sf . ${D}${libdir}/gcc
> }
> ```
> 
> The "hack" above was added before integrating `meta-clang` that provides a patch so that clang can find GCC triplets location correctly.

thanks for reporting the fix back. Really appreciated