kraj/meta-clang #743: clang-native uses host include dirs

Lorac, 2023-02-21:
**Describe the bug**
I'm trying to integrate https://codereview.qt-project.org/c/yocto/meta-qt6/+/412562 (pyside6) into our layer before qt eventually does it. During the compilation of pyside6 it invokes shiboken to generate python bindings. Shiboken is linked with libclang and pyside6 fails to compile because the invocation of shiboken with libClang includes host include dir `/usr/include`.

```
| qt.shiboken: (widgets) CLANG builtins includes directory: /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/clang/15.0.7/include
| (widgets) clang_parseTranslationUnit2(0x0, cmd[18]=-isystem/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/clang/15.0.7/include -isystem/usr/include/c++/10 -isystem/usr/include/x86_64-linux-gnu/c++/10 -isystem/usr/include/c++/10/backward -fPIC -Wno-constant-logical-operand -x c++ -std=c++17 -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6 -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/mkspecs/linux-g++ -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include /tmp/QtWidgets_global_XTbvqB.hpp "-DQT_ANNOTATE_ACCESS_SPECIFIER(a)=__attribute__((annotate(#a)))" "-DQT_ANNOTATE_CLASS(type,...)=static_assert(sizeof(#__VA_ARGS__),#type);" -DQSIMD_H)
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:24: error: unknown type name '__fpclassify'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:37: error: expected parameter declarator
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:40: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:60: error: unknown type name 'fpclassify'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:1: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:24:24: error: unknown type name '__signbit'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:24:34: error: expected parameter declarator
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:24:37: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:24:57: error: unknown type name 'signbit'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:24:1: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:29:24: error: unknown type name '__isinf'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:29:32: error: expected parameter declarator
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:29:35: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:29:55: error: unknown type name 'isinf'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:29:1: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:33:24: error: unknown type name '__finite'
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:33:33: error: expected parameter declarator
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:33:36: error: a type specifier is required for all declarations
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:33:56: error: unknown type name 'finite'
| fatal error: too many errors emitted, stopping now [-ferror-limit=]
| (widgets) Errors in -DQSIMD_H:
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:24: error: unknown type name '__fpclassify'
|    /tmp/QtWidgets_global_XTbvqB.hpp:1:10: note: in file included from /tmp/QtWidgets_global_XTbvqB.hpp:1:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h:5:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h:5:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnamespace.h:8:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnamespace.h:8:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qglobal.h:1405:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qglobal.h:1405:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnumeric.h:8:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnumeric.h:8:
|    /usr/include/c++/10/cmath:45:15: note: in file included from /usr/include/c++/10/cmath:45:
|    /usr/include/math.h:289:10: note: in file included from /usr/include/math.h:289:
| 
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:37: error: expected parameter declarator
| /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/bits/mathcalls-helper-functions.h:20:40: error: a type specifier is required for all declarations
|    /usr/include/math.h:285:20: note: expanded from macro '_Mdouble_'
```

**To Reproduce**
Steps to reproduce the behavior:
Copy and paste these to your layers: https://codereview.qt-project.org/c/yocto/meta-qt6/+/412562
Use meta-clang master
`bitbake python3-pyside6`

**Expected behavior**
Should build and not fail on host includes

**Desktop (please complete the following information):**
 - OS: Linux Debian Bullseye
 - Version: 11

kraj, 2023-03-30:
@Lorac do you still see this with clang16 in master ?

Lorac, 2023-03-30:
Different errors, but same problem.

```
| qt.shiboken: (widgets) CLANG builtins includes directory: /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/clang/16.0.1/include
| (widgets) clang_parseTranslationUnit2(0x0, cmd[18]=-isystem/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/clang/16.0.1/include -isystem/usr/include/c++/10 -isystem/usr/include/x86_64-linux-gnu/c++/10 -isystem/usr/include/c++/10/backward -fPIC -Wno-constant-logical-operand -x c++ -std=c++17 -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6 -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/mkspecs/linux-g++ -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore -I/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include /tmp/QtWidgets_global_KTCtuE.hpp "-DQT_ANNOTATE_ACCESS_SPECIFIER(a)=__attribute__((annotate(#a)))" "-DQT_ANNOTATE_CLASS(type,...)=static_assert(sizeof(#__VA_ARGS__),#type);" -DQSIMD_H)
| /usr/include/c++/10/cstddef:50:10: fatal error: 'stddef.h' file not found
| (widgets) Errors in -DQSIMD_H:
| /usr/include/c++/10/cstddef:50:10: fatal: 'stddef.h' file not found
|    /tmp/QtWidgets_global_KTCtuE.hpp:1:10: note: in file included from /tmp/QtWidgets_global_KTCtuE.hpp:1:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h:5:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h:5:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnamespace.h:8:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnamespace.h:8:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qglobal.h:10:12: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qglobal.h:10:
| 
| 
| (widgets) Clang: 1 diagnostic messages:
|   /usr/include/c++/10/cstddef:50:10: fatal: 'stddef.h' file not found
|    /tmp/QtWidgets_global_KTCtuE.hpp:1:10: note: in file included from /tmp/QtWidgets_global_KTCtuE.hpp:1:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h:5:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h:5:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnamespace.h:8:10: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qnamespace.h:8:
|    /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qglobal.h:10:12: note: in file included from /builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore/qglobal.h:10:
| 
| 
| Keeping temporary file: /tmp/QtWidgets_global_KTCtuE.hpp
| shiboken: Error running ApiExtractor.
| Command line:
|     "--generator-set=shiboken"
|     "--enable-parent-ctor-heuristic"
|     "--enable-pyside-extensions"
|     "--enable-return-value-heuristic"
|     "--use-isnull-as-nb_nonzero"
|     "--include-paths=/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6:/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/lib/mkspecs/linux-g++:/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include:/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include/QtCore:/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/recipe-sysroot/usr/include"
|     "--typesystem-paths=/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6:/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6:/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6/QtWidgets"
|     "--output-directory=/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets"
|     "--license-file=/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6/QtWidgets/../licensecomment.txt"
|     "--lean-headers"
|     "--api-version=6.4.2"
|     "/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/build/PySide6/QtWidgets_global.h"
|     "/builds/yocto/${MACHINE}/tmp/work/${DISTRO}/python3-pyside6/6.4.2-r0/git/sources/pyside6/PySide6/QtWidgets/typesystem_widgets.xml"
```

Lorac, 2023-07-13:
I have update the recipe to 6.5.1.1 and used mickeldore branch from meta-clang and now the build passes.

Lorac, 2023-07-13:
Closes it too fast. It only passes because I have GCC 12.2 on my host environment, but our CI runs with GCC 10 on the host and still fails with the same error previously stated :(