kraj/meta-clang #762: Compiler installation brokenu, undefined reference to `std::__1::cout'

mariuszmaximus, 2023-03-16:
I build Toradex [image ](https://developer.toradex.com/linux-bsp/os-development/build-yocto/build-a-reference-image-with-yocto-projectopenembedded/)

Toradex 5.7.2 dunfell with [meta-clang](https://github.com/kraj/meta-clang) dunfell-clang12
there were no problems, after install SDK I have clang compiler. 

But I have problem link "Hello world"  
```c++
#include <iostream>
int main(int, char**) {
    std::cout << "Hello, world!\n";
}
```

```bash
aarch64-tdx-linux-clang++ --target=aarch64-tdx-linux --sysroot=/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux -v  main.cpp
main.cpp:(.text+0x14): undefined reference to `std::__1::cout'
.......
clang-12: error: linker command failed with exit code 1 (use -v to see invocation)

```
[log.txt](https://github.com/kraj/meta-clang/files/10990187/log.txt)
where to find solution?

kraj, 2023-03-16:
@mariuszmaximus can you try 
```txt
${CLANGCXX} main.cpp
```
does it fail same way?

mariuszmaximus, 2023-03-16:
I try compile my project with nlohmann::json 
and have many problems with `<valarray>` on dunfell-clang12 

My first hello world  was:
```
#include <iostream>
#include <valarray>  

int main(int, char**) {
    std::cout << "Hello, world!\n";
}
```
`${CLANGCXX} -O2 -v  main.cpp` result
```
clang version 12.0.0 (https://github.com/llvm/llvm-project fa0971b87fb2c9d14d1bba2551e61f02f18f329b)
Target: aarch64-tdx-linux
Thread model: posix
InstalledDir: /opt/tdx-xwayland/5.7.2/sysroots/x86_64-tdxsdk-linux/usr/bin/aarch64-tdx-linux
Found candidate GCC installation: /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0
Selected GCC installation: /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0
Candidate multilib: .;@m64
Selected multilib: .;@m64
 "/opt/tdx-xwayland/5.7.2/sysroots/x86_64-tdxsdk-linux/usr/bin/clang-12" -cc1 -triple aarch64-tdx-linux -emit-obj --mrelax-relocations -disable-free -disable-llvm-verifier -discard-value-names -main-file-name main.cpp -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=non-leaf -fmath-errno -fno-rounding-math -mconstructor-aliases -munwind-tables -target-cpu generic -target-feature +neon -target-abi aapcs -fallow-half-arguments-and-returns -fno-split-dwarf-inlining -debugger-tuning=gdb -v -resource-dir /opt/tdx-xwayland/5.7.2/sysroots/x86_64-tdxsdk-linux/usr/lib/clang/12.0.0 -D _FORTIFY_SOURCE=2 -isysroot /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux -internal-isystem /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0 -internal-isystem /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/aarch64-tdx-linux -internal-isystem /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/backward -internal-isystem /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/local/include -internal-isystem /opt/tdx-xwayland/5.7.2/sysroots/x86_64-tdxsdk-linux/usr/lib/clang/12.0.0/include -internal-externc-isystem /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/include -internal-externc-isystem /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/include -O2 -Wformat -Wformat-security -Werror=format-security -fdeprecated-macro -fdebug-compilation-dir /opt/tdx-xwayland/5.7.2 -ferror-limit 19 -stack-protector 2 -fno-signed-char -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fcolor-diagnostics -vectorize-loops -vectorize-slp -faddrsig -o /tmp/main-57c2a7.o -x c++ main.cpp
clang -cc1 version 12.0.0 based upon LLVM 12.0.0 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/local/include"
ignoring nonexistent directory "/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0
 /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/aarch64-tdx-linux
 /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/backward
 /opt/tdx-xwayland/5.7.2/sysroots/x86_64-tdxsdk-linux/usr/lib/clang/12.0.0/include
 /opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/include
End of search list.
In file included from main.cpp:2:
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/valarray:1214:5: error: exception specification in declaration does not match previous declaration
    begin(valarray<_Tp>& __va) noexcept
    ^
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/bits/range_access.h:104:31: note: previous declaration is here
  template<typename _Tp> _Tp* begin(valarray<_Tp>&);
                              ^
In file included from main.cpp:2:
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/valarray:1224:5: error: exception specification in declaration does not match previous declaration
    begin(const valarray<_Tp>& __va) noexcept
    ^
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/bits/range_access.h:105:37: note: previous declaration is here
  template<typename _Tp> const _Tp* begin(const valarray<_Tp>&);
                                    ^
In file included from main.cpp:2:
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/valarray:1234:5: error: exception specification in declaration does not match previous declaration
    end(valarray<_Tp>& __va) noexcept
    ^
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/bits/range_access.h:106:31: note: previous declaration is here
  template<typename _Tp> _Tp* end(valarray<_Tp>&);
                              ^
In file included from main.cpp:2:
/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/lib/aarch64-tdx-linux/9.5.0/../../../include/c++/9.5.0/valarray:1249:5: error: exception specification in declaration does not match previous declaration
    end(const valarray<_Tp>& __va) noexcept
    ^
[... cut ...] ../../include/c++/9.5.0/bits/range_access.h:107:37: note: previous declaration is here
  template<typename _Tp> const _Tp* end(const valarray<_Tp>&);
                                    ^
4 errors generated.
```
I found on my sysroot two folders  with valarray 
`/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/include/c++/v1/valarray`
and 
`/opt/tdx-xwayland/5.7.2/sysroots/aarch64-tdx-linux/usr/include/c++/9.5.0/valarray'

I renamed v1 to 9.5.0  then clang compile .o but can't  link output

kraj, 2023-03-16:
Can you use 

`${CLANGCXX} -rtlib=compiler-rt --libunwind=libunwind -stdlib=libc++ main.cpp`

if it works then you need to set LD to same as above

mariuszmaximus, 2023-03-16:
clang-12: error: unsupported option '--libunwind=libunwind' 
but 
`${CLANGCXX} -rtlib=compiler-rt -stdlib=libc++ main.cpp` 
create `a.out`
I'll check tommorow on my linux

closed 2023-03-17 by mariuszmaximus