kraj/meta-clang #785: Honister syntax used in dunfell branch

smorris-witekio, 2023-05-08:
**Describe the bug**
This commit uses Honister syntax(colon instead of underscore), which is not supported in Dunfell, so builds will break:
https://github.com/kraj/meta-clang/commit/67ac939d47b3dfc383fec7244e4a0ff85ca4340f

Release notes on honister: https://docs.yoctoproject.org/migration-guides/migration-3.4.html

**To Reproduce**
Build the clang recipe using the dunfell version of Yocto.

I expect this was committed to the dunfell branch by mistake. But maybe the colon just needs to be changed to an underscore.

kraj, 2023-05-09:
Please update to latest release of `dunfell` for both `oe-core` and `bitbake` or `poky` whatever you use.

smorris-witekio, 2023-05-10:
Oh you're right, they do seem to support both syntax there now. Thank you. Closing...

closed 2023-05-10 by smorris-witekio