kraj/meta-clang #796: library missing for riscv target

abhishekwalia33, 2023-05-17:
Hi All,

I am working on riscv target boards, clang is compiled fine and simple hello world program is compiled successfully using clang-12 but when i am using the -fsanitize=address option then it is giving the error.

1 :~# cat test.c
#include <stdlib.h>
int main() {
  char *x = (char*)malloc(10 * sizeof(char*));
  free(x);
  return x[5];
}
2 :~# clang -O -g -fsanitize=address test.c
/usr/bin/riscv64-oe-linux-ld: cannot find /usr/lib/clang/12.0.0/lib/linux/libclang_rt.asan-riscv64.a: No such file or directory
clang-12: error: linker command failed with exit code 1 (use -v to see invocation)

I dont known why the clang is not generating this missing library Please suggest any solution.
Any help will be highly appreciated.

Thanks

kraj, 2023-05-17:
can you see if it happens with master/clang16 as well ?

abhishekwalia33, 2023-05-18:
I tried  but it is expecting the classes ,Are these classes should be present in meta-clang or we have to made it available from other layers, How to handle this issue.

**the layer cloned:**
$ git clone --recursive -b master https://github.com/kraj/meta-clang.git

**clang version:**
MAJOR_VER = "16"
MINOR_VER = "0"
PATCH_VER = "3"

**Issue:**
$ bitbake clang
Error: meta-clang/recipes-devtools/clang/openmp_git.bb:15: Could not inherit file classes/python3targetconfig.bbclass

kraj, 2023-05-18:
you need to use master branch of poky/openembedded-core repo as well.

abhishekwalia33, 2023-05-18:
I have cloned master branch of openembedded-core it provided the required classes but when compiling again facing issue of rust module no attribute, check it once why it is coming.

**issue:**
sources/meta-clang/recipes-devtools/bindgen-cli/bindgen-cli_0.64.0.bb
Traceback (most recent call last):
  File "Var <RUST_HOST_SYS>", line 1, in <module>
  File "/home/abhishek/abhi/nb2/sources/meta-clang/classes/rust-common.bbclass", line 54, in rust_base_triple(d=<bb.data_smart.DataSmart object at 0x7f060b2b4eb8>, thing='HOST'):
         else:
    >        arch = oe.rust.arch_to_rust_arch(d.getVar('{}_ARCH'.format(thing)))
     
bb.data_smart.ExpansionError: Failure expanding variable RUST_HOST_SYS, expression was ${@rust_base_triple(d, 'HOST')} which triggered exception AttributeError: module 'oe' has no attribute 'rust'

kraj, 2023-05-18:
where is `/home/abhishek/abhi/nb2/sources/meta-clang/classes/rust-common.bbclass` coming from?

abhishekwalia33, 2023-05-19:
I kept all the required classes in meta-clang class directory copying from master openembedded -core ,Is it correct way or not because we are using dunfell branch of openembedded-core in which the required classes are not present.

If we should not copy the required classes from openembedded-core master then how to handle it in dunfell branch , try to suggest how should i move forward.

kraj, 2023-05-19:
move all layers to use master, do not copy things around that will be difficult and you will differ a lot with upstream.

abhishekwalia33, 2023-05-22:
sure,but I cant do this because we are using dunfell branch as specific for our bsp,one other way i find to enable the -fsanitize=adress  flag in gcc as CFLAGS,CXXFLAGS anf LDFLAGS so that we can use sanitize feature on gcc itself,Could you please guide me.