kraj/meta-clang #806: dunfell: compiler-rt: invalid argument causes sanity check to fail

Escalion, 2023-06-02:
**Describe the bug**
When using dunfell branch with latest poky:dunfell and meta-oe:dunfell, compiler-rt fails to build.
Failure is caused by invalid argument '-fno-diagnostics-show-caret'.

**To Reproduce**
Steps to reproduce the behavior:
1. Create build as above
2. bitbake clang
3. bitbake compiler-rt
4. See error

**Expected behavior**
Build to succeed

Error:

```
Run Build Command(s):ninja cmTC_c3989 && [1/2] Building C object CMakeFiles/cmTC_c3989.dir/testCCompiler.c.o
|     FAILED: CMakeFiles/cmTC_c3989.dir/testCCompiler.c.o
|     /home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang   -target aarch64-poky-linux  -mcpu=cortex-a72+crc+crypto   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0=/usr/src/debug/compiler-rt/12.0.0-r0                      -fdebug-prefix-map=/home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0=/usr/src/debug/compiler-rt/12.0.0-r0                      -fdebug-prefix-map=/home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0/recipe-sysroot-native=  -fno-diagnostics-show-caret -target aarch64-poky-linux  -mcpu=cortex-a72+crc+crypto   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ryan/workspace/apollo/build/tmp/work/aarch64-poky-linux/compiler-rt/12.0.0-r0/recipe-sysroot -o CMakeFiles/cmTC_c3989.dir/testCCompiler.c.o   -c testCCompiler.c
|     clang-12: error: unknown argument: '-fno-diagnostics-show-caret'
|     ninja: build stopped: subcommand failed.
```

Escalion, 2023-06-05:
I've narrowed this down to ICECC setting -fno-diagnostics-show-caret.
Adding compiler-rt to ICECC_SYSTEM_PACKAGE_BL resolved the issue.
Could also set CFLAGS_append=" -Wno-error=unused-command-line-argument" for the same effect and allow distributed build.

kraj, 2023-06-06:
clang's option for similar effect is `-fno-caret-diagnostics` so somewhere ICECC needs to understand clang or gcc being the compiler. I dont know what we can do in meta-clang for this.