kraj/meta-clang #824: libcxx-native fails on imx8mp

tepperson2, 2023-07-17:
**Describe the bug**
I am unable to build the libcxx-native package. I am building on x86_64 ubuntu.

Here is part of the build output log.
```
| -- Check for working CXX compiler: /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++
| -- Check for working CXX compiler: /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++ - broken
| CMake Error at /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/share/cmake-3.25/Modules/CMakeTestCXXCompiler.cmake:63 (message):
|   The C++ compiler
| 
|     "/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/build/CMakeFiles/CMakeScratch/TryCompile-0HR4dy
| 
|     Run Build Command(s):ninja cmTC_a0b44 && [1/2] Building CXX object CMakeFiles/cmTC_a0b44.dir/testCXXCompiler.cxx.o
|     [2/2] Linking CXX executable cmTC_a0b44
|     FAILED: cmTC_a0b44
|     : && /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++ -target x86_64-linux   -isystem/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/include -O2 -pipe   -stdlib=libstdc++ -target x86_64-linux   -isystem/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/include -O2 -pipe   -stdlib=libstdc++  -L/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib                         -L/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/lib                         -Wl,-O1 -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/thomas/yocto2/build_xwayland/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -rtlib=libgcc --unwindlib=libgcc  -L/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib                         -L/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/lib                         -Wl,-O1 -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/thomas/yocto2/build_xwayland/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -rtlib=libgcc --unwindlib=libgcc CMakeFiles/cmTC_a0b44.dir/testCXXCompiler.cxx.o -o cmTC_a0b44   && :
|     /home/thomas/yocto2/build_xwayland/tmp/hosttools/ld: cannot find -lstdc++: No such file or directory
|     clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.

```

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser: chromium
 - Version 114.0.5735.198

**Additional context**
I am using the mickledore branch of meta-clang.

kraj, 2023-07-17:
Can you install libstdc++ on your. build machine ?

tepperson2, 2023-07-17:
I have libstdc++6, libstdc++-10-dev, libstdc++-11-dev, libx32stdc++-11-dev, and libx32stdc++6 installed.
No change in behavior.

kraj, 2023-07-17:
Can you try to compile a sample cxx file e.g.

hello.cpp
```c++
#include <iostream>
#include <cmath>
#include <cstdlib>
int main( int argc, char** argv )
{
  char* a = (char*)malloc ( 20 );
  std::cout << "hello world" << std::endl;
  return 0;
}
```

```
/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++ -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ hello.cpp
```

tepperson2, 2023-07-17:
I couldn't compile it because it couldn't find iostream. I reduced it even further to get it to try to compile.

```
int main( int argc, char** argv )
{
  return 0;
}
```
I get this in response.
```
/usr/bin/ld: cannot find -lstdc++: No such file or directory
clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
```

And for a bit more information
```
$ /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++ -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ groot.cpp -v
clang version 15.0.7 (https://github.com/llvm/llvm-project 8dfdcc7b7bf66834a761bd8de445840ef68e4d1a)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/10
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/11
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/12
Selected GCC installation: /usr/lib/gcc/x86_64-linux-gnu/12
Candidate multilib: .;@m64
Selected multilib: .;@m64
 "/home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang-15" -cc1 -triple x86_64-unknown-linux-gnu -emit-obj -mrelax-all --mrelax-relocations -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name groot.cpp -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=all -fmath-errno -ffp-contract=on -fno-rounding-math -mconstructor-aliases -funwind-tables=2 -target-cpu x86-64 -tune-cpu generic -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -fcoverage-compilation-dir=/home/thomas/temp -resource-dir /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib/clang/15.0.7 -internal-isystem /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++ -internal-isystem /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/x86_64-linux-gnu -internal-isystem /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/backward -internal-isystem /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib/clang/15.0.7/include -internal-isystem /usr/local/include -internal-isystem /usr/lib/gcc/x86_64-linux-gnu/12/../../../../x86_64-linux-gnu/include -internal-externc-isystem /usr/include/x86_64-linux-gnu -internal-externc-isystem /include -internal-externc-isystem /usr/include -fdeprecated-macro -fdebug-compilation-dir=/home/thomas/temp -ferror-limit 19 -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fcolor-diagnostics -faddrsig -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o /tmp/groot-724e50.o -x c++ groot.cpp
clang -cc1 version 15.0.7 based upon LLVM 15.0.7 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/x86_64-linux-gnu"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/backward"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/12/../../../../x86_64-linux-gnu/include"
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++
 /home/thomas/yocto2/build_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/lib/clang/15.0.7/include
 /usr/local/include
 /usr/include/x86_64-linux-gnu
 /usr/include
End of search list.
 "/usr/bin/ld" -pie -z relro --hash-style=gnu --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 -o a.out /lib/x86_64-linux-gnu/Scrt1.o /lib/x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/12/crtbeginS.o -L/usr/lib/gcc/x86_64-linux-gnu/12 -L/usr/lib/gcc/x86_64-linux-gnu/12/../../../../lib64 -L/lib/x86_64-linux-gnu -L/lib/../lib64 -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib64 -L/lib -L/usr/lib /tmp/groot-724e50.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /usr/lib/gcc/x86_64-linux-gnu/12/crtendS.o /lib/x86_64-linux-gnu/crtn.o
/usr/bin/ld: cannot find -lstdc++: No such file or directory
```

tepperson2, 2023-07-17:
I needed to install libstdc++-12-dev. I found a stackoverflow answer that pointed me in this direction. https://stackoverflow.com/questions/74543715/usr-bin-ld-cannot-find-lstdc-no-such-file-or-directory-on-running-flutte

closed 2023-07-17 by tepperson2

kraj, 2023-07-17:
OK so where is `libstdc++.so` located in `/usr/` ?