kraj/meta-clang #835: dunfell: Creating a dunfell-clang14 branch

rakuco, 2023-08-08:
This has been discussed in the meta-browser repository for a while, most recently in OSSystems/meta-browser#734. Clang 12 is too old for Chromium at this point as the project continues to adopt more C++20 features, and the lack of support for building the native bits with meta-clang's libc++ causes the build to fail either because the host's libstdc++ is too old (OSSystems/meta-browser#740) or just incompatible with the rest of the build (OSSystems/meta-browser#726).

A few weeks ago I created a dunfell-clang14 branch in my meta-clang fork (https://github.com/rakuco/meta-clang/tree/dunfell-clang14) that contains a lot of cherry-picks from meta-clang's kirkstone branch on top of a dunfell-clang12 branch. This tree has been tested with meta-browser, and chromium-x11 and chromium-ozone-wayland build and run fine on different architectures.

What would be the procedure for getting this work reviewed and integrated into this repository? I'm pretty sure I've cherry-picked some unneeded changes, and skipped several others that were not relevant to meta-browser (like all the bpf, bcc and android-tools commits).

nrpt-m, 2023-08-29:
@kraj,
It's been pending from last 3 weeks. Could you please let @rakuco know about the procedure for getting this work reviewed and integrated into this repository?

rwmacleod, 2023-09-07:
@rakuco seems that @kraj is waiting for a PR. Do you have time for that soon?

On Thu, Sep 7, 2023 at 4:16 AM Randy MacLeod <rwmacleod@gmail.com> wrote:
>
> Have a great vacation! Thanks for letting us know that you'll be away.
>
> Khem,
>
> Is meta-clang/dunfell-clang14 waiting on you having some time to work on it or is it blocked by something else?

I think Rakuco has already done the work, so I have asked him to send
a pull against this branch. He has not yet done it.

Hide quoted text

>
> Randy
>
>
> On Thu., Sep. 7, 2023, 06:49 Max Ihlenfeldt, <max@igalia.com> wrote:
>>
>> Hi,
>>
>> Just wanted to let you know that I'll be on vacation the next two weeks.
>>
>> I'd be happy to work on the update to 117 afterwards (i.e. starting September 25) if no one else wants to start working on it earlier. I suppose it depends on the meta-clang/dunfell-clang14 progress - not sure if we want to have three PRs open at the same time that depend on each other and are waiting for that branch.
>>
>> Best regards
>> Max

rakuco, 2023-09-08:
> I think Rakuco has already done the work, so I have asked him to send
> a pull against this branch. He has not yet done it.

#853 finally sent!

rakuco, 2023-09-08:
Hmm, dunno why this issue did not get closed when #853 was merged. The new branch is in, and it corresponds to what I had in my fork.

PR #853 (2023-09-08): dunfell-clang14: Update from clang 12 to clang 14
This gigantic pull request contains a lot of cherry-picks from the kirkstone branch that move clang from version 12 to version 14. This is necessary to update Chromium beyond M114 in meta-chromium, as clang 12 is too old for the language features used by Chromium.

This has been compiled and runtime-tested as described in https://github.com/OSSystems/meta-browser/issues/734#issuecomment-1661928446 and subsequent comments.

Parts of the changes had already landed separately in the dunfell-clang12 branch, so I did have to manually fix some confli [... cut]
closed 2023-09-08 by rakuco