kraj/meta-clang #836: Nothing PROVIDES 'virtual/nativesdk-x86_64-ekosdk-linux-binutils'

Lorac, 2023-08-09:
**Describe the bug**
Using poky latest tagged mickledore release and meta-clang latest master. I get this error:

```
ERROR: Nothing PROVIDES 'virtual/nativesdk-x86_64-ekosdk-linux-binutils' (but virtual:nativesdk:/home/maxime/devel/yocto-paris/meta-clang/recipes-devtools/clang/clang_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/nativesdk-x86_64-ekosdk-linux-compilerlibs
  virtual/nativesdk-x86_64-ekosdk-linux-go-runtime
  virtual/x86_64-eko-linux-binutils

ERROR: Required build target 'meta-toolchain-sonatest' has no buildable providers.
Missing or unbuildable dependency chain was: ['meta-toolchain-mysdkdistro', 'nativesdk-clang-format', 'virtual/nativesdk-x86_64-ekosdk-linux-binutils']
```

**To Reproduce**
Use mickledore poky, latest meta-clang master, and build a SDK with nativesdk-clang-format present.

kraj, 2023-08-10:
poky/mickledone + meta-clang/master wont work out of box. You need to backport https://github.com/YoeDistro/poky/commit/9a20fbe23f63d87e30b0291ae1dba863c0f5de31 to poky.

Lorac, 2023-08-10:
Awesome, thanks for the pointer. It worked.