kraj/meta-clang #840: bcc selftest failures on non-x86_64 arch machines

sundeep-kokkonda, 2023-08-17:
We enabled the _bcc_ in Yocto framework and found there are a few test cases (listed below) with bcc selftest are getting failed on non x86_64 architectures (arm64/riscv64).

In Poky, meta-openembedded & meta-clang layers added along with below _local.conf_ changes:

> IMAGE_INSTALL:append = "bcc bpftrace gdb gdbserver valgrind libc-mtrace strace libbpf"
> IMAGE_FEATURES += "ssh-server-dropbear"
> CORE_IMAGE_EXTRA_INSTALL += "bcc-ptest"
> EXTRA_IMAGE_FEATURES = "debug-tweaks dbg-pkgs dev-pkgs tools-sdk tools-debug tools-profile"
> MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS += "kernel-modules"
> IMAGE_ROOTFS_SIZE = "8192000"
> IMAGE_GEN_DEBUGFS = "1"
> KERNEL_DEBUG = "True"
> DEBUG_BUILD="1"

FAILED CASES ( cc resolve symbol addresses for a given PID; cc resolve symbols using /tmp/perf-pid.map; cc test bpf stack table; cc test bpf stack_id table; cc test hash of maps; cc test hash of maps using custom key; cc test array of maps; cc test read perf event; cc test usdt argument parsing; cc test finding a probe in our own process; cc test fine a probe in our own binary with C++ API; cc test fine probes in our own binary with C++ API; cc test fine a probe in our Process with C++ API; cc test find a probe in our process' shared libs with c++ API; cc test usdt partial init w/ fail init_usdt; cc Test uprobe refcnt semaphore activation;)

We tried selftest by compiling the bcc code by setting the cmake options - CMAKE_USE_LIBBPF_PACKAGE, DENABLE_LLVM_SHARED, ENABLE_USDT, ENABLE_CPP_API, ENABLE_CLANG_JIT, ENABLE_TESTS along with the kernel options given in https://github.com/iovisor/bcc/blob/master/docs/kernel_config.md but with all these settings the test failures are same. (There are several test executables generated by changing these settings like - test_libbcc, test_libbcc_no_libbpf, test_static. And, with all executables the test failures are same.)

There is also an upstream issue https://github.com/iovisor/bcc/issues/4681 is raised but did not get any update from community yet.

We've analyzed a few cases by attaching the gdb and found that there is a lot of diff in probe assertions w.r.t x86_64 (See below variables from gdb). We suspect that the probe assertions are not happening and do not know what could cause this.

_arm64:_
prevAssertions = {passed = 862809464, failed = 4, failedButOk = 549755809248}
assertions = {passed = 4, failed = 879586680, failedButOk = 0}

_x86_64:_
prevAssertions = {passed = 140737488346592, failed = 4, failedButOk = 913141112}
assertions = {passed = 140737488346624, failed = 4, failedButOk = 929918328}

Can you help us here to find what could be the reason for these failures and how to fix these test cases.

Thanks,
Sundeep K.

shr-project, 2023-09-06:
Don't want to hijack this issue ticket, but I've recently noticed bcc entries in my .gitconfig and they are added by:

https://github.com/iovisor/bcc/pull/4186/files

Should we remove all that as we're not using submodules and bitbake fetcher handles git safe directories?

shr-project, 2023-09-07:
bcc moved to right dynamic-layer and added patch to avoid .gitconfig changes in:
https://github.com/kraj/meta-clang/pull/852