kraj/meta-clang #856: kernel build error

lxy1579, 2023-09-11:
**Describe the bug**
make oldconfig

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kraj, 2023-09-11:
we do not use clang to compile by default for kernel it still uses gcc and binutils. Maybe you can disable lld to build kenrel.

```
KERNEL_LD = "${CCACHE}${HOST_PREFIX}ld.bfd ${HOST_LD_KERNEL_ARCH}"
```

lxy1579, 2023-09-12:
It will be OK if I compile kernel by removing `-fus-ld=lld` from KERNEL_CC, but I also set KERNEL_LD = "${CCACHE}${HOST_PREFIX}ld.lld". Verify that the final image(vmlinux) is also generated through ld.lld link

lxy1579, 2023-10-19:
> we do not use clang to compile by default for kernel it still uses gcc and binutils. Maybe you can disable lld to build kenrel.
> 
> ```
> KERNEL_LD = "${CCACHE}${HOST_PREFIX}ld.bfd ${HOST_LD_KERNEL_ARCH}"
> ```

Is there a plan to support `LLVM=1` parameter to kernel build?