kraj/meta-clang #857: libcxx-native fails to configure in kirkstone

pbeTrifork, 2023-09-19:
**Describe the bug**
When trying to compile libcxx-native, do_configure fails with the error message:
... ld:  cannot find -lstdc++: No such file or directory

**To Reproduce**
Steps to reproduce the behavior:
1. Check out poky kirkstone branch
2. add meta-clang kirkstone branch to bblayers.conf
3. `bitbake libcxx-native`
4. CMake fails to configure because clang++ is not able to compile a simple test program:

```
The C++ compiler
| 
|     "${TOPDIR}/tmp/work/x86_64-linux/compiler-rt-native/14.0.6-r0/recipe-sysroot-native/usr/bin/clang++"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
...
|     [2/2] Linking CXX executable cmTC_3d1f4
|     FAILED: cmTC_3d1f4
...
|     ${TOPDIR}/tmp/hosttools/ld: cannot find -lstdc++: No such file or directory
|     clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.

```
(the ${TOPDIR} is substituted in the bugreport)

**Expected behavior**
libcxx-native is built

**Desktop (please complete the following information):**
```
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
```
```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.12"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:5d822b31316663c838c5864ab68b28fb3ca41351"
meta-clang           = "kirkstone:2d08d6bf376a1e06c53164fd6283b03ec2309da4"
```

**Additional context**
The same issue arises with compiler-rt

kraj, 2023-09-19:
it needs libstdc++ installed on build host. So make sure this is installed.

pbeTrifork, 2023-09-20:
```
libstdc++-11-dev/jammy-updates,jammy-security,now 11.4.0-1ubuntu1~22.04 amd64 [installed]
```
however for recent 22.04 the package `libstdc++-12-dev` is used instead.

(closing as this solves it)

closed 2023-09-20 by pbeTrifork