kraj/meta-clang #859: lldb-tblgen not found in SDK generation

fidelski, 2023-09-21:
**populate_sdk fails on master**

This is happening on the master branch (just in case): When I want to install the SDK (with `CLANGSDK=1`) it fails with the following error:

```sh
| install: cannot stat '/mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/build/bin/lldb-tblgen': No such file or directory
| WARNING: /mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/temp/run.do_install.2498191:158 exit 1 from 'install -Dm 0755 /mnt/data/yocto/bf133/build/tmp/work/x86_64-nat
ivesdk-pokysdk-linux/nativesdk-clang/17.0.1/build/bin/lldb-tblgen /mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/image/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x8
6_64-pokysdk-linux/usr/bin/lldb-tblgen'
| WARNING: Backtrace (BB generated script):
|       #1: do_install, /mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/temp/run.do_install.2498191, line 158
|       #2: main, /mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/temp/run.do_install.2498191, line 198
ERROR: Task (virtual:nativesdk:/home/david/yocto/bf133/meta-clang/recipes-devtools/clang/clang_git.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6120 tasks of which 6119 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:nativesdk:/home/david/yocto/bf133/meta-clang/recipes-devtools/clang/clang_git.bb:do_install
Summary: There were 3 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.
```
But lldb-tblgen is there - just not in the searched directory.

```sh
❯ find . -iname 'lldb-tblgen'
./tmp/work/x86_64-linux/clang-native/17.0.1/image/mnt/data/yocto/bf133/build/tmp/work/x86_64-linux/clang-native/17.0.1/recipe-sysroot-native/usr/bin/lldb-tblgen
./tmp/work/x86_64-linux/clang-native/17.0.1/sysroot-destdir/mnt/data/yocto/bf133/build/tmp/work/x86_64-linux/clang-native/17.0.1/recipe-sysroot-native/usr/bin/lldb-tblgen
./tmp/work/x86_64-linux/clang-native/17.0.1/build/bin/lldb-tblgen
./tmp/work/x86_64-linux/clang-cross-powerpc64/17.0.1/recipe-sysroot-native/usr/bin/lldb-tblgen
./tmp/work/x86_64-linux/clang-crosssdk-x86_64/17.0.1/recipe-sysroot-native/usr/bin/lldb-tblgen
./tmp/work/ppc64e5500-vrag-linux/compiler-rt/17.0.1/recipe-sysroot-native/usr/bin/lldb-tblgen
./tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/recipe-sysroot-native/usr/bin/lldb-tblgen
./tmp/sysroots-components/x86_64/clang-native/usr/bin/lldb-tblgen
```

**To Reproduce**
Steps to reproduce the behavior:
1. Configure to build the clang-sdk (`CLANGSDK=1`)
2. Build the SDK with `bitbake core-image-minimal -c populate_sdk`
3. This gives the error that lldb-tblgen cannot be found

**Expected behavior**
The SDK gets build without errors.

**Desktop**
 - OS: Arch Linux

kraj, 2023-09-21:
Thanks for report. Can you also add the Build configuration that bitbake prints at the start ?

fidelski, 2023-09-21:
The output starts with the following:

```
❯ bitbake vrag-image -c populate_sdk                                                                                                                                                                   [8800/61316]
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)... (16:22:04.503012)
WARNING: Layer meta-vrag should set LAYERSERIES_COMPAT_meta-vrag in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer meta-bf133 should set LAYERSERIES_COMPAT_meta-bf133 in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Host distribution "arch" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:00
Loaded 3426 entries from dependency cache.
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:00:00
Parsing of 2101 .bb files complete (2100 cached, 1 parsed). 3426 targets, 411 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.6.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "powerpc64-vrag-linux"
MACHINE              = "bf133"
DISTRO               = "vrag"
DISTRO_VERSION       = "1.0.0"
TUNE_FEATURES        = "m64 fpu-hard e5500 bigendian"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "HEAD:aab44111d72d87528cfb9e210b6217a09d1636d4"
meta-vrag            = "HEAD:fb2958836891129175c02ec6b83e73308ea8e09a"
meta-bf133           = "HEAD:65d8ac7c91123e140254a37c5926369c4ac16b47"
meta-oe
meta-networking
meta-webserver       = "HEAD:250cb228ba98f7f7945ee5a5a94793aa19d0f53e"
meta-mingw           = "HEAD:65ef95a74f6ae815f63f636ed53e140a26a014ce"
meta-clang           = "HEAD:b525885a79c9110621e0808edf4b68d1bc2cc3a5"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 18 Local 0 Mirrors 0 Missed 18 Current 2710 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: nativesdk-clang-17.0.1-r0 do_install: ExecutionError('/mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/temp/run.do_install.2498191', 1, None, None)
ERROR: Logfile of failure stored in: /mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/temp/log.do_install.2498191
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/david/yocto/bf133/meta-clang/recipes-devtools/clang/clang-crosssdk_git.bb:do_populate_sysroot', '/home/david/yocto/bf133/meta/recipes-devtools/binutils/binutils-crosssdk_2
.41.bb:do_populate_sysroot', '/home/david/yocto/bf133/meta/recipes-devtools/cmake/cmake-native_3.27.4.bb:do_populate_sysroot', '/home/david/yocto/bf133/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populat
e_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-core/libxml/libxml2_2.11.5.bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/chrpath/chrpath_0.16.bb:do_popul
ate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/libedit/libedit_20230828-3.1.bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/ninja/ninja_1.11.1.
bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/pkgconfig/pkgconfig
_git.bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/python/python3
_3.11.5.bb:do_populate_sysroot', 'virtual:native:/home/david/yocto/bf133/meta/recipes-devtools/swig/swig_4.1.1.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/recipes-core/libxml/libxml2
_2.11.5.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/recipes-core/zlib/zlib_1.3.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/recipes-devtools/binutils/binu
tils_2.41.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/recipes-devtools/libedit/libedit_20230828-3.1.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/recipes-d
evtools/python/python3_3.11.5.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/recipes-extended/zstd/zstd_1.5.5.bb:do_populate_sysroot', 'virtual:nativesdk:/home/david/yocto/bf133/meta/re
cipes-support/libffi/libffi_3.4.4.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['clang-crosssdk-x86_64', 'binutils-crosssdk-x86_64-pokysdk-linux', 'cmake-native', 'quilt-native', 'libxml2-native', 'chrpath-native', 'libedit-native', 'ninja-nat
[... cut ...]
il-linux-libuuid', 'nativesdk-libxcrypt', 'nativesdk-openssl', 'nativesdk-libnsl2', 'binutils-native', 'nativesdk-linux-libc-headers', 'gettext-minimal-native', 'm4-native', 'nativesdk-libgcc', 'perl-native', 'm
ake-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| NOTE: DESTDIR=/mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/image VERBOSE=1 cmake --build /mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/native
sdk-clang/17.0.1/build --target install --
| Change Dir: '/mnt/data/yocto/bf133/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/17.0.1/build'
```

kraj, 2023-09-21:
This should be fixed now. Please test with tip of trunk and report back

closed 2023-09-21 by kraj

fidelski, 2023-09-22:
Thanks a lot for this reaction and bugfix at the speed of light. :heart: 
 All good now, the SDK gets built without errors.