kraj/meta-clang #869: Request for nanbield branch variant with LLVM 15.x

zboszor, 2023-10-19:
**Is your feature request related to a problem? Please describe.**

meta-clang master and meta-intel master/nanbield when used together fails to build intel-graphics-compiler because [IGC does not yet support anything past LLVM 15.x](https://github.com/intel/intel-graphics-compiler/projects/5).

As such, GPU acceleration must be disabled in oneDNN, and the whole Intel Compute Runtime must be left out.

**Describe the solution you'd like**

I will need a meta-clang nanbield branch variant that carries CLANG/LLVM 15.x latest point release so meta-intel can be used together.

**Describe alternatives you've considered**

I considered removing/replacing the Intel Compute Runtime (OpenCL driver) but Mesa Rusticl is not a viable alternative yet to support either SYCL or backend for oneDNN.

**Additional context**

kraj, 2023-10-19:
I am open for reviewing and accepting patches if someone works on it. Personally I do not have time or interest in supporting older clang on newer OE releases.

anujm1, 2023-12-13:
I think just adding nanbield to LAYERSERIES_COMPAT for mickledore branch is enough. I tried it and it compiles fine with oe-core, meta-intel, meta-oe, meta-python in bblayers.conf. If that is preferred over having a separate branch, I can send that change.

kraj, 2023-12-13:
> I think just adding nanbield to LAYERSERIES_COMPAT for mickledore branch is enough. I tried it and it compiles fine with oe-core, meta-intel, meta-oe, meta-python in bblayers.conf. If that is preferred over having a separate branch, I can send that change.

that should be fine.

saininav, 2023-12-19:
https://github.com/kraj/meta-clang/pull/892

zboszor, 2024-08-01:
See https://github.com/kraj/meta-clang/pull/924#issuecomment-2175957813

I am working on adding CLANG 15.0.7 to meta-clang with a set of versioned recipe names: `clang15`, etc.

This, together with small changes in `meta-intel` allows building IGC, Compute Runtime and opencl-clang 15 with CLANG 15 and linked to libLLVM-15.

All the recipes already adapted to the latest (non-versioned) LLVM/CLANG recipes are built with that version.

closed 2024-09-20 by zboszor