kraj/meta-clang #876: mickledore: build fails using Ubuntu 22.04

varigigi, 2023-10-29:
**Describe the bug**
Trying to build Chromium using the official NXP release mickledore-6.1.22-2.0.0 using a Ubuntu 22.04 LTS  machine, the build fails.
However, NXP approach is reasonably not relevant: please read the _Additional context_ section.

**To Reproduce**
To reproduce the behavior, you can follow the instructions reported in [L6.1.22_2.0.0_LINUX_DOCS](https://www.nxp.com/webapp/Download?colCode=L6.1.22_2.0.0_LINUX_DOCS)

**Additional context**
When installing packages for a Yocto build, the package _build-essential_ is installed, causing the installation of gcc-multilib and g++-multilib: on a stock Ubuntu 22.04 machine, this causes the installation of gcc-11-multilib and g++-11-multilib.
Ubuntu 22.04 originally comes with [dkms 2.8.7-2ubuntu2](https://packages.ubuntu.com/jammy/dkms) package from jammy repository: this specific package depends on the default gcc installation: on Ubuntu 22.04 this is gcc-11
When the "jammy-updates" repository is enabled (reasonably, a common condition for several users), an updated version [dkms 2.8.7-2ubuntu2.2](https://packages.ubuntu.com/jammy-updates/dkms) is installed: this specific package depends on the default gcc-12 which is automatically installed during the dkms update.
With these conditions, when meta-clang try building chromium dependencies, you will get
> Summary: 2 tasks failed:
  virtual:native:/my-yocto-mickledore/sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure
  virtual:native:/my-yocto-mickledore/sources/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_configure
Summary: There were 2 ERROR messages, returning a non-zero exit code.

Inspecting the origin of the build error, you will find
> /my-yocto-mickledore/build_mx8mp_xwayland/tmp/hosttools/ld: cannot find -lstdc++: No such file or directory
clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

This error is only present if the updated dkms package is installed.
Inspecting how clang works, I can see
> $ /my-yocto-mickledore/build_mx8mp_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin/clang++ --verbose
clang version 15.0.7 (GitHub - llvm/llvm-project: The LLVM Project is a collection of modular and reusable compiler and toolchain technologies.  8dfdcc7b7bf66834a761bd8de445840ef68e4d1a)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /my-yocto-mickledore/build_mx8mp_xwayland/tmp/work/x86_64-linux/libcxx-native/15.0.7-r0/recipe-sysroot-native/usr/bin
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/11
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/12
**Selected GCC installation: /usr/lib/gcc/x86_64-linux-gnu/12**
Candidate multilib: .;@ m64
Candidate multilib: 32;@ m32
Candidate multilib: x32;@ mx32
Selected multilib: .;@ m64

As far as I can understand, clang automatically select gcc-12 but g++-12 installation is missing, originating the build error.
The following workaround makes the Ubuntu 22.04 installation more _coherent_ from a clang perspective:
> $ sudo apt install gcc-12-multilib g++-12-multilib

Is this something that should be somehow either documented or fixed ?

kraj, 2023-11-02:
@varigigi thanks for report. The issue should be documented perhaps in README, that it depends on libstdc++ and g++ to be installed on build host and the versions should match the gcc version installed.

ferlzc, 2025-02-19:
This happend to me on Scarthgap too, have to add to my host system

apt install gcc-11-multilib

myabe this could be an addition to README of the layer

kraj, 2025-02-21:
@ferlzc I am open if someone cooks a patch and creates a PR, although I would like to ensure that its ubuntu specific issue or happens on other distros as well.