kraj/meta-clang #879: Kirkstone: libclang package is empty

fwuehr95, 2023-11-09:
Looking at the clang recipe in kirkstone branch FILES for the libclang package is set to the following:

FILES:libclang = "\
  ${libdir}/libclang.so.${MAJOR_VER} \
"
Unfortunately the llvm project has not changed the libclang version with the upgrade from 13 to 14, so there is a mismatch of the version naming of the resulting libclang (libclang.so.13) and the variable that shall be packaged which would equal to "libclang.so.14" (see: https://github.com/llvm/llvm-project/issues/53684).
As far as I understand it, they didn´t change the version due to the fact that there were no ABI changes. 

IMHO there are three options to resolve this: 

1. hardcode FILES:libclang to "${libdir}/libclang.so.13"
2. add an additional variable additionally to "MAJOR_VER" such as like "ABI_VERSION" if there is no steady relationship between these two numbers anymore
3. apply a patch to llvm-project-source at the file "CMakeLists.txt" of the libclang module and change the variable "CLANG_SONAME" to "14" instead of "13"

kraj, 2023-11-10:
I think it will be fine to hardcode it for 14 recipe, we do not allow multiple clang installs so it should be fine. I think with 15 onwards these versions match MAJOR_VER so we are good on master and newer releases.

fwuehr95, 2023-11-10:
Hi @kraj, thanks for the feedback, I opened a PR to fix this: https://github.com/kraj/meta-clang/pull/881

kraj, 2023-11-10:
Fixed with #881

closed 2023-11-10 by kraj