kraj/meta-clang #885: llvm-project-source breaks archiver.bbclass

nick-owens-eero, 2023-11-17:
**Describe the bug**
we use poky's archiver.bbclass to assist with open source copyright and license compliance. we do not redistribute parts of clang or llvm, but the archiver tasks still run when we otherwise want to use clang at build time.

the archiver tries to use a separate workdir for unpacking and patching, but this does not seem to work for `work-shared` cases like `llvm-project-source` and needs to be special cased.

`do_unpack_and_patch` should be using the archiver's special workdir, but tries to apply patches to the non-archiver workdir in work-shared.

```
ERROR: llvm-project-source-17.0.4-17.0.4-r0 do_unpack_and_patch: Applying patch '0001-libcxxabi-Find-libunwind-headers-when-LIBCXXABI_LIBU.patch' on target directory '/home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/git'
CmdError('quilt --quiltrc /home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/recipe-sysroot-native/etc/quiltrc push', 0, "stdout: Applying patch 0001-libcxxabi-Find-libunwind-headers-when-LIBCXXABI_LIBU.patch
can't find file to patch at input line 24
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 6a743f488230aa025016396208d1760835495264 Mon Sep 17 00:00:00 2001
|From: Khem Raj <raj.khem@gmail.com>
|Date: Sun, 27 Aug 2017 10:37:49 -0700
|Subject: [PATCH] libcxxabi: Find libunwind headers when
| LIBCXXABI_LIBUNWIND_INCLUDES is set
|
|Currently, when LIBCXXABI_LIBUNWIND_INCLUDES is set via CMake arguments
|then it ends up not searching the specified dir and unwind.h is not found
|especially for ARM targets
|
|This patch makes the searching synthesized directories and then set
|LIBCXXABI_LIBUNWIND_INCLUDES if its there in environment
|
|Upstream-Status: Pending
|Signed-off-by: Khem Raj <raj.khem@gmail.com>
|---
| libcxxabi/CMakeLists.txt | 16 +++++++++++-----
| 1 file changed, 11 insertions(+), 5 deletions(-)
|
|diff --git a/libcxxabi/CMakeLists.txt b/libcxxabi/CMakeLists.txt
|index f380fe6b6b92..145cb9d7ee88 100644
|--- a/libcxxabi/CMakeLists.txt
|+++ b/libcxxabi/CMakeLists.txt
--------------------------
No file to patch.  Skipping patch.
2 out of 2 hunks ignored
Patch 0001-libcxxabi-Find-libunwind-headers-when-LIBCXXABI_LIBU.patch does not apply (enforce with -f)

stderr: ")
ERROR: Logfile of failure stored in: /home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/temp/log.do_unpack_and_patch.1071623
ERROR: Task (/home/mischief/poky/meta-clang/recipes-devtools/clang/llvm-project-source.bb:do_unpack_and_patch) failed with exit code '1'
```

there's also a race here when trying to build clang native itself - something about the archiver is deleting or moving the llvm source tree.
```
$ bitbake clang-native
Loading cache: 100% |###########################################################################################################################| Time: 0:00:00
Loaded 1874 entries from dependency cache.                                
NOTE: Resolving any missing task queue dependencies
                                       
Build Configuration:                                                           
BB_VERSION           = "2.6.0"                                                 
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"   
MACHINE              = "qemux86-64"
DISTRO               = "poky"                                                  
DISTRO_VERSION       = "4.3+snapshot-cdecc1f6056a9f4ff15c6af91c15262172d04b57"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""                                                      
meta                                                                           
meta-poky                      
meta-yocto-bsp       = "master:cdecc1f6056a9f4ff15c6af91c15262172d04b57"
meta-clang           = "master:6eef6ed66097e404f826c038ff213f4dba461e1a"
                                       
Initialising tasks: 100% |######################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 15 Local 0 Mirrors 0 Missed 15 Current 282 (0% match, 94% complete)            
NOTE: Executing Tasks
WARNING: clang-native-17.0.4-r0 do_populate_lic: Could not copy license file /home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/git/clang/
LICENSE.TXT to /home/mischief/poky/build/tmp/work/x86_64-linux/clang-native/17.0.4/license-destdir/x86_64/clang-native/LICENSE.TXT.0: [Errno 2] No such file or
 directory: '/home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/git/clang/LICENSE.TXT'                             
WARNING: clang-native-17.0.4-r0 do_populate_lic: Could not copy license file /home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/git/llvm/L
ICENSE.TXT to /home/mischief/poky/build/tmp/work/x86_64-linux/clang-native/17.0.4/license-destdir/x86_64/clang-native/LICENSE.TXT.1: [Errno 2] No such file or 
directory: '/home/mischief/poky/build/tmp/work-shared/llvm-project-source-17.0.4-r0/git/llvm/LICENSE.TXT'
ERROR: clang-native-17.0.4-r0 do_populate_lic: QA Issue: clang-native: LIC_FILES_CHKSUM points to an invalid file: /home/mischief/poky/build/tmp/work-shared/ll
vm-project-source-17.0.4-r0/git/llvm/LICENSE.TXT [license-checksum]                                                                                            
ERROR: clang-native-17.0.4-r0 do_populate_lic: QA Issue: clang-native: LIC_FILES_CHKSUM points to an invalid file: /home/mischief/poky/build/tmp/work-shared/ll
vm-project-source-17.0.4-r0/git/clang/LICENSE.TXT [license-checksum] 
ERROR: clang-native-17.0.4-r0 do_populate_lic: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/mischief/poky/build/tmp/work/x86_64-linux/clang-native/17.0.4/temp/log.do_populate_lic.1146668
```

**To Reproduce**
Steps to reproduce the behavior:
1. setup poky (master), meta-clang layers
2. append
```
INHERIT += "archiver"
ARCHIVER_MODE[src] ?= "original"                                                                                                                               
ARCHIVER_MODE[diff] ?= "1"                                                                                                                                     
COPYLEFT_LICENSE_INCLUDE = '*'                                                                                                                                 
COPYLEFT_LICENSE_EXCLUDE = 'CLOSED Proprietary'                                                                                                                
```
to build/conf/local.conf

3. bitbake 

**Expected behavior**
archiver class should work with meta-clang / clang-native / llvm-project-source.

**Screenshots**

**Desktop (please complete the following information):**
- ubuntu 22.04

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kraj, 2023-11-18:
archiver functionality is implemented in a class and classes can not be extended via bbappends etc. so it has to be implemented in original archiver.bbclass in core layer.

nick-owens-eero, 2023-11-20:
i understand that this would need to go upstream. it seems to me that the archiver class is not very extensible in this special case of dealing with work-shared recipes - is this something that should be fixed more generally, or keep on special casing it?

tentatively here is my fix, but i am not sure if this is correct.
```diff
diff --git a/meta/classes/archiver.bbclass b/meta/classes/archiver.bbclass
index 4a5865d7b5..30aedd14d0 100644
--- a/meta/classes/archiver.bbclass
+++ b/meta/classes/archiver.bbclass
@@ -102,7 +102,7 @@ python () {
         return
 
     # TARGET_SYS in ARCHIVER_ARCH will break the stamp for gcc-source in multiconfig
-    if pn.startswith('gcc-source'):
+    if pn.startswith('gcc-source') or pn.startswith('llvm-project-source'):
         d.setVar('ARCHIVER_ARCH', "allarch")
 
     def hasTask(task):
@@ -278,7 +278,7 @@ python do_ar_configured() {
         pn = d.getVar('PN')
         # "gcc-source-${PV}" recipes don't have "do_configure"
         # task, so we need to run "do_preconfigure" instead
-        if pn.startswith("gcc-source-"):
+        if pn.startswith("gcc-source-") or pn.startswith('llvm-project-source'):
             d.setVar('WORKDIR', d.getVar('ARCHIVER_WORKDIR'))
             bb.build.exec_func('do_preconfigure', d)
 
@@ -460,6 +460,7 @@ def create_diff_gz(d, src_orig, src, ar_outdir):
 def is_work_shared(d):
     pn = d.getVar('PN')
     return pn.startswith('gcc-source') or \
+        (d.expand("${TMPDIR}/work-shared") in d.getVar('S')) or \
         bb.data.inherits_class('kernel', d) or \
         (bb.data.inherits_class('kernelsrc', d) and d.expand("${TMPDIR}/work-shared") in d.getVar('S'))
 
```

quaresmajose, 2023-11-21:
Your patch looks good imo, I would only change the last line:
```diff
@@ -460,6 +460,7 @@ def create_diff_gz(d, src_orig, src, ar_outdir):
 def is_work_shared(d):
     pn = d.getVar('PN')
-    return pn.startswith('gcc-source') \
+    return pn.startswith('gcc-source') or pn.startswith('llvm-project-source') \
         bb.data.inherits_class('kernel', d) or \
         (bb.data.inherits_class('kernelsrc', d) and d.expand("${TMPDIR}/work-shared") in d.getVar('S'))
```
please sent it to oe-core mailing list so you can get more help on review