kraj/meta-clang #893: kirkstone: incorrect packaging of libclang / broken bpftrace

miwojcik, 2023-12-19:
**Describe the bug**
As I mentioned in https://github.com/kraj/meta-clang/pull/881#issuecomment-1845472510, after PR https://github.com/kraj/meta-clang/pull/881 libclang is not being packaged correctly as it only contains a symlink:
```
dpkg -c ./tmp/deploy/ipk/cortexa57/libclang_14.0.6-r0_cortexa57.ipk
drwxr-xr-x root/root         0 2011-04-06 01:00 ./usr/
drwxr-xr-x root/root         0 2011-04-06 01:00 ./usr/lib/
lrwxrwxrwx root/root         0 2011-04-06 01:00 ./usr/lib/libclang.so.13 -> libclang.so.14.0.6
```
and the library is still being packaged in the clang package:
```
dpkg -c ./tmp/deploy/ipk/cortexa57/clang_14.0.6-r0_cortexa57.ipk
...
-rwxr-xr-x root/root  29451416 2011-04-06 01:00 ./usr/lib/libclang.so.14.0.6
...
```

Additionally, the packages that relied on clang package to contain the libclang library (for example bpftrace) don't work:
```
root@qemuarm64:~# bpftrace -l
bpftrace: error while loading shared libraries: libclang.so.13: cannot open shared object file: No such file or directory
```

**To Reproduce**
Reproduced on poky qemuarm64 build. Reverting the PR fixed the issue as it put back both files in the clang package:
```
$ dpkg -c ./tmp/deploy/ipk/cortexa57/clang_14.0.6-r0_cortexa57.ipk
...
lrwxrwxrwx root/root         0 2011-04-06 01:00 ./usr/lib/libclang.so.13 -> libclang.so.14.0.6
-rwxr-xr-x root/root  29451416 2011-04-06 01:00 ./usr/lib/libclang.so.14.0.6
...
```

**Expected behavior**
I see two possible solutions, but I'm open to discuss a different one:
1. Fix libclang packaging, so both symlink and the library are packaged together. Then the packages that relied on clang package to contain libclang will need to have dependencies fixed.
2. Revert the PR and keep packaging libclang with clang package.

In my opinion, reverting the PR introduces less damage as it won't trigger the chain of dependency fixes for LTS branch at a cost of not packaging the libclang correctly (that wasn't noticed for some time).

wentao-windriver, 2023-12-22:
I also got this problem, the following patch works for me.
https://github.com/kraj/meta-clang/pull/663/commits/fbd07014d73d68a9a02d7fb808c3311124eacb16

miwojcik, 2023-12-22:
Right, with this change libclang is packaged correctly:
```
$ dpkg -c ./tmp/deploy/ipk/cortexa57/libclang13_14.0.6-r0_cortexa57.ipk
drwxr-xr-x root/root         0 2011-04-06 01:00 ./usr/
drwxr-xr-x root/root         0 2011-04-06 01:00 ./usr/lib/
lrwxrwxrwx root/root         0 2011-04-06 01:00 ./usr/lib/libclang.so.13 -> libclang.so.14.0.6
-rwxr-xr-x root/root  29451416 2011-04-06 01:00 ./usr/lib/libclang.so.14.0.6
```
I was afraid it would still fail without fixing other package dependencies, but it looks like libclang is automatically picked as runtime dependency and installed. Thank you for checking @wentao-windriver! I'll submit a PR soon

closed 2023-12-29 by miwojcik