kraj/meta-clang #902: std::format not available on dunfell-clang14?

MaikVermeulen, 2024-01-10:
**Describe the bug**
It seems that `std::format` is not available when building our recipe. We're using the dunfell-clang14 branch.

**To Reproduce**
We're setting this in our recipe:
```
inherit cmake clang
TOOLCHAIN = "clang"
RUNTIME = "llvm"
LIBCPLUSPLUS = "-stdlib=libc++"
TC_CXX_RUNTIME = "llvm"
```

And using `std::format` in our C++ source file.

Then we get this error in our Yocto build process:
```
| /workdir/build-quatt-cic/tmp/work/cortexa9hf-neon-tps-linux-gnueabi/cic-cloud-metrics/1.0-r0/git/src/metrics/metrics.cpp:214:36: error: no member named 'format' in namespace 'std'; did you mean 'fmt::format'?
|         stats["time"]["tsHuman"] = std::format("{:%Y-%m-%dT%H:%M:%SZ}", now);
|                                    ^~~~~~~~~~~
|                                    fmt::format
```

**Expected behavior**
No error

kraj, 2024-01-10:
`std::format` is a c++ 20 standards feature. So please add  `-std=c++20` to CXXFLAGS or however you pass additional compiler flags to your components.

MaikVermeulen, 2024-01-11:
> `std::format` is a c++ 20 standards feature. So please add `-std=c++20` to CXXFLAGS or however you pass additional compiler flags to your components.

Thank you for your reply and good that you mention it.

We have this in our `CMakeLists.txt`:
```
set(CMAKE_POSITION_INDEPENDENT_CODE ON)
add_compile_options(-Wno-c99-extensions -Wno-pedantic)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++20")
set(CMAKE_CXX_FLAGS_DEBUG "-g3 -fsanitize=address,undefined")
set(CMAKE_CXX_FLAGS_RELEASE "-O1")
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```

I also tried adding these lines, but no luck yet:
```
add_compile_definitions(_LIBCPP_ENABLE_EXPERIMENTAL_LIBRARY)
add_definitions(-DLIBCXX_ENABLE_INCOMPLETE_FEATURES=ON)
```

kraj, 2024-01-11:
hmm rereading the comments again I see that you are using clang-14 (dunfell-clang14)
do you see it with clang-16 or clang-17 as well ?

MaikVermeulen, 2024-01-11:
Can we use that on dunfell? I'm afraid I don't see the branch for it

kraj, 2024-01-11:
> Can we use that on dunfell? I'm afraid I don't see the branch for it

We dont have anything newer than clang-14 for dunfell, but it would be a good excercise to try.

MaikVermeulen, 2024-01-17:
Hey @kraj, do you have any pointers on how to achieve e.g. clang15 on dunfell? Should I then also open a PR for `dunfell-clang15`?

I see the `clang.inc` and `layer.conf` files contain some important variables, but also I see many patch files etc. being committed

kraj, 2024-01-17:
it is a not a simple task but yes it is desirable to get to use newer clang on older yocto release.