kraj/meta-clang #932: nativesdk-clang build fails with DEBUG_BUILD enabled

lms-ts, 2024-03-28:
We are using the `nanbield` branch of meta-clang. The following behaviour can also be observed with the newest `scarthgap` commit.

We are building nativesdk-clang. The build is successful, however, with `DEBUG_BUILD = "1"` set in local.conf the build fails with following error:
```
2024-03-21 12:38:46 - INFO     - | FAILED: lib/libLLVM-17.so
[...]
2024-03-21 12:38:46 - INFO     - | /home/lms-yocto-worker-2/LY_project-lms-icecake_build_production/build/build/tmp/work/x86_64-nativesdk-oesdk-linux/nativesdk-clang/17.0.4/recipe-sysroot-native/usr/bin/x86_64-oesdk-linux/x86_64-oesdk-linux-ld: lib/libLLVMSupport.a(CrashRecoveryContext.cpp.o): relocation R_X86_64_TPOFF32 against `_ZN12_GLOBAL__N_121IsRecoveringFromCrashE' can not be used when making a shared object; recompile with -fPIC
2024-03-21 12:38:46 - INFO     - | /home/lms-yocto-worker-2/LY_project-lms-icecake_build_production/build/build/tmp/work/x86_64-nativesdk-oesdk-linux/nativesdk-clang/17.0.4/recipe-sysroot-native/usr/bin/x86_64-oesdk-linux/x86_64-oesdk-linux-ld: failed to set dynamic section sizes: bad value
2024-03-21 12:38:46 - INFO     - | x86_64-oesdk-linux-clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```
This error can be fixed by setting
```
CXXFLAGS:append:class-nativesdk = " -fPIC"
CFLAGS:append:class-nativesdk = " -fPIC"
```
in a clang_%.bbappend.

The question:
Would it be useful to always set the `-fPIC` flag for nativesdk in the clang_git.bb recipe? Or maybe conditionally, depending on the `DEBUG_BUILD` flag?

kraj, 2024-03-28:
hmm it seems to be a bug in LLVM build. Can you see why it happens with nativesdk alone ?

lms-ts, 2024-03-28:
I just verified that clang builds perfectly fine with `DEBUG_BUILD` enabled.

I don't see an obvious reason, why nativesdk specifically fails. I may need to take a closer look into the do_compile logs.

With mickledore the issue with nativesdk-clang wasn't there.

lms-ts, 2024-03-28:
[log.do_compile-nativesdkclang.txt](https://github.com/kraj/meta-clang/files/14792402/log.do_compile-nativesdkclang.txt)
[log.do_compile-clang.txt](https://github.com/kraj/meta-clang/files/14792404/log.do_compile-clang.txt)

I uploaded the logs for `bitbake clang` and the failing `bitbake nativesdk-clang` (without -fPIC added to the `CXXFLAGS`)

kraj, 2024-03-28:
if  `-DLLVM_ENABLE_PIC=ON -DCLANG_DEFAULT_PIE_ON_LINUX=ON ` is making into config step that should have done it

lms-ts, 2024-03-28:
Yeah, these 2 options are set in both the nativesdk-clang and clang, but only clang itself seems to set the `-fPIC` option without explicitly adding it to the `CXXFLAGS`.

lms-ts, 2024-03-28:
```
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Failed
-- Performing Test CXX_SUPPORTS_FPIC
-- Performing Test CXX_SUPPORTS_FPIC - Failed
CMake Warning at cmake/modules/HandleLLVMOptions.cmake:320 (message):
  -fPIC is not supported.
Call Stack (most recent call first):
  cmake/modules/HandleLLVMOptions.cmake:380 (add_flag_or_print_warning)
  CMakeLists.txt:936 (include)
```
That is in the log.do_configure of nativesdk-clang

kraj, 2024-03-28:
> ```
> -- LLVM default target triple: x86_64-unknown-linux-gnu
> -- Performing Test C_SUPPORTS_FPIC
> -- Performing Test C_SUPPORTS_FPIC - Failed
> -- Performing Test CXX_SUPPORTS_FPIC
> -- Performing Test [...]

that might be the right direction to look into. Find out why cmake test for fPIC support detection is failing ?
also see if this test fails for native or target clang builds too.

lms-ts, 2024-03-28:
The test only fails for nativesdk with debug enabled.

Thanks for your help so far, I will take a closer look next week why cmake thinks that the compiler does not support the `-fPIC` option even though I can set it manually via `CXXFLAGS` and build just fine.

lms-ts, 2024-03-29:
I got further into it.

The problem are the 

```
x86_64-oesdk-linux-clang++: warning: argument unused during compilation: '-feliminate-unused-debug-types' [-Wunused-command-line-argument]
```

warnings in the nativesdk-clang log (but strangely they appear only with `DEBUG_BUILD` enabled). This warning is not present in the clang log even though the same argument is used.

Taking a look at

https://github.com/llvm/llvm-project/blob/main/llvm/cmake/modules/HandleLLVMOptions.cmake#L336

one can see that the `-WError` flag is set during the check for `-fPIC`. So, due to the above warning, the check for the flag fails.

I have no idea yet why clang does not use the compilation flag under those specific circumstances.

kraj, 2024-03-29:
Perhaps try adding `DEBUG_FLAGS:append:class-nativesdk = "-Wno-error=unused-command-line-argument "`
but I think DEBUG_FLAGS should only be used for target packages, it does not make sense to use it for native, nativesdk, cross packages.

closed 2024-06-20 by kraj