kraj/meta-clang #935: undefined reference to `llvm::MachO::mapToArchitectureSet(llvm::ArrayRef<llvm::MachO::Target>)'

gtalarico, 2024-04-07:
Describe the bug
After I added `IMAGE_INSTALL_append += " chromium-ozone-wayland"`,  I get a clang compilation error.

Steps to reproduce the behavior:

Using Phythec BSP w/ `phytec-qt5demo-image` 
Docs - https://phytec.github.io/doc-bsp-yocto/bsp/imx8/imx8mm/pd22.1.1.html#accessing-peripherals

`bitbake phytec-qt5demo-image`

PS: sorry for the lack additional repro info, I am still trying to wrap my head around yocto

OS: : Ubuntu
Version : 18.04.6
gcc: tried 7, 8 and 10

Error:
```
| FAILED: bin/llvm-exegesis
| : && /home/gtalarico/dev/yocto/build/tmp/hosttools/g++ -isystem/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -isystem/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/include -O2 -pipe   -DNDEBUG -g0   -isystem/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/include -O2 -pipe   -L/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib                         -L/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/gtalarico/dev/yocto/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -L/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib                         -L/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/gtalarico/dev/yocto/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2    -Wl,-O3 -Wl,--gc-sections tools/llvm-exegesis/CMakeFiles/llvm-exegesis.dir/llvm-exegesis.cpp.o -o bin/llvm-exegesis  -Wl,-rpath,::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::  lib/libLLVMMCParser.a  lib/libLLVMSupport.a  lib/libLLVMX86CodeGen.a  lib/libLLVMX86AsmParser.a  lib/libLLVMX86Desc.a  lib/libLLVMX86Disassembler.a  lib/libLLVMX86Info.a  -lpthread  lib/libLLVMExegesis.a  lib/libLLVMExegesisX86.a  lib/libLLVMX86CodeGen.a  lib/libLLVMAsmPrinter.a  lib/libLLVMDebugInfoDWARF.a  lib/libLLVMCFGuard.a  lib/libLLVMX86AsmParser.a  lib/libLLVMX86Desc.a  lib/libLLVMX86Disassembler.a  lib/libLLVMX86Info.a  lib/libLLVMExegesis.a  lib/libLLVMGlobalISel.a  lib/libLLVMSelectionDAG.a  lib/libLLVMCodeGen.a  lib/libLLVMBitWriter.a  lib/libLLVMScalarOpts.a  lib/libLLVMAggressiveInstCombine.a  lib/libLLVMInstCombine.a  lib/libLLVMTransformUtils.a  lib/libLLVMMCDisassembler.a  lib/libLLVMMCJIT.a  lib/libLLVMExecutionEngine.a  lib/libLLVMTarget.a  lib/libLLVMAnalysis.a  lib/libLLVMProfileData.a  lib/libLLVMObjectYAML.a  lib/libLLVMRuntimeDyld.a  lib/libLLVMObject.a  lib/libLLVMMCParser.a  lib/libLLVMMC.a  lib/libLLVMDebugInfoCodeView.a  lib/libLLVMDebugInfoMSF.a  lib/libLLVMBitReader.a  lib/libLLVMCore.a  lib/libLLVMRemarks.a  lib/libLLVMBitstreamReader.a  lib/libLLVMTextAPI.a  lib/libLLVMBinaryFormat.a  lib/libLLVMSupport.a  -lrt  -ldl  -lpthread  -lm  /home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/recipe-sysroot-native/usr/lib/libz.so  lib/libLLVMDemangle.a && :
| lib/libLLVMTextAPI.a(TextStub.cpp.o): In function `llvm::yaml::MappingTraits<llvm::MachO::InterfaceFile const*>::NormalizedTBD_V4::denormalize(llvm::yaml::IO&)::{lambda(std::vector<(anonymous namespace)::SymbolSection, std::allocator<(anonymous namespace)::SymbolSection> > const&, llvm::MachO::SymbolFlags)#1}::operator()(std::vector<(anonymous namespace)::SymbolSection, std::allocator<(anonymous namespace)::SymbolSection> > const&, llvm::MachO::SymbolFlags) const':
| TextStub.cpp:(.text._ZZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE16NormalizedTBD_V411denormalizeERNS0_2IOEENKUlRKSt6vectorIN12_GLOBAL__N_113SymbolSectionESaISC_EENS2_11SymbolFlagsEE_clESG_SH_+0x88): undefined reference to `llvm::MachO::InterfaceFile::addSymbol(llvm::MachO::SymbolKind, llvm::StringRef, llvm::SmallVector<llvm::MachO::Target, 5u> const&, llvm::MachO::SymbolFlags)'
| TextStub.cpp:(.text._ZZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE16NormalizedTBD_V411denormalizeERNS0_2IOEENKUlRKSt6vectorIN12_GLOBAL__N_113SymbolSectionESaISC_EENS2_11SymbolFlagsEE_clESG_SH_+0xc3): undefined reference to `llvm::MachO::InterfaceFile::addSymbol(llvm::MachO::SymbolKind, llvm::StringRef, llvm::SmallVector<llvm::MachO::Target, 5u> const&, llvm::MachO::SymbolFlags)'
| TextStub.cpp:(.text._ZZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE16NormalizedTBD_V411denormalizeERNS0_2IOEENKUlRKSt6vectorIN12_GLOBAL__N_113SymbolSectionESaISC_EENS2_11SymbolFlagsEE_clESG_SH_+0xfb): undefined reference to `llvm::MachO::InterfaceFile::addSymbol(llvm::MachO::SymbolKind, llvm::StringRef, llvm::SmallVector<llvm::MachO::Target, 5u> const&, llvm::MachO::SymbolFlags)'
| TextStub.cpp:(.text._ZZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE16NormalizedTBD_V411denormalizeERNS0_2IOEENKUlRKSt6vectorIN12_GLOBAL__N_113SymbolSectionESaISC_EENS2_11SymbolFlagsEE_clESG_SH_+0x13b): undefined reference to `llvm::MachO::InterfaceFile::addSymbol(llvm::MachO::SymbolKind, llvm::StringRef, llvm::SmallVector<llvm::MachO::Target, 5u> const&, llvm::MachO::SymbolFlags)'
| TextStub.cpp:(.text._ZZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE16NormalizedTBD_V411denormalizeERNS0_2IOEENKUlRKSt6vectorIN12_GLOBAL__N_113SymbolSectionESaISC_EENS2_11SymbolFlagsEE_clESG_SH_+0x17b): undefined reference to `llvm::MachO::InterfaceFile::addSymbol(llvm::MachO::SymbolKind, llvm::StringRef, llvm::SmallVector<llvm::MachO::Target, 5u> const&, llvm::MachO::SymbolFlags)'
| lib/libLLVMTextAPI.a(TextStub.cpp.o):TextStub.cpp:(.text._ZZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE16NormalizedTBD_V411denormalizeERNS0_2IOEENKUlRKSt6vectorIN12_GLOBAL__N_113SymbolSectionESaISC_EENS2_11SymbolFlagsEE_clESG_SH_+0x1bb): more undefined references to `llvm::MachO::InterfaceFile::addSymbol(llvm::MachO::SymbolKind, llvm::StringRef, llvm::SmallVector<llvm::MachO::Target, 5u> const&, llvm::MachO::SymbolFlags)' follow
| lib/libLLVMTextAPI.a(TextStub.cpp.o): In function `llvm::yaml::MappingTraits<llvm::MachO::InterfaceFile const*>::NormalizedTBD::synthesizeTargets(llvm::MachO::ArchitectureSet, llvm::SmallSet<llvm::MachO::PlatformKind, 3u, std::less<llvm::MachO::PlatformKind> > const&)':
| TextStub.cpp:(.text._ZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE13NormalizedTBD17synthesizeTargetsENS2_15ArchitectureSetERKNS_8SmallSetINS2_12PlatformKindELj3ESt4lessISA_EEE[_ZN4llvm4yaml13MappingTraitsIPKNS_5MachO13InterfaceFileEE13NormalizedTBD17synthesizeTargetsENS2_15ArchitectureSetERKNS_8SmallSetINS2_12PlatformKindELj3ESt4lessISA_EEE]+0x93): undefined reference t [... cut ...] d/tmp/work-shared/llvm-project-source-12.0.0-r0/git/llvm/tools/llvm-mca/Views/ResourcePressureView.cpp
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/gtalarico/dev/yocto/build/tmp/work/x86_64-linux/clang-native/12.0.0-r0/temp/run.do_compile.24329' failed with exit code 1
ERROR: Task (virtual:native:/home/gtalarico/dev/yocto/sources/poky/../meta-clang/recipes-devtools/clang/clang_git.bb:do_compile) failed with exit code '1'
```

gtalarico, 2024-04-08:
By the way feel free to close this. This seems to be a problem with my build environment. But please let me know if you have any ideas on what the cause might be.

As a workaround, I was able to build it using the yocto builder docker images

kraj, 2024-04-08:
Have not seen this in meta-clang, I looked into manifest at

and it says
https://git.phytec.de/phy2octo/tree/BSP-Yocto-NXP-i.MX8MM-PD22.1.y.xml?h=imx8mm
```
	<project name="meta-clang" path="sources/meta-clang" remote="clang" revision="b0d805060791006d651efd3d7ae3dd5add8f70fe" upstream="hardknott"/>

```

hardknott is long dead.

closed 2024-04-08 by kraj