kraj/meta-clang #938: Linkage error with Clang compiler

drula, 2024-04-18:
I'm trying to use Clang for my C++ project in Yocto. I've added the following to the recipe for my project:
```
TOOLCHAIN = "clang"
TC_CXX_RUNTIME = "llvm"
```

But I got an error during the configuration of the C++ compiler: cannot find `crtbeginS.o` and `-lgcc`.

> ERROR: compiler-rt-17.0.6-r0 do_configure: ExecutionError('/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/temp/run.do_configure.23915', 1, None, None)
ERROR: Logfile of failure stored in: /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/temp/log.do_configure.23915
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/path/to/src/lfs-core/meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/path/to/src/lfs-core/src/poky/../../meta-lfs-core-poky/recipes-devtools/gcc12/gcc-runtime_12.2.bb:do_populate_sysroot', '/path/to/src/lfs-core/src/poky/../../meta-lfs-core-poky/recipes-devtools/gcc12/libgcc_12.2.bb:do_populate_sysroot', '/path/to/src/lfs-core/src/poky/../../meta-lfs-core-upstream/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/path/to/src/lfs-core/src/poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/path/to/src/lfs-core/src/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/path/to/src/lfs-core/src/poky/../../meta-lfs-core-poky/recipes-devtools/python/python3_3.10.13.bb:do_populate_sysroot', 'virtual:native:/path/to/src/lfs-core/src/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/path/to/src/lfs-core/src/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/path/to/src/lfs-core/src/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/path/to/src/lfs-core/src/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['clang-cross-aarch64', 'gcc-runtime', 'libgcc', 'glibc', 'cmake-native', 'quilt-native', 'python3-native', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'bzip2-native', 'zlib-native', 'curl-native', 'zstd-native', 'ncurses-native', 'xz-native', 're2c-native', 'clang-native', 'binutils-cross-aarch64', 'linux-libc-headers', 'openssl-native', 'sqlite3-native', 'libffi-native', 'util-linux-native', 'gdbm-native', 'libtool-native', 'libtirpc-native', 'readline-native', 'libnsl2-native', 'attr-native', 'gettext-minimal-native', 'binutils-native', 'swig-native', 'libxml2-native', 'libedit-native', 'texinfo-dummy-native', 'gnu-config-native', 'flex-native', 'perl-native', 'gettext-native', 'lzo-native', 'libpcre-native', 'm4-native', 'db-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is Clang 17.0.6
| -- The CXX compiler identification is Clang 17.0.6
| -- The ASM compiler identification is Clang with GNU-like command-line
| -- Found assembler: /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native/usr/bin/aarch64-linux-gnu/aarch64-linux-gnu-clang
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - failed
| -- Check for working C compiler: /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native/usr/bin/aarch64-linux-gnu/aarch64-linux-gnu-clang
| -- Check for working C compiler: /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native/usr/bin/aarch64-linux-gnu/aarch64-linux-gnu-clang - broken
| CMake Error at /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):
|   The C compiler
| 
|     "/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native/usr/bin/aarch64-linux-gnu/aarch64-linux-gnu-clang"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build/CMakeFiles/CMakeTmp
| 
|     Run Build Command(s):ninja cmTC_18b0b && [1/2] Building C object CMakeFiles/cmTC_18b0b.dir/testCCompiler.c.o
|     [2/2] Linking C executable cmTC_18b0b
|     FAILED: cmTC_18b0b
|     : && /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native/usr/bin/aarch64-linux-gnu/aarch64-linux-gnu-clang --target=aarch64-linux-gnu -target aarch64-linux-gnu -mcpu=neoverse-n2+crypto -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -ffile-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0=/usr/src/debug/compiler-rt/17.0.6-r0 -fmacro-prefix-map=/path/to/yoctobuild/tmp/work-shared/llvm-project-source-17.0.6-r0/git=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work-shared/llvm-project-source-17.0.6-r0/git=/usr/src/debug/compiler-rt/17.0.6-r0  -fmacro-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot=  -fmacro-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot=  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native= -target aarch64-linux-gnu -mcpu=neoverse-n2+crypto -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -ffile-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0=/usr/src/debug/compiler-rt/17.0.6-r0 -fmacro-prefix-map=/path/to/yoctobuild/tmp/work-shared/llvm-project-source-17.0.6-r0/git=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work-shared/llvm-project-source-17.0.6-r0/git=/usr/src/debug/compiler-rt/17.0.6-r0  -fmacro-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot=  -fmacro-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot=  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++   -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -ffile-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0=/usr/src/debug/compiler-rt/17.0.6-r0 -fmacro-prefix-map=/path/to/yoctobuild/tmp/work-shared/llvm-project-source-17.0.6-r0/git=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work-shared/llvm-project-source-17.0.6-r0/git=/usr/src/debug/compiler-rt/17.0.6-r0  -fmacro-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/build=/usr/src/debug/compiler-rt/17.0.6-r0  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot=  -fmacro-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot=  -fdebug-prefix-map=/path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ CMakeFiles/cmTC_18b0b.dir/testCCompiler.c.o -o cmTC_18b0b   && :
[... cut ...]
| 	#3: main, /path/to/yoctobuild/tmp/work/aarch64-linux-gnu/compiler-rt/17.0.6-r0/temp/run.do_configure.23915, line 205
ERROR: Task (/path/to/src/lfs-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1254 tasks of which 1253 didn't need to be rerun and 1 failed.
Summary: 1 task failed:
  /path/to/src/lfs-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure
Summary: There were 50 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

kraj, 2024-04-19:
@drula thanks for trying meta-clang out and reporting the issue you observed. I am not entirely clear what could be causing this error. I am guessing you are on a release and I would suggest that you try master branch and see if you still run into this error.

Secondly, what is your build host, perhaps that will provide some leads maybe.

Thirdly try [yoe distro](https://github.com/yoedistro/yoe-distro), it uses clang as default compiller and see if it works out ok for your case.

kraj, 2024-04-19:
and try setting
```
  # Use clang as default compiler
   TOOLCHAIN ?= "clang"

   # Build clang into SDK as well
   CLANGSDK = "1"

   # Use compile-rt/libcxx/libunwind as default runtime libraries
   TC_CXX_RUNTIME ?= "llvm"
   RUNTIME ?= "llvm"
```