kraj/meta-clang #956: cannot use oe-debuginfod with clang built code

abuibrahim, 2024-06-22:
**Describe the bug**
When trying to use the `oe-debuginfod` script as described in https://docs.yoctoproject.org/next/dev-manual/debugging.html#using-the-debuginfod-server-method to debug a Clang built recipe, no symbols/source can be found on remote servers from `DEBUGINFOD_URLS`

**To Reproduce**
Steps to reproduce the behavior:
1. Set `TOOLCHAIN = "clang"` and build core-image-minimal.
2. Run the image inside qemu and make sure you can ping the host.
3. On the host, run `oe-debuginfod`
4. In the VM run `export DEBUGINFOD_URLS="http://<your-host-ip>:8002/" ; gdb /bin/my-prog`

**Expected behavior**
You should see debug info and sources automatically downloaded during debuging session. This works when the same recipe is built with GCC.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

abuibrahim, 2024-06-24:
Update - adding `EXTRA_OECMAKE += "-DENABLE_LINKER_BUILD_ID=ON"` fixed it, so perhaps an optional `build-id` knob could be added to `PACKAGECONFIG` for those who want to enable it.

kraj, 2024-06-24:
> Update - adding `EXTRA_OECMAKE += "-DENABLE_LINKER_BUILD_ID=ON"` fixed it, so perhaps an optional `build-id` knob could be added to `PACKAGECONFIG` for those who want to enable it.

Thanks for following up and root causing it, this option is enabled by default in GCC in oe-core, therefore we should add it as a `packageconfig` and also enable it by default. So there is an option for someone to disable it but default can be to enable it.

closed 2024-07-04 by kraj