kraj/meta-clang #963: [Kirkstone][Aarch64] libunwind is broken with current compiler flags

ofacklam, 2024-07-03:
**Describe the bug**
Applications linked against `libunwind` as unwind library are failing to handle exceptions properly.

This happens on the kirkstone branch when compiling for an aarch64 target.

This issue is probably related to the following set of changes on the `poky` `kirkstone` branch:
- https://github.com/yoctoproject/poky/commit/8585e789650d1c55d9ed8074976ae3229d6d9a39
- https://github.com/yoctoproject/poky/commit/a2bf83842a72c3dfe24a824c650a2f64d3fdf1b0
alongside this issue in LLVM which seems to be patched only since v15:
- https://reviews.llvm.org/D123692#change-npxa8asgzwtw

**To Reproduce**
- `kirkstone` branch of `meta-clang` + `poky`
- targeting aarch64
- `TOOLCHAIN=clang`
- `RUNTIME=llvm`
- `CLANGSDK=1`

Test program:
```c++
#include <iostream>

void do_except() { throw std::runtime_error("Exception world"); }

int main() {
    try {
        do_except();
    } catch (const std::exception &e) {
        std::cout << e.what() << std::endl;
    }

    return 0;
}
```

**Expected behavior**
Expected to print the exception reason when running this on the target.

Instead, we get the following error message:
```
libc++abi: terminating with uncaught exception of type std::runtime_error: Exception world
```

**Additional context**
On our end, we managed to work around the issue by adding this to our `local.conf`
```
TUNE_CCARGS:remove:toolchain-clang = "-mbranch-protection=standard"
```
and this to our `meta-environment.bbappend`:
```
TARGET_CLANGCC_ARCH:remove = "-mbranch-protection=standard"
```

kraj, 2024-07-04:
@ofacklam thanks for detailed report, I think you have found a real problem. For now we are stuck with clang-14 on kirkstone, perhaps we will do a another backport but not planned as of now. Could you try backporting

https://github.com/llvm/llvm-project/commit/c218fd3d7d3764eb123c8429bbcd33bacfe2e633

to meta-clang/kirkstone and see if it fixes the problem ?

kraj, 2024-07-04:
Hmm, the issue shows up on master/clang-19 as well.

closed 2024-07-05 by kraj