kraj/meta-clang #975: building of clang for SDK fails due to -fPIC is not supported

ceggers-arri, 2024-07-13:
**OE Version:** Yocto Scarthgap

I've tried to build clang for SDK by adding the meta-clang
layer to my build and setting ```CLANGSDK = "1"``` in my local.conf.
But when running ```bitbake myrootfs -c populate_sdk```, linking of
libLLVM.so.18.1 fails with the following error messages:

```
/build/devel-scarthgap/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/18.1.5/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-ld: lib/libLLVMSupport.a(CrashRecoveryContext.cpp.o): relocation R_X86_64_TPOFF32 against `_ZN12_GLOBAL__N_121IsRecoveringFromCrashE' can not be used when making a shared object; recompile with -fPIC
/build/devel-scarthgap/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/18.1.5/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-ld: failed to set dynamic section sizes: bad value
x86_64-pokysdk-linux-clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

I can see that ```-fPIC``` as actually not set during compilation of ```CrashRecoveryContext.cpp```. In the ```log.do_compile``` this switch is only set for 36 out of 6320 source files. When I compare this with the ```native``` clang build, I see that ```-fPIC``` is set for almost all files.

The ```log.do_configure``` for ```native``` includes the following:
```
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Performing Test CXX_SUPPORTS_FPIC
-- Performing Test CXX_SUPPORTS_FPIC - Success
-- Building with -fPIC
```
But for ```nativesdk``` this is different:
```
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Failed
-- Performing Test CXX_SUPPORTS_FPIC
-- Performing Test CXX_SUPPORTS_FPIC - Failed
CMake Warning at cmake/modules/HandleLLVMOptions.cmake:320 (message):
  -fPIC is not supported.
Call Stack (most recent call first):
  cmake/modules/HandleLLVMOptions.cmake:380 (add_flag_or_print_warning)
  CMakeLists.txt:936 (include)wq
```

kraj, 2024-07-13:
@ceggers-arri  are you also using `DEBUG_BUILD = "1"` ? if so this could be duplicate of https://github.com/kraj/meta-clang/issues/932 we fixed it for master branch already, please test the patch from https://github.com/kraj/meta-clang/pull/955 and report back, if it works then please create a backport pull request for `scarthgap`

ceggers-arri, 2024-07-14:
Hi Khem, 

after some investigation it turned out that it's in fact a duplicate of #932. On my setup I've set ```BUILDSDK_OPTIMIZATION``` similar as for target, so  also ```-feliminate-unused-debug-types``` was set (independently of ```BUILD_DEBUG```).

In ```build/CMakeFiles/CMakeConfigureLog.yaml``` then I see the following errors:
```
        x86_64-pokysdk-linux-clang: error: argument unused during compilation: '-feliminate-unused-debug-types' [-Werror,-Wunused-command-line-argument]
        x86_64-pokysdk-linux-clang: error: argument unused during compilation: '-feliminate-unused-debug-types' [-Werror,-Wunused-command-line-argument]
```

So it clear now why the ```native``` build works fine while the ```nativesdk``` build doesn't (the former uses the host gcc, the latter clang). After removing ```-feliminate-unused-debug-types``` from ```BUILDSDK_OPTIMIZATION```, everything looks fine.

Just on Friday Richard sent a [patch](https://lists.openembedded.org/g/openembedded-core/topic/patch_1_3_bitbake_conf/107177953) which removes ```-feliminate-unused-debug-types``` at all. Looking into the GCC source I found an entry in [Changelog-2012](https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/ChangeLog-2012;h=900077783c9086feb232083c66d415b8d5330616;hb=refs/heads/master#l10781) which states that ```-feliminate-unused-debug-types``` is enabled by default since then. So instead of working around this problem as in #955, it may be worth to remove this switch for all maintained OE core branches.