kraj/meta-clang #991: Regression of  #645 - compiler-rt and libcxx contain reference to TMPDIR [buildpaths]

fidelski, 2024-08-06:
**What is happening**
In the current master the build of a system that requests `CLANGSDK = "1"` fails due to errors relating to QA Issues:

```
ERROR: compiler-rt-18.1.8-r0 do_package_qa: QA Issue: File /usr/lib/clang/18.1.8/lib/linux/libclang_rt.builtins-powerpc64.a in package compiler-rt-staticdev contains reference to TMPDIR
File /usr/lib/clang/18.1.8/lib/linux/liborc_rt-powerpc64.a in package compiler-rt-staticdev contains reference to TMPDIR [buildpaths]
ERROR: compiler-rt-18.1.8-r0 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /mnt/data/yocto/bf133/build/tmp/work/ppc64e5500-vrag-linux/compiler-rt/18.1.8/temp/log.do_package_qa.4021297
ERROR: Task (/mnt/data/yocto/bf133/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_package_qa) failed with exit code '1'
ERROR: libcxx-18.1.8-r0 do_package_qa: QA Issue: File /usr/lib64/.debug/libc++abi.so.1.0 in package libcxx-dbg contains reference to TMPDIR
File /usr/lib64/.debug/libc++.so.1.0 in package libcxx-dbg contains reference to TMPDIR [buildpaths]
ERROR: libcxx-18.1.8-r0 do_package_qa: QA Issue: File /usr/lib64/libc++.a in package libcxx-staticdev contains reference to TMPDIR
File /usr/lib64/libc++abi.a in package libcxx-staticdev contains reference to TMPDIR
File /usr/lib64/libc++experimental.a in package libcxx-staticdev contains reference to TMPDIR [buildpaths]
ERROR: libcxx-18.1.8-r0 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /mnt/data/yocto/bf133/build/tmp/work/ppc64e5500-vrag-linux/libcxx/18.1.8/temp/log.do_package_qa.4018378
ERROR: Task (/mnt/data/yocto/bf133/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_package_qa) failed with exit code '1'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use current master of meta-clang
2. Enter `CLANGSDK = "1"` in the build configuration
4. run `bitbake {myimage}`
6. The build fails with the error shown

**Expected behavior**
The build succeeds and the Clang SDK is built correctly.

kraj, 2024-08-07:
I think we need to find the place in llvm sources where the source path is being emitted into `clang -v`  output it should just emit

`clang version 18.1.8 `

kraj, 2024-08-08:
@fidelski Are you using tip of master branch or some other branch ? we have recently switched away from building the compiler from git checkout and resorted to tarballs  but I see your build is coming out of a git tree which is in `TMPDIR`

I think cmake's git version and repo url detection logic is getting triggerred and doing unwanted things here

I have a patch here
https://github.com/kraj/meta-clang/pull/993

Please try it out and lemme know.

fidelski, 2024-08-08:
Thanks a lot for your quick response and patch! I was using tip of master branch, no idea why it should build differently here :man_shrugging: 
Your patch does work, I needed to add `-DLLVM_APPEND_VC_REV=OFF` to basically all occurrences of `EXTRA_OECMAKE` though, only setting it in `clang_git.bb` was not sufficient.

kraj, 2024-08-09:
should be fixed now.

closed 2024-08-09 by kraj